KejPi/AbracaDABra #14: Airspy TCP server does not work

KejPi, 2022-06-02:
For some reason Airspy TCP server does not work a application does not connect. Different protocol?

andimik, 2022-06-11:
see https://github.com/TLeconte/airspy_tcp , in the README.md according to the sentence

> It will work with any rtl_tcp capable frontend

it should work.

The tcp connection has successfully established:

```
$ ./airspy_tcp
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control parameters (frequency, gain, ...).
client accepted!
start rx
```

But the app says:

> RTLTCP: Trying to connect to: 127.0.0.1
> RTLSDR_TUNER_R820T
> RTL_TCP WARNING: unexpected number of gain values reported by server 22

This is the output of `airspy_info `

```
$ airspy_info 
airspy_lib_version: 1.0.11

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-6-g4008185 2020-05-08
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x04A464C8377B3D0B
Supported sample rates:
	6.000000 MSPS
	3.000000 MSPS
Close board 1
```

andimik, 2022-06-11:
I tried the airspy_tcp with Welle.io

```
$ ./airspy_tcp    
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control parameters (frequency, gain, ...).
client accepted!
start rx
sample rate 2048000 not supported
```
(its output is 3M)

I will try under Windows, there is an Airspyinferface from Andi_Bayern from rundfunkforum : https://www.ukwtv.de/cms/index.php?option=com_content&view=article&id=281&Itemid=550&jsmallfib=1&dir=JSROOT/dabplayer/AirspyInterface

KejPi, 2022-06-12:
3 MHz sample rate is a problem, the library needs 2.048 MHz that is standard for DAB, I would have to implement downsampling to support higher rate.

andimik, 2022-06-12:
I fear you are right.

KejPi, 2022-06-12:
I will try to look at that for 1.2 version. Now I am finishing 1.1, I hope I could release it soon, but now I do not have much time to work on that 🙁

andimik, 2022-07-28:
I suggest to close the issue as this is no longer needed as Airspy is now supported directly without this TCP server.

closed 2022-07-29 by KejPi