KejPi/AbracaDABra #21: error when building radiocontrol.cpp

andimik, 2022-06-11:
```
$ make
[  2%] Automatic MOC and UIC for target AbracaDABra
[  2%] Built target AbracaDABra_autogen
[  5%] Generating qrc_resources.cpp
[  8%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o
[ 10%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/qrc_resources.cpp.o
[ 13%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/main.cpp.o
[ 16%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/mainwindow.cpp.o
[ 18%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/clickablelabel.cpp.o
[ 21%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/dabtables.cpp.o
[ 24%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/radiocontrol.cpp.o
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:810:2: warning: #warning "Debug - remove or make implementation clean" [-Wcpp]
  810 | #warning "Debug - remove or make implementation clean"
      |  ^~~~~~~
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp: In member function 'void RadioControl::eventFromDab(RadioControlEvent*)':
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:74:9: error: 'dabsdrNtfResetFlags_t' was not declared in this scope; did you mean 'dabsdrNtfEnsemble_t'?
   74 |         dabsdrNtfResetFlags_t flags = static_cast<dabsdrNtfResetFlags_t>(pEvent->pData);
      |         ^~~~~~~~~~~~~~~~~~~~~
      |         dabsdrNtfEnsemble_t
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:75:17: error: 'flags' was not declared in this scope; did you mean 'QFlags'?
   75 |         switch (flags)
      |                 ^~~~~
      |                 QFlags
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:77:14: error: 'DABSDR_RESET_INIT' was not declared in this scope
   77 |         case DABSDR_RESET_INIT:
      |              ^~~~~~~~~~~~~~~~~
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:81:14: error: 'DABSDR_RESET_NEW_EID' was not declared in this scope
   81 |         case DABSDR_RESET_NEW_EID:
      |              ^~~~~~~~~~~~~~~~~~~~
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp: In function 'void dabNotificationCb(dabsdrNotificationCBData_t*, void*)':
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:1302:50: error: 'dabsdrNtfResetFlags_t' was not declared in this scope; did you mean 'dabsdrNtfEnsemble_t'?
 1302 |         pEvent->pData = static_cast<intptr_t>(*((dabsdrNtfResetFlags_t*) p->pData));
      |                                                  ^~~~~~~~~~~~~~~~~~~~~
      |                                                  dabsdrNtfEnsemble_t
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:1302:72: error: expected primary-expression before ')' token
 1302 |         pEvent->pData = static_cast<intptr_t>(*((dabsdrNtfResetFlags_t*) p->pData));
      |                                                                        ^
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:1302:73: error: expected ')' before 'p'
 1302 |         pEvent->pData = static_cast<intptr_t>(*((dabsdrNtfResetFlags_t*) p->pData));
      |                                                ~                        ^~
      |                                                                         )
/home/andreas/apps/AbracaDABra/gui/radiocontrol.cpp:1302:84: error: expected ')' before ';' token
 1302 |         pEvent->pData = static_cast<intptr_t>(*((dabsdrNtfResetFlags_t*) p->pData));
      |                                                                                    ^
      |                                                                                    )
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:196: gui/CMakeFiles/AbracaDABra.dir/radiocontrol.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:99: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

I randomly tried several previous commits and found b195b77f20 from 3rd June which worked for me.

KejPi, 2022-06-12:
New version of DABSDR library is needed, binaries a headers are not committed yet.

closed 2022-06-12 by KejPi

andimik, 2022-06-12:
Thanks for the explanation

andimik, 2022-06-22:
Thanks for the latest update, now it works.

KejPi, 2022-06-23:
Very good, I was about to ask you to try 😉 I am going to release v1.1 soon.