KejPi/AbracaDABra #24: wrong codec detection - therefore no audio

andimik, 2022-07-01:
I have realized that the audio codec is very often not detected correctly and therefore I don't get a sound, no DLS and no SLS slideshow.

[image]

This should be HE-AAC, not AAC-LC, same here:

[image]

[image]

The console says for instance

```
=== MCI complete
AAC-LC 32 kHz mono
Output SR = 32000, channels = 2
```

I wonder the line

```
Muting... [available 2880 samples]
```
because then there is no following

```
Unmuting audio
```
for such silent audio services.

I need to try several other services before I get a sound. Switching back to the previous service may result (= not always) in correct codec detection and I may get audio.

This following screenshot was taken after 10 attempts ...

[image]

And I tried both, RTLTCP and RTLSDR. And it seems, rawfiles are not affected.

Hint: I don't think this is the issue #17

KejPi, 2022-07-01:
I have never seen this issue on my antenna signals and test stream I have. In theory there should be no difference for the audio detection between input devices. In other words it should be reproducible also from raw file. This would be actually the only possibility for me to reproduce if you can share some raw files, longer better. Can you share some recordings? 

Additionally, could you try apply the attached patch and check what number is printed in command line in faulty case? Is the number always the same? 

[printAACheader.zip](https://github.com/KejPi/AbracaDABra/files/9026746/printAACheader.zip)

andimik, 2022-07-01:
Let me try on another computer in the next days.

KejPi, 2022-07-01:
One additional question. Is this issue new for 1.1 or you have observed this also in previous version?

andimik, 2022-07-01:
Good question. Let me test it.

andimik, 2022-07-01:
no, [1e3a6c5](https://github.com/KejPi/AbracaDABra/commit/1e3a6c5aa601dcceb84427f28aaa83c45806b8f4) did not help in 1.1+

[image]

[image]
[image]

There is only a little chance for audio.

Switching back to 1.0:

```
git reset --hard 87aa017
```
There is still the same issue, no sound, no SLS, no DLS for the most attempts. Only a few attempts with audio.

Back to 1.1 I've applied the patch file you sent me.

For those with no audio I get lots of `F0` (I guess that is for HE-AACv1 services) or `E8` (all HE-AACv2 services) or `20`, 

For working audio `68` (HE-AACv2), `70` (HE-AACv1), `20` (HE-AACv1 32 kHz mono) or `60` (don't know what that stands for)

[image]
(20, with SLS and DLS, but no sound) 

So, `20` is not unique and could be either sound or no sound (but with SLS and DLS)

The service Radio Student has normal audio in Qt-DAB, so this is not related to the multiplex.
(Hint: the red bar means mono)

[image]

I found another hint, maybe that could help you to identify it.

For sound, the error rate is 0

[image]

for no sound there are CRC errors  **only**.

[image]

Here is the log for Austrian's bundesmux, I clicked each service for around 5 seconds.

[abracaDABra_log_patch.txt](https://github.com/KejPi/AbracaDABra/files/9031715/abracaDABra_log_patch.txt)

Remark: all other tools (Qt-DAB, welle.io, Qirx, etc) are working with all audio services.

Could you download this 150 MB raw file (recorded with AbracaDABra) and test it on your machine? (BTW: the constellation diagram of this recording looks very poor compared to live, don't know if this is another bug)

https://we.tl/t-T5rXMToUq1

KejPi, 2022-07-02:
Thanks for the recording, I will try next week. It seems that there is a problem with CRC, but not clear why, because it seems that the signal is good enough. Can you also copy text from ensemble information for good and bad case? And I assume that with https://github.com/KejPi/AbracaDABra/commit/1e3a6c5aa601dcceb84427f28aaa83c45806b8f4 there is either no aac format detected or correct format. No longer wrong format, can you confirm?

andimik, 2022-07-03:
I tried another laptop with the same Ubuntu 22.04 and there it works fine.

So I reinstalled all necessary packages, but no luck.

Later in the afternoon I will purge and install them again.

andimik, 2022-07-04:
Update: The Windows version works well on the same laptop where I have problems under Ubuntu 22.04. So it's not related to the tool itself and seems to be a local issue on my computer under Ubuntu only.

I also tried the appimage under Ubuntu, same result.

I just need to investigate, if one of the packages (needed for e.g. `dablin`, `ODR-DabMod` and `eti-tools`, which I have installed in addition to my other PC) causes that issue.

Therefore, I close this issue unless I find out what causes the incompatibility.

andimik, 2022-07-04:
I found out that it might has to do with 48 kHz sampling rate.

Because all (!) stations with 32 kHz are working all the time.

```
Found XPAD app 2
HE-AAC 32 kHz mono
Output SR = 32000, channels = 2
Unmuting audio
```

I tried the following muxes

| Mux  | all services ok? | content |
| ------------- | ------------- | --- |
| TRT Turkey  | yes, all ok  | no SPI, only 48 kHz  |
| ERT Greece | partly | only 48 kHz |
| DAB Italia | partly | only 48 kHz |
| Bundesmux Germany | partly | only 48 kHz |
| RAS 10D Südtirol | nearly all | only 48 kHz |
| RAS 10B Südtirol | some | only 48 kHz |
| Mux 1 Slovenia | some, all 32 kHz | 32 and 48 kHz |
| Mux 2E Slovenia | partly | only 48 kHz |
| Mux 2W Slovenia | partly | only 48 kHz |
| Bundesmux Austria | some | only 48 kHz |
| Salisbury UK | all 32 kHz | DAB 48 kHz and DAB+ 32 kHz |

```
$ dpkg -l | grep portau
ii  libportaudio2:amd64                        19.6.0-1.1                              amd64        Portable audio I/O - shared library
ii  libportaudiocpp0:amd64                     19.6.0-1.1                              amd64        Portable audio I/O C++ bindings - shared library
ii  portaudio19-dev:amd64                      19.6.0-1.1                              amd64        Portable audio I/O - development files
```

andimik, 2022-07-05:
Further attempts:

I tried to unselect portaudio in cmake and even uninstall portaudio (= as this is not used under Windows).

And I built it with option `cmake .. -DUSE_FDK_AAC=ON`.

Still 100% audio loss on lots of 48 kHz programs (some work from time to time). 32 kHz programs always work.

[image]

More and more I feel this is really a AbracaDABra bug and not a misleading constellation on my system.

KejPi, 2022-07-05:
I would say the problem is not the audio decoder or audio output. The numbers printed by experimental patch show problems with CRC (bit 7 is set). It will be probably in libdabsdr, I will look at this next week, I think there is nothing wrong with your system and there is not much you can do to fix it. But in any case, thank you for all your reports a trials. Still any recording you can share is highly appreciated!

KejPi, 2022-07-06:
If the issue can be reproduced using files and ODR-DabMod then I should be able to reproduce the problem too. I have ODR-DABMOD on my Mac.

andimik, 2022-07-07:
If you could send me another patch which prints more information in the console, then we can locate it where the bug comes from. 

Because I fear only I can reproduce it on my machine according to my hardware or/and software packages.

For me I can reproduce it on a raw file as well, but you won't see a difference.

KejPi, 2022-07-09:
Please find attached the debug version of libdabsdr with some messages that might help me to understand what is going on. This library was built under Ubuntu 22.04, apart from debug messages it has a small fix that might solve the problem. This was the only potential problem I was able to find. Please let me know your results, I have tried to reproduce the issue with the recording you have shared and with other recordings from the forum as well as on my antenna signals (most of the services @ 48kHz) but it always works on Mac.

[libdabsdr_1.1.1_linux_x86_64.tar.gz](https://github.com/KejPi/AbracaDABra/files/9077609/libdabsdr_1.1.1_linux_x86_64.tar.gz)

andimik, 2022-07-10:
Thanks, it did not help, but attached you find the console output. I have clicked through all my stations I can receive, 5 seconds each.

[abracadabra.log](https://github.com/KejPi/AbracaDABra/files/9079633/abracadabra.log)

KejPi, 2022-07-10:
Well, I do not know what the problem could be. Obviously there is a lot of errors that is why you do not have audio, the question is why they are there. I do not understand why the problem is not reproducible from raw file recorded using IQ dump (Ensemble info -> Dump raw data). Can you confirm that if you dump the stream in AbracaDABra and then you play it via raw file device the problem is not present any more? 
I can see huge frequency offset in some of the screenshots you have shared (~13kHz). Can you confirm this offset also for the log you shared? What was the signal quality? And are you using SW AGC?

andimik, 2022-07-10:
> I do not understand why the problem is not reproducible from raw file recorded using IQ dump

I told you: This is only related to my Ubuntu installation on that particular laptop. I still need to find out what could be the reason.

Is it my soundcard? Also tried an external one.
Is it the kernel? Also tried a previous kernel.
Is it the one of the software packages? Also tried reinstall of all required packages.

```
$ sudo apt-get install --reinstall git cmake build-essential mesa-common-dev libusb-dev librtlsdr-dev libfaad2 mpg123 libmpg123-dev libfaad-dev portaudio19-dev qt6-base-dev qt6-multimedia-dev libqt6svg6-dev rtl-sdr
```

Next try is to use the appimage under a live system (where I still need to install the rtl-sdr driver)

KejPi, 2022-07-10:
I would exclude sound card and kernel. If the input stream is correct (this can be checked by dumping it and playing it from file) then it is inside the application related to live playback. The dump takes input data and stores them to file as they are coming from rtldsr device. If it is playing from file correctly then it means that data are correct and the problem is caused in the application during the processing. The fact that you see a lot of RS/CRC errors means that this happens before audio decoder so there is no problem with audio decoder, audio output and sound card. 
One idea would be the slow CPU (what is the CPU load when AbracaDABra is running?) the other is big frequency offset that I could not test. 

Please try to set
` #define RTLSDR_DOC_ENABLE  0`
on rtlsdrinput.h line 10. 

This is the last idea I have to explain difference between rtlsdr input and raw file input. Otherwise there is no difference apart from CPU load caused by handling USB input.

andimik, 2022-07-10:
I am writing these lines now under Ubuntu 22.04 live CD on the same laptop.

Booting from ISO file and chose `try Ubuntu`.
Then I had to install the rtl-driver:

```
sudo add-apt-repository universe
sudo apt-get install libusb-dev librtlsdr-dev rtl-sdr
```

then 

```
sudo apt-get install fuse libfuse2
```
in order to get the appimage running.
Then as root:
```
echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb
```

**It still shows the same issue.**

It has nothing to do with reception (= because same issue for raw files, same with generated raw files from ETI input). 
No changing when disable the software AGC (= because same issue for raw files).

Your suggestions don't seem reasonable.

- My laptop is fast enough (for all other kind of DAB software, this is the 10th or so software for DAB applications I have).
- define RTLSDR_DOC_ENABLE 0 makes no sense (= because same issue for raw files)
- It's not the soundcard, tried with external
- I feared it had to do with DVB drivers (which caused problems with my graphics card), but it's not the case

So my conclusion is the problem needs to be found in your code. Please try to focus on 48 kHz/32 kHz.

> If it is playing from file correctly

No, it does not. When I first wrote the issue I tested it with a raw file where - by chance - all were in 32 kHz, so please ignore the sentence

> And it seems, rawfiles are not affected.

because it's wrong.

andimik, 2022-07-11:
> I just need to investigate, if one of the packages (needed for e.g. dablin, ODR-DabMod and eti-tools, which I have installed in addition to my other PC) causes that issue.

No, see above. Clean Live CD without any other tools installed.

KejPi, 2022-07-11:
Let me summarize my current understanding of the issue - please correct whatever is wrong:
- it happens on one of your Linux PC, the other is without problem although both are running Ubuntu 22.04
- it can be reproduced on both rtlsdr input as well as on raw filer input
- it seems that 32kHz services are always working while 48kHz services fail
- it can be reproduces on more raw files, not specific to single raw file
- not related to Ubuntu configuration, issue can be reproduced on clean Live system using released AppImage

Open points / questions - please answer:
- can it be reproduced on any raw file? 
- can you reproduce the issue on these files recorded from my antenna? 
    [2022-07-11_105219_5A.raw](https://www.dropbox.com/s/tgbavn0lgigv9tn/2022-07-11_105219_5A.raw?dl=0)
    [2022-07-11_105405_7A.raw](https://www.dropbox.com/s/dxks4xhofmtdepx/2022-07-11_105405_7A.raw?dl=0)
    [2022-07-11_105700_12C](https://www.dropbox.com/s/skb4cb2fznpz7ki/2022-07-11_105700_12C.raw?dl=0)
- can it be always reproduced on 48kHz services or sometimes it works and sometimes not?
- what CPU is in the laptop that is affected by the problem? 
- any other observations that I missed?

KejPi, 2022-07-11:
FDK-AAC issue should be fixed now.

Please try with attached library [linux_x86_64_1.1.2.tar.gz](https://github.com/KejPi/AbracaDABra/files/9087100/linux_x86_64_1.1.2.tar.gz)

This is my last idea - it this is not helping I will probably give it up until I have some way to reproduce the problem :-(

andimik, 2022-07-11:
Wow, progress! 

Now it works for all station as long as I don't select a station with a different sampling rate. 

For example I first select all 48 kHz services then they play all perfectly, but the first 32 kHz breaks it.

Important: Then all (!) following stations (48 and 32 kHz) have no audio (but SLS and DLS works) any more. You need to restart the tool.

Your 5A plays 100% (also RADIO UKRAJINA). All on 7A work, but on 12C I have selected **UKRAJINSKE-RADIO** at the end, which is in 32 kHz (no audio, but SLS and DLS work). And then all others won't have audio any more.

Log:

After the start I have chosen a 48 kHz station
```
=== MCI complete
23:41:39.748916 [I] : Superframe SYNC :-)
23:41:39.748978 [D] : AU header: 0x70
23:41:39.749022 [D] :   => ps_flag=0, dac_rate=1, sbr_flag=1, aac_channel_mode=1
23:41:39.749070 [D] :   => HE-AAC 48 kHz stereo
HE-AAC 48 kHz stereo
Output SR = 48000, channels = 2
Unmuting audio
```

Then after selecting a 32 kHz service no audio:
```
Muting... [available 2880 samples]
Found XPAD app 2
23:41:47.326193 [I] : RS detected uncorrectable errors
23:41:47.332030 [I] : RS detected uncorrectable errors
23:41:47.333064 [I] : RS detected uncorrectable errors
23:41:47.334088 [I] : RS detected uncorrectable errors
23:41:47.335090 [I] : Superframe SYNC :-)
23:41:47.335110 [D] : AU header: 0x20
23:41:47.335150 [D] :   => ps_flag=0, dac_rate=0, sbr_flag=1, aac_channel_mode=0
23:41:47.335197 [D] :   => HE-AAC 32 kHz mono
HE-AAC 32 kHz mono
Output SR = 32000, channels = 2
```

Back to a 48 kHz station, still no audio, but no error on the console.

```
Found XPAD app 2
23:42:53.327055 [I] : RS detected uncorrectable errors
23:42:53.333329 [I] : Superframe SYNC :-)
23:42:53.333350 [D] : AU header: 0x68
23:42:53.333401 [D] :   => ps_flag=1, dac_rate=1, sbr_flag=1, aac_channel_mode=0
23:42:53.333455 [D] :   => HE-AAC v2 48 kHz stereo
```

KejPi, 2022-07-12:
Thanks for the report. Now it looks more like an issue with AAC decoder or audio output. I would like to ask you to do another trial with the attached patch. Please apply it on 056b5d6e7ad4005931186472f848d92d93d5341e using default FAAD decoder.

[logAAC_audio2file.zip](https://github.com/KejPi/AbracaDABra/files/9091242/logAAC_audio2file.zip)

After you compile the code I would like to ask you to follow these steps:

_Preparation_

1. Run app
2. Go to settings
3. Choose Raw file input 
4. Select 12C raw file
5. Apply settings
6. Exit the applications

_Test_ 

1. Run app
2. Choose any DAB channel (this start processing of 12C raw file)
3. Select any 48kHz service
4. You should have audio at this point -> let it play for few seconds
5. Choose UKRAJINSKE-RADIO 
6. There should be no audio - let it play for few seconds
7. Go back to 48kHz service
8. There should be no audio, let it play for few seconds
9. Exit app
 
The application will print a lot of logs during the test, please copy it and share.
Additionally it will create 2 files called audio.raw and audio.aac in the folder where the executable is. Please share these files too.

andimik, 2022-07-12:
ok, first patch these two files (BTW: I had to learn how to patch 2 instead of 1 files ...) 

```
~/apps/AbracaDABra/gui$ patch < /tmp/logAAC_audio2file.patch 
patching file audiodecoder.cpp
patching file audiodecoder.h
```
Then `cmake` without any options and `make` afterwards.

```
~/apps/AbracaDABra/build$ cmake ..
make
```

delete the `ini` file.
```
rm ~/.config/AbracaDABra/AbracaDABra.ini
```
Start the tool, stop the scan, go to expert settings and chose RAW. Close the app. 
Restart the app. Scan 12C (does not matter). Play a 48 kHz station, then switch to the only 32 kHz audio.

This is the log file from console.
[audio.zip](https://github.com/KejPi/AbracaDABra/files/9097025/audio.zip)
[abracadabra_12c.log](https://github.com/KejPi/AbracaDABra/files/9097034/abracadabra_12c.log)

KejPi, 2022-07-13:
You can apply the patch using _git apply_ command. 
I am sorry my instructions were probably misleading. Please correct my if I am wrong, but it seems that you used the DABSDR library from repository not the one that I shared recently. 
Could you please redo the test with [linux_x86_64_1.1.2.tar.gz](https://github.com/KejPi/AbracaDABra/files/9087100/linux_x86_64_1.1.2.tar.gz) 

Just to be clear: 
1. Get AbracaDABra source revision 056b5d6e7ad4005931186472f848d92d93d5341e 
2.  apply patch ` git apply logAAC_audio2file.patch`
3. Copy  [linux_x86_64_1.1.2.tar.gz](https://github.com/KejPi/AbracaDABra/files/9087100/linux_x86_64_1.1.2.tar.gz)  to lib folder
4. Compile without special options: `mkdir build && cd build && cmake .. && make`
5. Do the test procedure with preparation phase as described above (no need to delete ini file)
6. Share log and audio files.

Thanks and sorry for additional request.

andimik, 2022-07-13:
this is the log:

Switched between CRo-C.BUDEJOVICE, then UKRAJINSKE-RADIO and back to CRo-C.BUDEJOVICE

```
$ ./gui/AbracaDABra
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
	Service: CRo-VLTAVA SID = 0xE2232D, SCIdS = 0
	Service: CRo-RADIOZURNAL SID = 0xE2232F, SCIdS = 0
	Service: CRo-C.BUDEJOVICE SID = 0xE2250D, SCIdS = 0
	Service: CRo-RZ SPORT SID = 0xE22F2C, SCIdS = 0
	Service: UKRAJINSKE-RADIO SID = 0xE22FE0, SCIdS = 0
	Service: CRo-PARDUBICE SID = 0xE2280C, SCIdS = 0
	Service: CRo-LIBEREC SID = 0xE22706, SCIdS = 0
	Service: CRo-DAB PRAHA SID = 0xE22F1A, SCIdS = 0
	Service: CRo-POHODA SID = 0xE22F1B, SCIdS = 0
	Service: CRo-PLZEN SID = 0xE22608, SCIdS = 0
	Service: CRo-REGION SID = 0xE2243F, SCIdS = 0
	Service: CRo-WAVE SID = 0xE22F23, SCIdS = 0
	Service: CRo-JAZZ SID = 0xE22F27, SCIdS = 0
	Service: CRo-DVOJKA SID = 0xE2232E, SCIdS = 0
	Service: CRo-JUNIOR SID = 0xE22F25, SCIdS = 0
	Service: CRo-DDUR SID = 0xE22F26, SCIdS = 0
	Service: CRo-PLUS SID = 0xE22424, SCIdS = 0
=== MCI complete
Found XPAD app 2
22:25:39.590610 [I] : RS detected uncorrectable errors
22:25:39.595825 [I] : RS detected uncorrectable errors
22:25:39.597295 [I] : RS detected uncorrectable errors
22:25:39.598717 [I] : Superframe SYNC :-)
22:25:39.598735 [D] : AU header: 0x68
22:25:39.598765 [D] :   => ps_flag=1, dac_rate=1, sbr_flag=1, aac_channel_mode=0
22:25:39.598803 [D] :   => HE-AAC v2 48 kHz stereo
Decoder: AAC header 0x68
HE-AAC v2 48 kHz stereo
Output SR = 48000, channels = 2
Decoder: error = 0 | bytesconsumed = 199 | samples = 0
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 240 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 209 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 208 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 220 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 220 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 210 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 240 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 198 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 194 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 241 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 213 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 211 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 238 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 199 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 201 | samples = 3840
Decoder: AAC header 0x68
Decoder: error = 0 | bytesconsumed = 247 | samples = 3840
Decoder: AAC header 0x68
[... 435 lines cut ...]
Decoder: AAC header 0x20
Decoder: error = 0 | bytesconsumed = 103 | samples = 3840
Decoder: AAC header 0x20
Decoder: error = 0 | bytesconsumed = 108 | samples = 3840
Decoder: AAC header 0x20
Decoder: error = 0 | bytesconsumed = 114 | samples = 3840
Decoder: AAC header 0x20
Decoder: error = 0 | bytesconsumed = 97 | samples = 3840
Decoder: AAC header 0x20
Decoder: error = 0 | bytesconsumed = 112 | samples = 3840
Decoder: AAC header 0x20
Decoder: error = 0 | bytesconsumed = 99 | samples = 3840
Decoder: void AudioDecoder::handleAudioOutputFAAD(const NeAACDecFrameInfo&, const uint8_t*) 368640: 0 < 7680
Found XPAD app 2
22:25:57.590679 [I] : RS detected uncorrectable errors
22:25:57.595741 [I] : Superframe SYNC :-)
22:25:57.595764 [D] : AU header: 0x68
22:25:57.595881 [D] :   => ps_flag=1, dac_rate=1, sbr_flag=1, aac_channel_mode=0
22:25:57.596005 [D] :   => HE-AAC v2 48 kHz stereo
22:26:13.187989 [I] : Finishing thread...

22:26:13.188929 [I] : Thread joined
```

[audio.zip](https://github.com/KejPi/AbracaDABra/files/9105814/audio.zip)

andimik, 2022-07-13:
BTW: The app does not close correctly ... Need to use `Ctrl + c`

KejPi, 2022-07-14:
Thanks for the report. If I understand well, the application behaved during the test like this:
1. CRo-C.BUDEJOVICE: audio out, no problem
2. UKRAJINSKE-RADIO: no audio
3. CRo-C.BUDEJOVICE: no audio, no further logs and application needs to be stopped by CTRL+C

If this was the behavior, then the problem is with audio output. It is not able to switch between sample rates for some reason. 
I am sorry I have to ask you to do few more tests :-( 

**Test 1:**
1. Restore git working copy to its original state: `git reset --hard`
2. Apply this patch:  [logAAC_audio2file_portaudioDeviceList.zip](https://github.com/KejPi/AbracaDABra/files/9109826/logAAC_audio2file_portaudioDeviceList.zip)
`git apply logAAC_audio2file_portaudioDeviceList.patch`
3. Copy  [linux_x86_64_1.1.2.tar.gz](https://github.com/KejPi/AbracaDABra/files/9087100/linux_x86_64_1.1.2.tar.gz) to lib folder.
4. Compile `mkdir build && cd build && cmake .. && make`
5. Do the test as before, share logs

**Test 2:**
1. Leave the working copy as it is
2. Delete build folder
3. Compile application without portaudio, thus:
      ```
      mkdir build
      cd build 
      cmake .. -DUSE_PORTAUDIO=OFF
      make
      ```
      Please note that it is mandatory to delete build folder before.
4. Do the test as before, share logs and results.

andimik, 2022-07-14:
> If this was the behavior

Yes, correct.

I will test it in the evening.

As I seem to be the first Linux user, I am grad to help here.

The tool itself is amazing and has much better performance (reception) than all other software.

andimik, 2022-07-14:
# Test 1

Now CRo-C.BUDEJOVICE and UKRAJINSKE-RADIO have both audio
The app quits normally.

[abracadabra_12C_14july2022.txt](https://github.com/KejPi/AbracaDABra/files/9114072/abracadabra_12C_14july2022.txt)
[audio.zip](https://github.com/KejPi/AbracaDABra/files/9114074/audio.zip)

# Test 2

> Please note that it is mandatory to delete build folder before.

Yes, sure. I always did it (resp. `rm -r *`)

CRo-C.BUDEJOVICE and UKRAJINSKE-RADIO have both audio
The app quits normally.

[abracadabra_12C_portaudiooff_14july2022.txt](https://github.com/KejPi/AbracaDABra/files/9114112/abracadabra_12C_portaudiooff_14july2022.txt)
[audio_portaudiooff.zip](https://github.com/KejPi/AbracaDABra/files/9114118/audio_portaudiooff.zip)

# Hint

There is still a message saying 

> Selected file : "/home/andreas/Downloads/2022-07-11_105700_12C.raw" Sample format unknown

Guess this is from `sampledialog.cpp` Line 311 as the extension is raw.

KejPi, 2022-07-14:
Test 2 result is surprising, I would expect that it would not work. However more important for me is that Test1 works - this is the default configuration. Please try without patch but with modified library for the ensembles you have and please report any issue. If it is working I would consider audio problem solved and I will focus on DABSDR library optimizations.

andimik, 2022-07-14:
I will test it later regarding the audio, but at the start now I get lots of infos about `portaudio` on the console compared to before.

```
$ ./gui/AbracaDABra
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
ALSA lib pcm_dsnoop.c:601:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=Datei oder Verzeichnis nicht gefunden)
attempt to connect to server failed
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=Datei oder Verzeichnis nicht gefunden)
attempt to connect to server failed
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
connect(2) call to /dev/shm/jack-1000/default/jack_0 failed (err=Datei oder Verzeichnis nicht gefunden)
attempt to connect to server failed
PortAudio version: 0x00130600
Version text: 'PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4'
Number of devices = 5
--------------------------------------- device #0
Name                        = HDA ATI HDMI: 0 (hw:0,3)
Host API                    = ALSA
Max inputs = 0, Max outputs = 8
Default low input latency   =  -1,0000
Default low output latency  =   0,0058
Default high input latency  =  -1,0000
Default high output latency =   0,0348
Default sample rate         = 44100,00
Supported standard sample rates
 for half-duplex 16 bit 8 channel output = 
	32000,00, 44100,00, 48000,00, 88200,00,
	96000,00, 192000,00
--------------------------------------- device #1
Name                        = HD-Audio Generic: CX20590 Analog (hw:1,0)
Host API                    = ALSA
Max inputs = 2, Max outputs = 0
Default low input latency   =   0,0058
Default low output latency  =  -1,0000
Default high input latency  =   0,0348
Default high output latency =  -1,0000
Default sample rate         = 44100,00
Supported standard sample rates
 for half-duplex 16 bit 2 channel input = 
	44100,00, 48000,00, 96000,00
--------------------------------------- device #2
Name                        = hdmi
Host API                    = ALSA
Max inputs = 0, Max outputs = 8
Default low input latency   =  -1,0000
Default low output latency  =   0,0058
Default high input latency  =  -1,0000
Default high output latency =   0,0348
Default sample rate         = 44100,00
Supported standard sample rates
 for half-duplex 16 bit 8 channel output = 
	32000,00, 44100,00, 48000,00, 88200,00,
	96000,00, 192000,00
--------------------------------------- device #3
Name                        = pulse
Host API                    = ALSA
Max inputs = 32, Max outputs = 32
Default low input latency   =   0,0087
Default low output latency  =   0,0087
[... 42 lines cut ...]
	192000,00
Supported standard sample rates
 for full-duplex 16 bit 32 channel input, 32 channel output = 
	 8000,00,  9600,00, 11025,00, 12000,00,
	16000,00, 22050,00, 24000,00, 32000,00,
	44100,00, 48000,00, 88200,00, 96000,00,
	192000,00
	Service: CRo-VLTAVA SID = 0xE2232D, SCIdS = 0
	Service: CRo-DVOJKA SID = 0xE2232E, SCIdS = 0
	Service: CRo-RADIOZURNAL SID = 0xE2232F, SCIdS = 0
	Service: CRo-PLUS SID = 0xE22424, SCIdS = 0
	Service: CRo-REGION SID = 0xE2243F, SCIdS = 0
	Service: CRo-C.BUDEJOVICE SID = 0xE2250D, SCIdS = 0
	Service: CRo-PLZEN SID = 0xE22608, SCIdS = 0
	Service: CRo-LIBEREC SID = 0xE22706, SCIdS = 0
	Service: CRo-PARDUBICE SID = 0xE2280C, SCIdS = 0
	Service: CRo-DAB PRAHA SID = 0xE22F1A, SCIdS = 0
	Service: CRo-POHODA SID = 0xE22F1B, SCIdS = 0
	Service: CRo-WAVE SID = 0xE22F23, SCIdS = 0
	Service: CRo-JUNIOR SID = 0xE22F25, SCIdS = 0
	Service: CRo-DDUR SID = 0xE22F26, SCIdS = 0
	Service: CRo-JAZZ SID = 0xE22F27, SCIdS = 0
	Service: CRo-RZ SPORT SID = 0xE22F2C, SCIdS = 0
	Service: UKRAJINSKE-RADIO SID = 0xE22FE0, SCIdS = 0
```

andimik, 2022-07-14:
> EDIT: yes the message is because from .raw extension it is not clear what the sample format is. I should probably use .u8 for all files that the application dumps.

Use `*.u8.raw` which then is clearer ...

KejPi, 2022-07-14:
Yes, I have added PortAudio info in the patch to understand what sample rates are supported by your HW. I will probably add the check for supported sample rates in the code as default to avoid problems in future.

I will maybe change the filename later, I still consider it as debug option not to be used by average user.

If and only if everything works with clean repository + and dabsdr library 1.1.2 shared before, please try with attached version of the library. I have re-enabled the optimization but changed compiler flags that might help with compatibility. 

[linux_x86_64_1.1.3.tar.gz](https://github.com/KejPi/AbracaDABra/files/9114416/linux_x86_64_1.1.3.tar.gz)

andimik, 2022-07-14:
Do you mean 3cb5f0e166dd in combination with 1.1.3 without the patches? Well, it does not work

```
21:12:33.256759 [I] : RS detected uncorrectable errors
21:12:33.257540 [I] : RS detected uncorrectable errors
21:12:33.258345 [I] : RS detected uncorrectable errors
```

KejPi, 2022-07-14:
It seems I was too optimistic :-( 
Could you please try this?
[linux_x86_64_1.1.4.tar.gz](https://github.com/KejPi/AbracaDABra/files/9114754/linux_x86_64_1.1.4.tar.gz)

andimik, 2022-07-14:
no, does not work

1.1.2 is the only one with 100% audio

KejPi, 2022-07-14:
It seems to be some problem with SSE. Could you please compile and check FFTS:
```
git clone https://github.com/anthonix/ffts.git
cd ffts
mkdir build
cd build
cmake ..
make 
./ffts_test
```
Please share output from cmake and from ffts_test, thanks!

andimik, 2022-07-14:
# ffts

```
$ cmake ..
-- The C compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for malloc.h
-- Looking for malloc.h - found
-- Looking for mm_malloc.h
-- Looking for mm_malloc.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stdlib.h
-- Looking for stdlib.h - found
-- Looking for string.h
-- Looking for string.h - found
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for aligned_alloc
-- Looking for aligned_alloc - found
-- Looking for memalign
-- Looking for memalign - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for valloc
-- Looking for valloc - found
-- Looking for _mm_malloc
-- Looking for _mm_malloc - not found
-- Looking for aligned_alloc
-- Looking for aligned_alloc - found
-- Looking for memalign
-- Looking for memalign - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for valloc
-- Looking for valloc - found
-- Looking for _mm_malloc
-- Looking for _mm_malloc - not found
-- Looking for xmmintrin.h
-- Looking for xmmintrin.h - found
-- Looking for emmintrin.h
-- Looking for emmintrin.h - found
-- Looking for pmmintrin.h
-- Looking for pmmintrin.h - found
-- Performing Test HAVE_GCC_VISIBILITY
-- Performing Test HAVE_GCC_VISIBILITY - Success
-- Looking for pow in m
-- Looking for pow in m - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andreas/apps/ffts/build
```
```
$ make
[  8%] Building C object CMakeFiles/ffts_static.dir/src/ffts.c.o
[ 16%] Building C object CMakeFiles/ffts_static.dir/src/ffts_chirp_z.c.o
[ 25%] Building C object CMakeFiles/ffts_static.dir/src/ffts_nd.c.o
[ 33%] Building C object CMakeFiles/ffts_static.dir/src/ffts_real.c.o
[ 41%] Building C object CMakeFiles/ffts_static.dir/src/ffts_real_nd.c.o
[ 50%] Building C object CMakeFiles/ffts_static.dir/src/ffts_transpose.c.o
[ 58%] Building C object CMakeFiles/ffts_static.dir/src/ffts_trig.c.o
[ 66%] Building C object CMakeFiles/ffts_static.dir/src/ffts_static.c.o
/home/andreas/apps/ffts/src/ffts_static.c:239:36: warning: ‘ffts_constants_inv_64f’ defined but not used [-Wunused-const-variable=]
  239 | static const FFTS_ALIGN(16) double ffts_constants_inv_64f[16] = {
      |                                    ^~~~~~~~~~~~~~~~~~~~~~
/home/andreas/apps/ffts/src/ffts_static.c:195:36: warning: ‘ffts_constants_64f’ defined but not used [-Wunused-const-variable=]
  195 | static const FFTS_ALIGN(16) double ffts_constants_64f[16] = {
      |                                    ^~~~~~~~~~~~~~~~~~
/home/andreas/apps/ffts/src/ffts_static.c:141:36: warning: ‘ffts_constants_small_inv_64f’ defined but not used [-Wunused-const-variable=]
  141 | static const FFTS_ALIGN(16) double ffts_constants_small_inv_64f[24] = {
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/andreas/apps/ffts/src/ffts_static.c:77:36: warning: ‘ffts_constants_small_64f’ defined but not used [-Wunused-const-variable=]
   77 | static const FFTS_ALIGN(16) double ffts_constants_small_64f[24] = {
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~
[ 75%] Building C object CMakeFiles/ffts_static.dir/src/codegen.c.o
/home/andreas/apps/ffts/src/codegen.c: In function ‘ffts_generate_func_code’:
/home/andreas/apps/ffts/src/codegen.c:282:19: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int32_t’ {aka ‘int’} [-Wsign-compare]
  282 |         if (ws_is != pLUT) {
      |                   ^~
/home/andreas/apps/ffts/src/codegen.c:121:15: warning: unused variable ‘len’ [-Wunused-variable]
  121 |     ptrdiff_t len;
      |               ^~~
[ 83%] Linking C static library libffts.a
[ 83%] Built target ffts_static
[ 91%] Building C object CMakeFiles/ffts_test.dir/tests/test.c.o
[100%] Linking C executable ffts_test
[100%] Built target ffts_test
```

```
$ ./ffts_test 
 Sign |      Size |     L2 Error
------+-----------+-------------
  -1  |         2 | 8.659561E-17
  -1  |         4 | 1.145552E-16
  -1  |         8 | 1.210162E-08
  -1  |        16 | 2.220916E-08
  -1  |        32 | 2.396272E-08
  -1  |        64 | 2.272872E-08
  -1  |       128 | 1.940460E-08
  -1  |       256 | 2.081312E-08
  -1  |       512 | 2.045939E-08
  -1  |      1024 | 2.079071E-08
  -1  |      2048 | 2.074052E-08
  -1  |      4096 | 2.124981E-08
  -1  |      8192 | 2.086463E-08
  -1  |     16384 | 2.071494E-08
  -1  |     32768 | 2.062090E-08
  -1  |     65536 | 2.054694E-08
  -1  |    131072 | 2.060355E-08
  -1  |    262144 | 2.054523E-08
   1  |         2 | 8.659561E-17
   1  |         4 | 1.145552E-16
   1  |         8 | 1.210162E-08
   1  |        16 | 2.220916E-08
   1  |        32 | 2.396272E-08
   1  |        64 | 2.272872E-08
   1  |       128 | 1.940460E-08
   1  |       256 | 2.081312E-08
   1  |       512 | 2.045939E-08
   1  |      1024 | 2.079071E-08
   1  |      2048 | 2.074052E-08
   1  |      4096 | 2.124981E-08
   1  |      8192 | 2.086463E-08
   1  |     16384 | 2.071494E-08
   1  |     32768 | 2.062090E-08
   1  |     65536 | 2.054694E-08
   1  |    131072 | 2.060355E-08
   1  |    262144 | 2.054523E-08
```

KejPi, 2022-07-14:
Thanks, so it works. I have to review the code, it will take some time. Meanwhile you should be able to run the app with library 1.1.2, in case you find any problem please report. Once I get some idea how to fit this issue, I will ask you to do another test. Thank you for your help!

KejPi, 2022-07-14:
One additional question. For the same raw file input, do you see the same frequency offset when using library 1.1.2 and 1.1.4? (you can check it in Ensemble Information dialog). it should be about 263 Hz for 12C raw file.

andimik, 2022-07-14:
Yes, around 263 Hertz

# 1.1.4

[image]

# 1.1.2

[image]

KejPi, 2022-07-18:
Could you please check with this library when you have time: [linux_x86_64_1.1.5.tar.gz](https://github.com/KejPi/AbracaDABra/files/9134585/linux_x86_64_1.1.5.tar.gz)

andimik, 2022-07-18:
# 1.1.5

Strange new bug (in 1.1.5 only) detected:

I can **hardly** listen to **any** DAB+ station (AAC) when its bitrate is exactly **80 kBit** (lots, lots, lots of attempts to get a sound). I have tried many files from Germany (Bayern), Norway (nearly all public ones from NRK have 80 kBit), Czech Republic (including yours, for example EVROPA 2, FREKVENCE 1) and live reception from Austria (Technikum ONE) and Slovenia (Rock Radio) 

For all those 80 kBit stations I nearly always (99% of all attempts) get

> 21:45:11.518578 [I] : RS detected uncorrectable errors

The rest works.

Hint: 80 kBit mp2 works for my very old RAW-recordings (Irish RTE from 2017, RAI Südtirol from 2016), on the other hand.

# 1.1.2

80 kBit bug not existing

KejPi, 2022-07-18:
I do not have 80kbps issue on my Linux machine (tried with 7A raw file shared before) and I am running out of ideas. The same code works under Windows, Mac and my Linux machine but not on your HW running the same Ubuntu 22.04. It is very difficult to find problem that I cannot reproduce whatever I try to do :-(

andimik, 2022-07-18:
But you have changed something in 1.1.5 (= it worked in 1.1.2) which prevents me listening to 80 kBit stations. All other bitrates are working.

KejPi, 2022-07-19:
The last result is even more confusing than everything before :-(

I have no explanation for the 80kbps services problem, there is not difference from library point of vie between 80kbps and 64kbps.

Some additional questions: Do all non-working services have the same parameters? I mean these parameters (EVROPA 2 and FREKVENCE 1 from 7A raw file: _NumCU: 80, EEP 2-A, Coderate: 3/8, Bitrate: 80kbps_). For example Schwarzwaldradio in the raw file below is also 80kbps but with different protection  _NumCU: 60, EEP 3-A, Coderate: 1/2, Bitrate: 80kbps_ Does it work or not? 
[2022-07-19_083557_7C.raw](https://www.dropbox.com/s/xi259xadnsj7r6u/2022-07-19_083557_7C.raw?dl=0)

And what about other services from this raw file - are all working?

andimik, 2022-07-19:
It's not related to the protection level. Tested 2A and 3A. Only services with 80 kBit AAC are affected. All others work (incl. 80 kBit mp2). 

Let me generate some more test files with nearly all Bitrates and PLs. I did it in the past, but used 64  kBit each...

andimik, 2022-07-19:
I will test your file in the evening.

KejPi, 2022-07-19:
I have meanwhile identified two potential OS dependent issues (not related to 80kbps specifically) - I will fix them in the evening and provide the library update. Please wait with your tests for this new library. Maybe it will fix the problem or there will create new problem again :-(

KejPi, 2022-07-19:
Please try with this library: [linux_x86_64_1.1.6.tar.gz](https://github.com/KejPi/AbracaDABra/files/9143676/linux_x86_64_1.1.6.tar.gz)

KejPi, 2022-07-19:
Thank YOU for all your tests, but I would not close it yet. The latest 1.1.6 is not the final version, I still need more of your tests.

andimik, 2022-07-19:
BTW: If you are interested, this is an eti file with four 80 kBit services in PLs 1A, 2A, 3A, and 4A.
[test_protlev_80kbit.zip](https://github.com/KejPi/AbracaDABra/files/9144409/test_protlev_80kbit.zip)

```
odr-dabmod ./test_protlev_80kbit.eti -f ./test_protlev_80kbit.raw -F u8 -a 0.2
```

KejPi, 2022-07-19:
Please try again with this version that has all optimizations enabled. This should be the final version from functionality point of view, I still need to clean up the code 
[linux_x86_64_1.1.7.tar.gz](https://github.com/KejPi/AbracaDABra/files/9144424/linux_x86_64_1.1.7.tar.gz)

KejPi, 2022-07-19:
For sure I am interested, I am collecting all test files ;-)

andimik, 2022-07-19:
This is an eti with all 8 possible DAB+ protection levels I have recorded some time ago by myself ...
[test_protlev.zip](https://github.com/KejPi/AbracaDABra/files/9144811/test_protlev.zip)

```
odr-dabmod ./test_protlev.eti -f ./test_protlev.raw -F u8 -a 0.8
```

KejPi, 2022-07-20:
Please check and if everything works we can finally close this issue. Thanks for reporting and testing!

andimik, 2022-07-21:
I'll check next week and let you know.

andimik, 2022-07-21:
I've audio on all channels. Tested all terrestrial ensembles and some files which made problems before.

[image]

So let's close it.

[1 more reply]

closed 2022-07-21 by KejPi