KejPi/AbracaDABra #25: fifo raw file not supported

andimik, 2022-07-01:
I've realized that a fifo is not supported (in Qt-DAB or welle.io this works). The raw file needs to be a real file, because at the start the tools checks its length:

```
Number of samples: 0 => 0.000000 sec

Expected NULLs: 0
```

I am using ODR-DabMod's output (which is a standard 2048kHz raw file 8u) sent to a fifo-file which I want to use in the various tools.

So in this case, I save ODR-DabMod's output to a (temporary) file and open it then.

```
Number of samples: 45613056 => 22.272000 sec

Expected NULLs: 232
```

KejPi, 2022-07-09:
Could you please share more details about the use-case? What is the command line od ODR-DabMod and what exactly do you want to play?

andimik, 2022-07-10:
Well, the idea is 

a) to use much smaller eti files instead of huge raw recordings
and
b) use live reception of satellite feeds

for example

```
mkfifo ~/stdin.raw
```

then load an eti file and modulate it as raw file

```
$ cat /media/andreas/Data/dab/nrk1.eti | odr-dabmod /dev/stdin -f ~/stdin.raw -F u8 
```
This is then the **Qt-DAB** window (I have selected the `~/stdin.raw` file as input)

[image]

And - of course it should work with live reception, see list in https://github.com/piratfm/eti-tools#satellite-dab-feeds 

But in **AbracaDABra** the same (valid) fifo-file cannot be loaded as there is a check for the file length.

**But I found the bug in `rawfileinput.cpp` in lines 50 - 53.**

When I comment out these lines, then it works in **AbracaDABra**:

```
//    if (getNumSamples() > 0)
//    {
//        emit deviceReady();
//    }
```
[image]
[image]

Do you really need this `if`-command?

KejPi, 2022-07-10:
Well, I do not need this check - I will remove it - as I wrote somewhere on forum - I consider raw file input as a kind of debuging feature, I did expect that someone would like to use it regularly.

closed 2022-07-14 by andimik