KejPi/AbracaDABra #26: Feature: SoapySDR support

nmaster2042, 2022-07-22:
Support for SoapySDR lib would bring support for many SDR devices instead of adding specific support for each one.
It would make usable SDRPlay, Limesdr, hackrf and other.

KejPi, 2022-10-13:
SoapySDR support added in 2114d31b374313e0c3a1803a4429567f4c3ce500 and previous commits. Implementation was only tested with RTL-SDR and Airspy Mini, volunteers for testing with other supported HW are welcome. Please report bugs and problems.

SoapySDR is disabled by default, enable it by (requires SoapySDR installation):

`cmake .. -DSOAPYSDR=ON`

nmaster2042, 2022-10-19:
Thank you for adding this feature. I will make a try.

nmaster2042, 2022-10-24:
Tried with SDRPlay RSP1, works fine.
Thank you for the great work.

KejPi, 2022-10-24:
I am happy it works, thank you for your feedback!

closed 2022-10-24 by KejPi