KejPi/AbracaDABra #28: Prevent entry Frequency=0 in ini-file

andimik, 2022-07-27:
During listening to this radio station it suddenly came a 2nd entry for the complete mux (maybe due to reception problems) but without the Ensemble Label and no DAB channel information.

[image]

And the `ini`-file [AbracaDABra.zip](https://github.com/KejPi/AbracaDABra/files/9202728/AbracaDABra.zip) got this invalid entry, which might cause this issue:

```
1\Ensemble\1\Frequency=0
1\Ensemble\1\Label=
1\Ensemble\1\ShortLabel=
```

So I would suggest the tool not to save such invalid frequencies.

KejPi, 2022-07-29:
This is probably caused by combination of two ensembles in the same channel. I would be interested to get a recording 😉 This is a consequence of implementing endsemble change (issue #8). For sure I can implement the proposed workaround but it will not help with the cause.

andimik, 2022-07-29:
I can't receive two different ensembles on one channel. I guess this was a reception problem.

KejPi, 2022-07-29:
OK, I will check that.

andimik, 2022-07-30:
> This is probably caused by combination of two ensembles in the same channel.

No, otherwise frequency would not be 0.

KejPi, 2022-08-03:
It should be fixed now, closing.

closed 2022-08-03 by KejPi