KejPi/AbracaDABra #29: use XMLHeader for rawfile recordings

andimik, 2022-07-28:
Instead of having a simple raw file recording in raw 8bit I suggest to replace the current format by adding the XMLHeader as described in https://qirx.softsyst.com/RawXmlHeader

Clem Schmidt and Jan van Katwijk have implemented this in their tools **Qirx** and **Qt-DAB**, too.

You will also want to read them afterwards, too.

For example:

```
<?xml version="1.0" encoding="utf-8"?>
<SDR>
 <Recorder Name="AbracaDABra" Version="1.1+"/>
 <Device Model="Generic RTL2832U OEM" Name="rtlsdr"/>
 <Time Value="Thu Jul 28 23:23:00 2022&#xa;" Unit="UTC"/>
 <Sample>
  <Samplerate Value="2048000" Unit="Hz"/>
  <Channels Container="uint8" Bits="8" Ordering="LSB">
   <Channel Value="I"/>
   <Channel Value="Q"/>
  </Channels>
 </Sample>
 <Datablocks>
  <Datablock Number="1" Count="14456914" Unit="Channel">
   <Frequency Value="215072" Unit="KHz"/>
   <Modulation Value="DAB"/>
  </Datablock>
 </Datablocks>
</SDR>

```

[image]

I've made an example with **Qt-DAB** and patched the header 

[Generic RTL2832U OEM-10D-Do--Juli-28-2022-23-22-50.zip](https://github.com/KejPi/AbracaDABra/files/9214434/Generic.RTL2832U.OEM-10D-Do--Juli-28-2022-23-22-50.zip)

[image]
(offset of the rawfile is 1388 bytes)

KejPi, 2022-07-29:
I Iike the idea but I have doubts about the format itself. When I have time I will contact authors.

KejPi, 2023-01-27:
Implemented in 80f90bbc5b81d09151bd7efa5a92fa2c2b079157 however, the application always records the IQ stream @ 2048kHz (i.e. after potential downsampling) and not the real input stream coming from the device. It is also the case for reading - XML header is recognised and parsed but input file must be recoded @ 2048kHz either in uint8 or int16 format to play correctly.

closed 2023-01-27 by KejPi

andimik, 2023-01-28:
As the header shows the sampling rate, a recording made by AbracaDABra should play in Qt-DAB and Qirx, I suppose.

Thanks for the implementation! I will test it later.

KejPi, 2023-01-28:
In theory yes but I have not tested, neither is running on Apple Silicon.