KejPi/AbracaDABra #30: Expert mode: Change service source icon shown for SFNs

andimik, 2022-07-29:
[image]

The icon **Change service source** should only appear when a mux can be received on different DAB channels, right?

In **Expert mode** it appears even without having duplicates (and has no function of course). Seems the icon is always appearing after the 3rd manual scan (although all of the ensembles are in an SFN). I tried different combinations of all 4 DAB channels I can receive here.

I guess the issue is whenever I do **not** click on a service (see left top corner: **No service**).

This is not related to **Basic mode**, as the first entry will be highlighted immediately after the scan.

KejPi, 2022-07-29:
What do you mean by 3rd manual scan? Is it that you do the band scan 3 times in row?
Actually first service should start playing after scan for expert mode too.

andimik, 2022-07-29:
> What do you mean by 3rd manual scan?

The icon **always** appears as soon as there are 3 ensembles in the list. No matter, if they have duplicates.

Not **Band scan**, but selecting the 5A...13F channels manually, see screenshots below.

> Is it that you do the band scan 3 times in row?

No, not really. I select all DAB channels one after the other. In my case I have 5 channels. After the 3rd the icon appears suddenly. Non of the services found have duplicates (not even data services).

# Step 1, empty list

[image]

# Step 2, selecting 10D (1st channel)

[image]

# Step 3, selecting 12B (2nd channel)

[image]

# Step 4, selecting 12C (3rd channel) and icon appears

[image]

KejPi, 2022-07-29:
I can reproduce that :-) thanks for reporting, I will fix it.

closed 2022-07-29 by andimik