KejPi/AbracaDABra #34: "Tuner Gain" issue

gvanem, 2022-08-19:
When running the official 1.2 version under a debugger (cdb.exe), I note a lot of these `qDebug()` messages:
```
[unknown:0] - RTLSDR: Failed to set tuner gain
[unknown:0] - RTLSDR: Failed to set tuner gain
...
```

Now I've built it myself using `clang-cl.exe` (x64) and see no such traces. Linked as a console program:
[image]

AFAICS an issue with the `librtlsdr` that is bundled with this release. Where is it from?

*PS*. In my build, I use this [RTLSDR](https://github.com/old-dab/rtlsdr) library using *WinUSB* directly. No libusb involved.

KejPi, 2022-08-19:
RTLK-SDR comes form here https://github.com/osmocom/rtl-sdr.git
I have built all my libs in MingGW64 (including also libusb-1.0.25). All was built with gcc. I use the app under Windows most of the time and never saw any problem with gain settings using RTL-SDR dongle v3. I have not seen any problem with Nooelec with E4k tuner either, but I use this dongle under Mac.

gvanem, 2022-08-19:
Not a big deal. Probably some issue with libusb (?). Works like a charm with WinUSB though.

closed 2022-09-01 by gvanem