KejPi/AbracaDABra #40: AUDIOOUTPUT_RAW_FILE_OUT (debug function) is on per default

andimik, 2022-10-17:
In https://github.com/KejPi/AbracaDABra/commit/381769ef64bea65acfc2850dae0add2d2d84d206 you have enabled the debug function `AUDIOOUTPUT_RAW_FILE_OUT` by mistake (?), so all audio files are written unnecessarily to the disc.

[image]

See the audio.raw file.

[image]

On the other hand, if I really want to enable it, in particular ` #define AUDIO_DECODER_RAW_OUT ` then there is a build error:

```
$ make
[  2%] Automatic MOC and UIC for target AbracaDABra
[  2%] Built target AbracaDABra_autogen
Consolidate compiler generated dependencies of target AbracaDABra
[  5%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/audiodecoder.cpp.o
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp: In member function 'void AudioDecoder::handleAudioOutputFAAD(const NeAACDecFrameInfo&, const uint8_t*)':
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:818:16: error: 'outBufferPtr' was not declared in this scope; did you mean 'm_outBufferPtr'?
  818 |         fwrite(outBufferPtr, 1, bytesToWrite, m_rawOut);
      |                ^~~~~~~~~~~~
      |                m_outBufferPtr
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:365: gui/CMakeFiles/AbracaDABra.dir/audiodecoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:99: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

And when I want to set `#define AUDIO_DECODER_AAC_OUT` the error is a bit different:

```
$ make
[  2%] Automatic MOC and UIC for target AbracaDABra
[  2%] Built target AbracaDABra_autogen
Consolidate compiler generated dependencies of target AbracaDABra
[  5%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o
[  7%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/main.cpp.o
[ 10%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/mainwindow.cpp.o
[ 12%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/audiodecoder.cpp.o
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp: In member function 'void AudioDecoder::writeAACOutput(const char*, uint16_t)':
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:876:10: error: 'aacHeader' was not declared in this scope; did you mean 'm_aacHeader'?
  876 |     if ((aacHeader.bits.dac_rate == 0) && (aacHeader.bits.sbr_flag == 1))
      |          ^~~~~~~~~
      |          m_aacHeader
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:881:10: error: 'aacHeader' was not declared in this scope; did you mean 'm_aacHeader'?
  881 |     if ((aacHeader.bits.dac_rate == 1) && (aacHeader.bits.sbr_flag == 1))
      |          ^~~~~~~~~
      |          m_aacHeader
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:886:10: error: 'aacHeader' was not declared in this scope; did you mean 'm_aacHeader'?
  886 |     if ((aacHeader.bits.dac_rate == 0) && (aacHeader.bits.sbr_flag == 0))
      |          ^~~~~~~~~
      |          m_aacHeader
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:891:10: error: 'aacHeader' was not declared in this scope; did you mean 'm_aacHeader'?
  891 |     if ((aacHeader.bits.dac_rate == 1) && (aacHeader.bits.sbr_flag == 0))
      |          ^~~~~~~~~
      |          m_aacHeader
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:917:29: error: 'aacHeader' was not declared in this scope; did you mean 'aac_header'?
  917 |     uint_fast8_t sbr_flag = aacHeader.bits.sbr_flag;
      |                             ^~~~~~~~~
      |                             aac_header
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:365: gui/CMakeFiles/AbracaDABra.dir/audiodecoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:99: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

andimik, 2022-10-17:
This is my patch in order to fix it.

[diff.txt](https://github.com/KejPi/AbracaDABra/files/9801998/diff.txt)

KejPi, 2022-10-17:
Thanks for checking and fixing my bugs :-)

closed 2022-10-17 by KejPi