KejPi/AbracaDABra #42: Confusing Ensemble view

gvanem, 2022-12-21:
In the below picture when I select to scan ensemble 12D, programs in the previous ensemble 12B shows up too. 
*P5 Hits* is part of 12D but *NRK P2* is part of 12B. Why is it listed in this *Riks* ensemble list? Rather confusing.

[image]

andimik, 2022-12-21:
@gvanem

Dear Gisle, the column on the left shows all services of all scanned muxes in alphabetical order.

I guess the drop down box 12D was confusing you.

gvanem, 2022-12-21:
>  the column on the left shows all services of all scanned muxes

The header (*Riks*) doesn't indicate that. And neither the drop-down *12D* either.
Replace with one only; some kind of dynamic drop-down list?
```
New ensemble ->
  |___    .... fill up as it's scanned
  ...
 NRK Reg4Vest ->
 |___ NRK Jazz
 |___ NRK P2
  ....
Riks ->
  |___ P5 Hits
  |___ P6 Rock
  ....

```

andimik, 2022-12-22:
Please see right column, which is ordered by channel and then by service.

If you switch to basic mode you might not complain... 😁

KejPi, 2022-12-22:
The information on top of the service list (left column) is currently tuned ensemble. In fact this information is redundant and according to DAB Rules of implementation should not be displayed because service is the only information user needs to know. Nevertheless I wanted to show more information especially in expert mode. Currently I have no plans to change the application UI unless something significantly better is proposed. For sure I will keep service list (left column) as it is, I might consider reshuffling the information about currently tuned ensemble to make it less confusing but as already stated - this is an expert view, you might find basic view less confusing.

gvanem, 2022-12-23:
Okay, very well. Closing.

*PS*. Adding transmitter locations would be nice. Using the CSV from:
https://www.dablist.org/qtdab/qtdab_dabtx_data.php

closed 2022-12-23 by gvanem

andimik, 2022-12-23:
Not necessarily needed

You can save it as raw file and load it in Qirx.