KejPi/AbracaDABra #43: Freeze when selecting other DAB channels

andimik, 2023-01-04:
When scrolling through the DAB channels (5A - 13F) I am regularly facing a freeze.

This is the `gdb` backtrace output:

```
$ gdb ~/apps/AbracaDABra/build/gui/AbracaDABra
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra...
(No debugging symbols found in /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra)
(gdb) run
Starting program: /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[New Thread 0x7ffff11c8640 (LWP 12761)]
[New Thread 0x7fffeb776640 (LWP 12762)]
[New Thread 0x7fffeaf52640 (LWP 12763)]
[New Thread 0x7fffea719640 (LWP 12764)]
[New Thread 0x7fffe9f18640 (LWP 12765)]
[New Thread 0x7fffe9286640 (LWP 12766)]
[New Thread 0x7fffd3fff640 (LWP 12767)]
[New Thread 0x7fffd37fe640 (LWP 12768)]
[New Thread 0x7fffd2ffd640 (LWP 12769)]
[New Thread 0x7fffd27fc640 (LWP 12770)]
[New Thread 0x7fffd1ffb640 (LWP 12771)]
[New Thread 0x7fffd17fa640 (LWP 12772)]
[New Thread 0x7fffd0ff9640 (LWP 12773)]
ALSA lib pcm_dsnoop.c:601:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
[New Thread 0x7fffe8175640 (LWP 12774)]
Cannot connect to server socket err = Datei oder Verzeichnis nicht gefunden
Cannot connect to server request channel
jack server is not running or cannot be started
[Thread 0x7fffe8175640 (LWP 12774) exited]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[New Thread 0x7fffe8175640 (LWP 12775)]
Cannot connect to server socket err = Datei oder Verzeichnis nicht gefunden
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[Thread 0x7fffe8175640 (LWP 12775) exited]
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
[New Thread 0x7fffaffff640 (LWP 12776)]
[Thread 0x7fffaffff640 (LWP 12776) exited]
[New Thread 0x7fffaffff640 (LWP 12777)]
[Thread 0x7fffaffff640 (LWP 12777) exited]
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
[... 137 lines cut ...]
#5  0x00007ffff6770265 in qErrnoWarning(char const*, ...) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#6  0x00007ffff6780451 in  () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#7  0x0000555555581087 in RtlSdrWorker::~RtlSdrWorker() ()
#8  0x00007ffff6874dfb in QObjectPrivate::deleteChildren() () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#9  0x00007ffff68774ef in QObject::~QObject() () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#10 0x00005555555b254d in RtlSdrInput::~RtlSdrInput() ()
#11 0x000055555558cca7 in MainWindow::initInputDevice(InputDeviceId const&) ()
#12 0x00007ffff686db8f in QObject::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#13 0x00007ffff7619d36 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#14 0x00007ffff6820eb8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#15 0x00007ffff68247ca in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#16 0x00007ffff6a49a97 in  () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#17 0x00007ffff5cbbd3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff5d106c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff5cb93e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#20 0x00007ffff6a4930e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/x86_64-linux-gnu/libQt6Core.so.6
#21 0x00007ffff682df3b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#22 0x00007ffff6829553 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#23 0x000055555557a03f in main ()
```
I guess it has to do when a user changes the drop-down entries too fast.

[image]

andimik, 2023-01-04:
Sometimes it's even a crash or the following line appears:

```
[...]
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
[Thread 0x7fffeaf52640 (LWP 13347) exited]
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
RTL-SDR: Worker thread not finished after timeout - this should not happen :-(
[Thread 0x7fffd2ffd640 (LWP 13354) exited]
[Thread 0x7fffd37fe640 (LWP 13353) exited]
[Thread 0x7fffd27fc640 (LWP 13355) exited]
[Thread 0x7fffd3fff640 (LWP 13352) exited]
[...]
```

KejPi, 2023-01-04:
Thanks for reporting, I can reproduce the issue. I will try to find some solution.