KejPi/AbracaDABra #45: Airspy is off, but required

andimik, 2023-01-14:
Although I have not compiled it with Airspy support, Airspy is needed anyway.

I fear, there is an `if` missing somewhere.

```
$ cmake ..
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- libdabsdr found: /home/andreas/apps/AbracaDABra/lib/linux_x86_64/libdabsdr.so
-- Checking for one of the modules 'rtlsdr'
-- Checking for one of the modules 'usb-1.0'
-- Checking for one of the modules 'faad'
-- Checking for one of the modules 'mpg123'
-- Checking for one of the modules 'portaudio-2.0'
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE  
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found WrapOpenGL: TRUE  
-- Found XKB: /usr/lib/x86_64-linux-gnu/libxkbcommon.so (found suitable version "1.4.0", minimum required is "0.5.0") 
-- Found WrapVulkanHeaders: /usr/include  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andreas/apps/AbracaDABra/build
andreas@andreas-HP-EliteDesk-800-G2-SFF:~/apps/AbracaDABra/build$ LANG=C make
[  2%] Automatic MOC and UIC for target AbracaDABra
[  2%] Built target AbracaDABra_autogen
[  5%] Generating qrc_resources.cpp
[  8%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o
In file included from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/EWIEGA46WW/../../../../gui/setupdialog.h:11,
                 from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/EWIEGA46WW/../../../../gui/mainwindow.h:18,
                 from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/EWIEGA46WW/moc_mainwindow.cpp:10,
                 from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/mocs_compilation.cpp:11:
/home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/EWIEGA46WW/../../../../gui/airspyinput.h:7:10: fatal error: libairspy/airspy.h: No such file or directory
    7 | #include <libairspy/airspy.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:129: gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:99: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

My version is:

```
$ git show
commit 08923171f745a1cb79ec7d4a6f38b060a08012a2 (HEAD -> main, origin/main, origin/HEAD)
Merge: 154eaaa 97e9da3
Author: KejPi <xkejpi@gmail.com>
Date:   Fri Jan 13 14:21:14 2023 +0100

    Merge branch 'main' of https://github.com/KejPi/AbracaDABra
```

KejPi, 2023-01-14:
Please do "cmake .." from empty folder or at least delete cmake cache. Options are cached by cmake and they are not overwritten subsequent calls.

andimik, 2023-01-14:
I did of course

closed 2023-01-14 by andimik