KejPi/AbracaDABra #46: Debug switches don't work anymore

andimik, 2023-01-20:
If I remove the comments in `audiodecoder.h` in order to let the tool write the AAC files to the current directory, this is no more working in the latest commits.

```
// debug switches
#define AUDIO_DECODER_RAW_OUT
#define AUDIO_DECODER_AAC_OUT
#define AUDIO_DECODER_MP2_OUT
```

```
$ LANG=C make
[  2%] Automatic MOC and UIC for target AbracaDABra
[  2%] Built target AbracaDABra_autogen
[  5%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o
[  7%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/main.cpp.o
[ 10%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/mainwindow.cpp.o
[ 13%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/audiodecoder.cpp.o
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp: In member function 'void AudioDecoder::processMP2(RadioControlAudioData*)':
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:512:32: error: no match for call to '(std::vector<unsigned char>) ()'
  512 |     writeMP2Output(inData->data() + 2, inData->size() - 2);
      |                    ~~~~~~~~~~~~^~
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:512:48: error: 'struct RadioControlAudioData' has no member named 'size'
  512 |     writeMP2Output(inData->data() + 2, inData->size() - 2);
      |                                                ^~~~
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp: In member function 'void AudioDecoder::processAAC(RadioControlAudioData*)':
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:789:20: error: 'aacData' was not declared in this scope
  789 |     writeAACOutput(aacData, len);
      |                    ^~~~~~~
/home/andreas/apps/AbracaDABra/gui/audiodecoder.cpp:789:29: error: 'len' was not declared in this scope
  789 |     writeAACOutput(aacData, len);
      |                             ^~~
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:367: gui/CMakeFiles/AbracaDABra.dir/audiodecoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

In the past it was working (tried with v1.2, commit 58ab081b95819e45136f8f2a598944f878c32048 )

[image]

The idea is to use the tool for writing the audio to a file when the reception is too bad for all other tools (Qt-DAB, QIRX and so on).

KejPi, 2023-01-20:
Compilation works, but not tested. Please check the functionality and close this issue if everything OK. 

If other tools do not work, you should just switch to AbracaDABra ;-)

andimik, 2023-01-20:
Yes, that [64f076b](https://github.com/KejPi/AbracaDABra/commit/64f076bb5358da1345efa6cce730a3a10fe8167d) solved the issue. The AAC contains all MOT data that were received via DAB+.

[image]

> If other tools do not work, you should just switch to AbracaDABra ;-)

That's why I need audio AAC recording ...

closed 2023-01-20 by andimik