KejPi/AbracaDABra #48: int_fast32_t instead of of int

andimik, 2023-02-05:
During compiling under Ubuntu 22.04 64bit I get the following hint:

```
[ 89%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/spiapp.cpp.o
/home/andreas/apps/AbracaDABra/gui/spiapp.cpp: In member function ‘uint32_t SPIApp::parseTag(const uint8_t*, uint8_t, int)’:
/home/andreas/apps/AbracaDABra/gui/spiapp.cpp:292:33: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘int_fast32_t’ {aka ‘long int’} [-Wformat=]
  292 |     qDebug("Tag = %2.2X, len = %d", tag, len);
      |                                ~^        ~~~
      |                                 |        |
      |                                 int      int_fast32_t {aka long int}
      |                                %ld
[ 92%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/slsview.cpp.o
```
All other files are ok. 

If I set the line to 

```
int len = *(dataPtr+1);
```

it compiles without message, but I don't know if this ok for your.

This is the diff:

[abra.diff.txt](https://github.com/KejPi/AbracaDABra/files/10610874/abra.diff.txt)

KejPi, 2023-02-05:
Yes, I know, I was thinking to remove the code completely because it is non-functional mess, I was trying to understand, how SPI works and did not remove it from the project before creating Github project. I will fix it.

andimik, 2023-02-05:
Fixed in https://github.com/KejPi/AbracaDABra/commit/a52ab7eef1c4f23d4b3143087748472f3652d707

closed 2023-02-05 by andimik