KejPi/AbracaDABra #50: File format to be taken from xml header

andimik, 2023-02-06:
https://github.com/KejPi/AbracaDABra/blob/cf7282408606f9186fa53da852e03ef135fba727/gui/setupdialog.cpp#L904

Do we really need these line(s)? If no, delete them. 

If yes, we could now read the file format from the xml header, if applicable. Currently, the console output is depending on the file extension (`*.u8`, `*.s16`)

KejPi, 2023-02-06:
There is plenty of useless messages I have added for debug purposes. Most of them could be probably removed. It probably worries only Linux users, others do not see them. I will clean it up when I have some time. Currently I am trying to implement support for translations.

andimik, 2023-02-06:
ok fine :-)

KejPi, 2023-02-20:
Messages removed, if XML header is detected, then format is taken from header. Still only uint8 or int16 LE supported.

closed 2023-02-20 by KejPi