KejPi/AbracaDABra #52: Cannot build - 'StringLiterals' is not a namespace-name

andimik, 2023-02-10:
latest f5c2cf74acff1f88a0afe31da826df953983f86b cannot be built:

```
[ 17%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/qrc_resources.cpp.o
Reaping winning child 0x560693e005e0 PID 8006 
Live child 0x560693e005e0 (gui/CMakeFiles/AbracaDABra.dir/qrc_resources.cpp.o) PID 8008 
Reaping winning child 0x560693e005e0 PID 8008 
Removing child 0x560693e005e0 PID 8008 from chain.
    Successfully remade target file 'gui/CMakeFiles/AbracaDABra.dir/qrc_resources.cpp.o'.
    Considering target file 'gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o'.
     File 'gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o' does not exist.
      Pruning file 'gui/CMakeFiles/AbracaDABra.dir/compiler_depend.ts'.
      Pruning file 'gui/CMakeFiles/AbracaDABra.dir/flags.make'.
      Considering target file '../gui/airspyinput.cpp'.
       Looking for an implicit rule for '../gui/airspyinput.cpp'.
       No implicit rule found for '../gui/airspyinput.cpp'.
       Finished prerequisites of target file '../gui/airspyinput.cpp'.
      No need to remake target '../gui/airspyinput.cpp'.
     Finished prerequisites of target file 'gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o'.
    Must remake target 'gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o'.
Putting child 0x560693dfc550 (gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o) PID 8012 on the chain.
Live child 0x560693dfc550 (gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o) PID 8012 
[ 19%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o
Reaping winning child 0x560693dfc550 PID 8012 
Live child 0x560693dfc550 (gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o) PID 8014 
Reaping winning child 0x560693dfc550 PID 8014 
Removing child 0x560693dfc550 PID 8014 from chain.
    Successfully remade target file 'gui/CMakeFiles/AbracaDABra.dir/airspyinput.cpp.o'.
    Considering target file 'gui/CMakeFiles/AbracaDABra.dir/main.cpp.o'.
     File 'gui/CMakeFiles/AbracaDABra.dir/main.cpp.o' does not exist.
      Pruning file 'gui/CMakeFiles/AbracaDABra.dir/compiler_depend.ts'.
      Pruning file 'gui/CMakeFiles/AbracaDABra.dir/flags.make'.
      Considering target file '../gui/main.cpp'.
       Looking for an implicit rule for '../gui/main.cpp'.
       No implicit rule found for '../gui/main.cpp'.
       Finished prerequisites of target file '../gui/main.cpp'.
      No need to remake target '../gui/main.cpp'.
     Finished prerequisites of target file 'gui/CMakeFiles/AbracaDABra.dir/main.cpp.o'.
    Must remake target 'gui/CMakeFiles/AbracaDABra.dir/main.cpp.o'.
Putting child 0x560693dfd460 (gui/CMakeFiles/AbracaDABra.dir/main.cpp.o) PID 8018 on the chain.
Live child 0x560693dfd460 (gui/CMakeFiles/AbracaDABra.dir/main.cpp.o) PID 8018 
[ 21%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/main.cpp.o
Reaping winning child 0x560693dfd460 PID 8018 
Live child 0x560693dfd460 (gui/CMakeFiles/AbracaDABra.dir/main.cpp.o) PID 8020 
/home/andreas/apps/AbracaDABra/gui/main.cpp:33:21: error: 'StringLiterals' is not a namespace-name
   33 | using namespace Qt::StringLiterals;
      |                     ^~~~~~~~~~~~~~
/home/andreas/apps/AbracaDABra/gui/main.cpp: In function 'int main(int, char**)':
/home/andreas/apps/AbracaDABra/gui/main.cpp:64:44: error: unable to find string literal operator 'operator""_s' with 'const char16_t [12]', 'long unsigned int' arguments
   64 |     if (translator.load(QLocale::system(), u"AbracaDABra"_s, u"_"_s, u":/i18n"_s))
      |                                            ^~~~~~~~~~~~~~~~
/home/andreas/apps/AbracaDABra/gui/main.cpp:64:62: error: unable to find string literal operator 'operator""_s' with 'const char16_t [2]', 'long unsigned int' arguments
   64 |     if (translator.load(QLocale::system(), u"AbracaDABra"_s, u"_"_s, u":/i18n"_s))
      |                                                              ^~~~~~
/home/andreas/apps/AbracaDABra/gui/main.cpp:64:70: error: unable to find string literal operator 'operator""_s' with 'const char16_t [7]', 'long unsigned int' arguments
   64 |     if (translator.load(QLocale::system(), u"AbracaDABra"_s, u"_"_s, u":/i18n"_s))
      |                                                                      ^~~~~~~~~~~
Reaping losing child 0x560693dfd460 PID 8020 
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:195: gui/CMakeFiles/AbracaDABra.dir/main.cpp.o] Error 1
Removing child 0x560693dfd460 PID 8020 from chain.
Reaping losing child 0x55e1e409ed30 PID 7999 
make[1]: *** [CMakeFiles/Makefile2:132: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
Removing child 0x55e1e409ed30 PID 7999 from chain.
Reaping losing child 0x55dde98f2d00 PID 7924 
make: *** [Makefile:136: all] Error 2
Removing child 0x55dde98f2d00 PID 7924 from chain.
```

andimik, 2023-02-10:
code has been introduced in 50610614

KejPi, 2023-02-10:
This is the documented way for loading translations. I had the same compilation problem on Mac and resolved it by adding StringLiterals namespace. It is probably missing in older version of Qt6. I was able to compile it under ArchLinux ARM yesterday. I will check that later.

andimik, 2023-02-10:
There is a missing 

```
#include <StringLiterals>
```

in main.cpp, but it still does not compile.

https://doc.qt.io/qt-6/qt-literals-stringliterals.html

andimik, 2023-02-10:
Ubuntu 22.04 LTS installs 6.2.4

Do you really want me to use 6.4?

KejPi, 2023-02-10:
StringLiteral related function have been introduced in Qt 6.4. Thre should be some solution for provious version of Qt6, I just need to find it somewhere.

andimik, 2023-02-10:
https://doc.qt.io/qt-6.2/qtliterals.html

This?

KejPi, 2023-02-10:
Yes, it seems to be what I need for < Qt6.4.

KejPi, 2023-02-10:
Could you please try if 7392075c1ffa017833fa40151d34f96d6021be3a works?

andimik, 2023-02-10:
Yes, this works, thank you!

```
[ 89%] Linking CXX executable AbracaDABra
[100%] Built target AbracaDABra
```

closed 2023-02-10 by andimik