KejPi/AbracaDABra #58: Favourite not reliable

andimik, 2023-02-27:
Sometimes a click on a favorite station will not play it, but the previous heard station.

[image]

Seems this only is in combination with a station on two muxes or channels.

I tried to find the steps how to re-produce it, but I couldn't find a way for describing that 100% accurate.

KejPi, 2023-02-27:
I do not have this experience on Mac/Windows with the ensembles I have on my antenna. Waiting for steps to reproduce.

andimik, 2023-02-27:
I had this behaviour 4 times up to now. But not 100% accurate steps. Selecting favorite and non-favorite and toggling.

KejPi, 2023-02-27:
Is it really only favorite - non favorite toggling? Have you tried 2 favorites or 2 non-favorites? There should be no difference between favorite and others, it is only different way to display service list item.

andimik, 2023-02-27:
Bingo, also non-favorite are affected.

Was toggling between 2 non-favorite stations

[image]

andimik, 2023-02-27:
Not related to Austrian muxes ...

[image]

Hint: I'm clicking on the entry in the middle, not on the right.

KejPi, 2023-02-27:
Interesting, I am not able to reproduce it on Mac. When this happens - is it the service displayed on top playing it does the service selected in the service list play? And can you confirm the problem with a service that is coming from 1 ensemble? In your case both services are coming from 2 ensembles.

andimik, 2023-02-27:
Another attempt. Delete all entries. Only lock both Slovenian muxes R2West and R2East.

Radio Antena and Bob are in both muxes.

Switching between Antena (on 2 muxes) an Aktual Kum (only on 1 mux). Sometimes pressing the toggling muxes icon.

[image]

You will hear the sound of the station displayed on the left (in this case Antena)

BUT the toggling function will NOT work here as the marked (!) service (in this case Aktual Kum) does NOT have a 2nd.

I guess the second (!!) toggling produces this error.

andimik, 2023-02-27:
This is a video:

[Bildschirmaufzeichnung vom 27.02.2023, 22:41:46.webm](https://user-images.githubusercontent.com/24510556/221692532-ba76dcc2-68b5-4eae-a9b2-12dfa7e936a5.webm)

KejPi, 2023-02-28:
Thank for the video, it helped me to reproduce the behavior 👍 Just to be clear - was the issue always related to switching of the service source (button on top) or simple clicking in the service list did that too?

andimik, 2023-02-28:
Yes, I already thought about that. If you compare how long locking is taking, then you understand it.

KejPi, 2023-02-28:
Steps to reproduce are actually very simple. Just 2 ensembles are needed:

1. Select any service in ensA
2. Wait until it is tuned
3. Select service1 in ensB
4. Immediately when UI allows you select another service2 in ensB 

As a result, service1 is playing but service2 is selected in service list. 

This way I can reproduce the issue in 100% cases, I hope it is the same issue that you have reported originally.

andimik, 2023-02-28:
ok, seems to be fixed with that commit, therefore, I'll close it.

closed 2023-02-28 by andimik