KejPi/AbracaDABra #60: "Removing from ens" message from wrong ensemble

andimik, 2023-02-28:
I wonder why I got the following message on the console:

```
	Removing from ens E0A101: Radio Sraka SID = 0xE4943D, SCIdS = 0
	Removing from ens E0A101: MMR SID = 0xE49456, SCIdS = 0
	Removing from ens E0A101: Radio Maxi SID = 0xE49428, SCIdS = 0
	Removing from ens E0A101: Radio Maribor SID = 0xE49426, SCIdS = 0
	Removing from ens E0A101: Radio Krka SID = 0xE49423, SCIdS = 0
	Removing from ens E0A101: Radio Celje SID = 0xE49413, SCIdS = 0
	Removing from ens E0A101: Radio Ptuj SID = 0xE49434, SCIdS = 0
	Removing from ens E0A101: Antena Celje SID = 0xE4965E, SCIdS = 0
	Removing from ens E0A101: Radio Murski Val SID = 0xE4942A, SCIdS = 0
	Removing from ens E0A101: Koroski Radio SID = 0xE49452, SCIdS = 0
	Removing from ens E0A101: Stajerski Val SID = 0xE4943F, SCIdS = 0
	Removing from ens E0A101: Radio Aktual Kum SID = 0xE49443, SCIdS = 0
	Removing from ens E0A101: Radio Velenje SID = 0xE49448, SCIdS = 0
```

as these services are all on Slovenian

```
Ensemble:
0x9300 SLO DAB+ R2 E [ SLO R2 E ] ECC = 0xE4, UTC 60 min, INT = 1, alarm announcements = 0
```

and not on Austrian Bundesmux

```
Ensemble:
0xA101 DAB+ Austria [ DAB+ A ] ECC = 0xE0, UTC 60 min, INT = 1, alarm announcements = 0
```

andimik, 2023-02-28:
I could not reproduce so far.

KejPi, 2023-03-02:
What device do you use?

andimik, 2023-03-02:
RTL Stick.

KejPi, 2023-03-02:
Connected directly or over TCP-IP?

andimik, 2023-03-02:
Direct

KejPi, 2023-03-02:
Hmm, interesting, I cannot reproduce this behavior. Could you please share more details? 

I have understood that you switch 2 services each from different ensemble and during switching you see these messages. Is any of that 2 services in both ensembles? Is service list broken or the problem after switch? 

Maybe sharing video would help again.

andimik, 2023-03-02:
A video won't help you. I just saw this message in the console. The list of services was looking good.

I know that it has to do with weak signals, but I wonder why AbracaDABra mixed up the ensembles...

KejPi, 2023-03-03:
Yes, mixing of the IQ data was my idea too, that is why I have asked about device. It should not happen when RTLSDR is used directly because I always stop the stream and then I start it again to avoid mixing of data. This is unfortunately making tune slower but I have not found better way to avoid mixing of the stream. On the other hand it should be safe unless the underlying layer is leaving old data in buffers and sending them on new start. I will try to reproduce the issue on Linux, on Mac it does not happen. I was trying to switch 2 services from 2 different ensembles many times and it did not happen. I assume you compiled the latest version from repository on Linux, you are not using AppImage, correct?

andimik, 2023-03-03:
I am always referring to the latest commit, which I am compiling myself.

KejPi, 2023-03-03:
This is what I have in terminal output under Ubuntu 22.04 when switching between Classic Praha and Expres FM. Both are in both ensembles, I ma switching between Classic @ 7C and Express @ 7A. There is obviously no problem.

```
Allocating 15 zero-copy buffers
	Service: DAB Test HQ SID = 0xE22F75, SCIdS = 0
	Service: Classic Praha SID = 0xE2244A, SCIdS = 0
	Service: DAB Test LQ SID = 0xE22F77, SCIdS = 0
	Service: 'Rádio SÁZAVA SID = 0xE22FBA, SCIdS = 0
	Service: Schwarzwaldradio SID = 0xE0100D, SCIdS = 0
	Service: Schlager Radio SID = 0xE01735, SCIdS = 0
	Service: Kiss SID = 0xE22FA0, SCIdS = 0
	Service: Country Radio SID = 0xE22F37, SCIdS = 0
	Service: COLOR MusicRadio SID = 0xE22442, SCIdS = 0
	Service: Spin Radio SID = 0xE22F43, SCIdS = 0
	Service: HEY Radio SID = 0xE224F8, SCIdS = 0
	Service: Expres FM SID = 0xE2203B, SCIdS = 0
AAC-LC 48 kHz stereo
Output SR = 48000, channels = 2
Allocating 15 zero-copy buffers
	Service: RADIO BONTON SID = 0xE22430, SCIdS = 0
	Service: Radio BEAT SID = 0xE22F31, SCIdS = 0
	Service: BLANIK DAB SID = 0xE22F3C, SCIdS = 0
	Service: Expres FM SID = 0xE2203B, SCIdS = 0
	Service: HITRADIO CITY SID = 0xE22F32, SCIdS = 0
	Service: Classic Praha SID = 0xE2244A, SCIdS = 0
	Service: RadioZET SID = 0xE2210E, SCIdS = 0
	Service: PIGY RADIO SID = 0xE22FB7, SCIdS = 0
	Service: EVROPA 2 SID = 0xE22204, SCIdS = 0
	Service: FREKVENCE 1 SID = 0xE22205, SCIdS = 0
	Service: Cesky Impuls SID = 0xE22FB4, SCIdS = 0
	Service: Radio Dechovka SID = 0xE224B2, SCIdS = 0
	Service: ROCKZONE SID = 0xE22445, SCIdS = 0
	Service: Radio Impuls SID = 0xE22203, SCIdS = 0
HE-AAC v2 48 kHz stereo
Output SR = 48000, channels = 2
Allocating 15 zero-copy buffers
	Service: 'Rádio SÁZAVA SID = 0xE22FBA, SCIdS = 0
	Service: Schwarzwaldradio SID = 0xE0100D, SCIdS = 0
	Service: Schlager Radio SID = 0xE01735, SCIdS = 0
	Service: Kiss SID = 0xE22FA0, SCIdS = 0
	Service: Country Radio SID = 0xE22F37, SCIdS = 0
	Service: COLOR MusicRadio SID = 0xE22442, SCIdS = 0
	Service: Spin Radio SID = 0xE22F43, SCIdS = 0
	Service: HEY Radio SID = 0xE224F8, SCIdS = 0
	Service: Expres FM SID = 0xE2203B, SCIdS = 0
	Service: DAB Test HQ SID = 0xE22F75, SCIdS = 0
	Service: Classic Praha SID = 0xE2244A, SCIdS = 0
	Service: DAB Test LQ SID = 0xE22F77, SCIdS = 0
AAC-LC 48 kHz stereo
Output SR = 48000, channels = 2
```

andimik, 2023-03-10:
I am using the version from Ubuntu.

andimik, 2023-04-24:
Just had the same message with Airspy, so this is not related to the dongle.

```
RadioControlEvent::SERVICE_STOP error 2
	Service: PRVI SID = 0xE49201, SCIdS = 0
	Service: Radio 1 DAB SID = 0xE49442, SCIdS = 0
	Service: Ekspres Slo SID = 0xE49462, SCIdS = 0
	Service: Radio 1 80-a SID = 0xE49459, SCIdS = 0
	Service: Radio Salomon SID = 0xE49463, SCIdS = 0
	Service: Rock Radio Slo SID = 0xE49516, SCIdS = 0
	Service: Center Slo SID = 0xE49414, SCIdS = 0
	Service: Radio Ognjisce SID = 0xE4932F, SCIdS = 0
	Service: Radio Si SID = 0xE4933A, SCIdS = 0
	Service: VAL 202 SID = 0xE49202, SCIdS = 0
	Service: ARS SID = 0xE49203, SCIdS = 0
	Service: Radio Veseljak SID = 0xE49449, SCIdS = 0
	Service: Radio Aktual SID = 0xE4941C, SCIdS = 0
	Service: Radio City SID = 0xE49415, SCIdS = 0
	Service: Radio Student SID = 0xE49440, SCIdS = 0
	Service: Radio Net FM SID = 0xE4942B, SCIdS = 0
	Removing from ens E49200: 'WELLE 1 SID = 0xE0AD58, SCIdS = 0
	Removing from ens E49200: *STEPHANSDOM* SID = 0xE0AC52, SCIdS = 0
	Removing from ens E49200: Antenne Österrei SID = 0xE0AD57, SCIdS = 0
	Removing from ens E49200: KLASSIK RADIO SID = 0xE0AD53, SCIdS = 0
	Removing from ens E49200: Rock Antenne SID = 0xE0AD27, SCIdS = 0
	Removing from ens E49200: arabella HOT SID = 0xE0AD59, SCIdS = 0
	Removing from ens E49200: Radio Maria SID = 0xE0A3DD, SCIdS = 0
	Removing from ens E49200: jö.live SID = 0xE0AD56, SCIdS = 0
	Removing from ens E49200: ERF Süd SID = 0xE0AD24, SCIdS = 0
	Removing from ens E49200: -AUSTRIA SID = 0xE0A3E0, SCIdS = 0
	Removing from ens E49200: arabella RELAX SID = 0xE0AD54, SCIdS = 0
	Removing from ens E49200: #Technikum ONE SID = 0xE0AD2A, SCIdS = 0
	Removing from ens E49200: Mein Kinderradio SID = 0xE0AD55, SCIdS = 0
	Removing from ens E49200: ENERGY SID = 0xE0AC51, SCIdS = 0
	Removing from ens E49200: * 88.6 * SID = 0xE0AC47, SCIdS = 0
	Removing from ens E49200: Radio Flamingo SID = 0xE0AD5A, SCIdS = 0
	Service: PRVI SID = 0xE49201, SCIdS = 0
	Service: Radio 1 DAB SID = 0xE49442, SCIdS = 0
	Service: Ekspres Slo SID = 0xE49462, SCIdS = 0
	Service: Radio 1 80-a SID = 0xE49459, SCIdS = 0
	Service: Radio Salomon SID = 0xE49463, SCIdS = 0
	Service: Rock Radio Slo SID = 0xE49516, SCIdS = 0
	Service: Center Slo SID = 0xE49414, SCIdS = 0
	Service: Radio Ognjisce SID = 0xE4932F, SCIdS = 0
	Service: Radio Si SID = 0xE4933A, SCIdS = 0
	Service: VAL 202 SID = 0xE49202, SCIdS = 0
	Service: ARS SID = 0xE49203, SCIdS = 0
	Service: Radio Veseljak SID = 0xE49449, SCIdS = 0
	Service: Radio Aktual SID = 0xE4941C, SCIdS = 0
	Service: Radio City SID = 0xE49415, SCIdS = 0
	Service: Radio Student SID = 0xE49440, SCIdS = 0
	Service: Radio Net FM SID = 0xE4942B, SCIdS = 0
	Removing from ens E49200: 'WELLE 1 SID = 0xE0AD58, SCIdS = 0
	Removing from ens E49200: *STEPHANSDOM* SID = 0xE0AC52, SCIdS = 0
	Removing from ens E49200: Antenne Österrei SID = 0xE0AD57, SCIdS = 0
	Removing from ens E49200: KLASSIK RADIO SID = 0xE0AD53, SCIdS = 0
	Removing from ens E49200: Rock Antenne SID = 0xE0AD27, SCIdS = 0
	Removing from ens E49200: arabella HOT SID = 0xE0AD59, SCIdS = 0
	Removing from ens E49200: Radio Maria SID = 0xE0A3DD, SCIdS = 0
	Removing from ens E49200: jö.live SID = 0xE0AD56, SCIdS = 0
	Removing from ens E49200: ERF Süd SID = 0xE0AD24, SCIdS = 0
	Removing from ens E49200: -AUSTRIA SID = 0xE0A3E0, SCIdS = 0
	Removing from ens E49200: arabella RELAX SID = 0xE0AD54, SCIdS = 0
	Removing from ens E49200: #Technikum ONE SID = 0xE0AD2A, SCIdS = 0
	Removing from ens E49200: Mein Kinderradio SID = 0xE0AD55, SCIdS = 0
	Removing from ens E49200: ENERGY SID = 0xE0AC51, SCIdS = 0
	Removing from ens E49200: * 88.6 * SID = 0xE0AC47, SCIdS = 0
	Removing from ens E49200: Radio Flamingo SID = 0xE0AD5A, SCIdS = 0
```
9200 is Slovenian DAB, not Austria ...

Could you add a debug message when tuning to a block (like `tuning to block xxx`) and service (`playing Radio xxx`).

andimik, 2023-04-24:
This is an interesting screenshot. Found 2 ensembles (6A and 8A, both `DAB+ Austria`), but only shows one line this time. 

Seems this may cause the above mentioned issue?

[image]

Normally, both blocks are listed here.

[image]

KejPi, 2023-04-26:
I have added the messages in e6598e5e77828878de4767c087887f1ed94aac11 
Do you have any test case to reproduce the issue or it is more like random?

andimik, 2023-04-26:
I fear it's more like random.

andimik, 2024-10-18:
I'll close here, not checked in the past months.

closed 2024-10-18 by andimik

KejPi, 2026-02-17:
According to the log, you tuned from 10C to 5A, what is in 10C? And where are the removed services coming from? Are they coming from single ensemble or from more ennsembles? Please share ensemble info of all involved ensebles - the one on 5A, the ensemble on 10C and all ensembles that contain removed services.