KejPi/AbracaDABra #61: MUX 12B

WIHDIA, 2023-03-09:
Hi,

I cannot get anything on strong local 12B DAB Bruxelles. Strong  local 12A and nearby 12C NPO are perfect. I can get 12B on decoders like QT_DAB and QIRX.

73

Herman Wijnants

[My logblog](https://www.wijnants.info/fmdx/)

WIHDIA, 2023-03-09:
Hi,

I tried the manual gain control before posting this issue and I cannot get 12B to work using it.

A raw file recorded with QIRX on the other hand works well on AbracaDABra.

I'm sure it's common Mode 1 as this mux plays well on QT/QIRX and even my DAB radio set in the living room.

73

Herman Wijnants

[My logblog](https://www.wijnants.info/fmdx/)

[BELGIUM editor fmlist.org](https://www.fmlist.org/)

andimik, 2023-03-09:
And what about rtl_tcp and try some bandwidth settings?

WIHDIA, 2023-03-09:
Hi,

I also tried Airspy and it works a bit but not great. It's a hassle to get the manual setting right and these settings won't work after a few seconds. RTL-SDR works great but not on 12B, alas.  There's a LNA connected to the DAB antennes and RTL cannot switch it off in AbracaDABra, but it can do so with Airspy.

I'm not sure how to geet rtl_tcp running.

73

Herman Wijnants

[My logblog](https://www.wijnants.info/fmdx/)

[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-09:
This is very interesting issue but it will be difficult to debug. 

My understanding of the issue is following:
- recording from QIRX works in AbracaDABra
- it is not possible to make it working using RTL SDR (both SW and manual gain), it only somehow works with Airspy

Conclusions:
- DAB signal is OK because recording from other tool works
- the problem seems to be AGC or BW. Since manual gain control does not work it seems to be problem with bandwidth. Also because there are signals in adjacent channels I would say it is bandwidth problem. 

Questions/Hints:
1. Have you tried to play recording from AbracaDABra in QIRX? Does it work?
2. Could you please try to reduce bandwidth by editing INI file - see this https://github.com/KejPi/AbracaDABra#expert-settings I would recommend to try `bandwidth=1530000` for RTLSDR device. 
3. You can run rtl_tcp from installation folder of QIRX and connect to that server by AbracaDABra (RTL TCP device). Does it work with SW gain and/or with device gain?
4. Could you please share both recording from other tool and from AbracaDABra from RTLSDR device with SW gain?
5. It would be great if you could share full bandwidth recording using airspy_rx tool (with 4096kHz sample rate) so that I can analyse what is going on.

WIHDIA, 2023-03-09:
Hi,

See replies in **bold**.

> This is very interesting issue but it will be difficult to debug.
>
>     * recording from QIRX works in AbracaDABra
>     * it is not possible to make it working using RTL SDR (both SW and manual gain), [...]

**Correct!**
 
>     * the problem seems to be AGC or BW. Since manual gain control does not work it seems to be problem with bandwidth. Also because there are signals in adjacent channels I would say it is bandwidth problem.
> 
**Yes, that's plausible. I remember QT having the same problems way back.**
> Questions/Hints:
> 
>     1. Have you tried to play recording from AbracaDABra in QIRX? Does it work?

**Works perfectly well in QT and QIRX.**
> 
>     2. Could you please try to reduce bandwidth by editing INI file - see this https://github.com/KejPi/AbracaDABra#expert-settings I would recommend to try `bandwidth=1530000` for RTLSDR device.

**I should have studied the ini file. I only had a quick glance.
This is what I changed:**

**bandwidth=0
bias-T=false**

**TO**

**bandwidth=1530000
bias-T=true**

**Changing the bandwidth and NOT the bias-T won't work as well in my case with LNA near the antennas.
The problem seems to be fixed.**

> 
>     3. You can run rtl_tcp from installation folder of QIRX and connect to that server by AbracaDABra (RTL TCP device). Does it work with SW gain and/or with device gain?

**I'll have a look later on today, but still necessary after previous remarks?**

>     4. Could you please share both recording from other tool and from AbracaDABra from RTLSDR device with SW gain?

**Still necessary? But I'll gladly do so. Just let me know. Share via PM?**

> 
>     5. It would be great if you could share full bandwidth recording using airspy_rx tool (with 4096kHz sample rate) so that I can analyse what is going on.

**Same as above.**

**BTW AbracaDABra doesn't show TII codes or am I missing something. I must admit I mostly use QT or QIRX for DX and listening and AbracaDABra for occasional listening. So AbracaDABra is still a but new to me :-)**

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

WIHDIA, 2023-03-09:
Hi,

1530000 still gave some problems. I tried a few lower and highter settings and now 1400000 seems to work pretty well on all muxes, including 12B.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-09:
AbracaDABra does not decode TII, application is mainly focused on radio listening. 
Yes, I am still very interested to get recordings for analysis. It would help me to improve the application because typically AbracaDABra does not have reception problems, from sensitivity point of view the app is typically better than others or at least comparable and this is the first time anybody reports reception problems.

WIHDIA, 2023-03-09:
Hi

Recordings can be found here:

https://wijnants.stackstorage.com/s/lfTyAzd5b4fcGaF3   (link works for 2 days only).

2023_03_09_12_25_44_2048000_fHz225648000.raw  made with QIRX
2023-03-09_145835_12C.uff AbracaDABra + RTL-SDR
2023-03-09_193202_12B.uff AbracaDABra + RTL-SDR
2023-03-09_194656_12B.uff AbracaDABra + rtl_tcp
2023-03-09_195644_12B.uff AbracaDABra + Airspy R2

After the AbracaDABra +rtl_tcp recording RTL-SDR wouldn't work with AbracaDABra. A pc reboot fixed it.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

andimik, 2023-03-10:
> After the AbracaDABra +rtl_tcp recording RTL-SDR wouldn't work with AbracaDABra. A pc reboot fixed it.

Have you closed resp. killed rtl_tcp?

WIHDIA, 2023-03-10:
Hi,

I checked and the process wasn't running at all. Odd. Perhaps I overlooked it. Anyway, it's not a big deal to me.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-10:
Thanks for sharing the files!
I have seen the problem with some rtl_tcp binaries under Windows too. The solution is to unplug the USB stick and plug it again. I am not sure what is wrong it seems that the process does not close the device properly, but maybe AbracaDABra does not do correct init. I will create separate ticket for that so that I do not forget to look at this problem.

WIHDIA, 2023-03-10:
> Thanks for sharing the files! I have seen the problem with some rtl_tcp binaries under Windows too. The solution is to unplug the USB stick and plug it again. I am not sure what is wrong it seems [...]

Thanks, KejPi!

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-14:
Selectivity filter is implemented is DABSDR 2.1.0 released few minutes ago. This should fix selectivity issues reported in this ticket as well as on forum (RTLSDR default bandwidth was already set to 1530kHz before). All shared raw files play fine with exception of Airspy. I did some analysis of Airspy file and it seems to be a problem of suppression caused by AGC driven by strong adjacent channel. Unfortunately Airspy does not have API for BW selection thus it is very difficult to fix this problem. It might help to switch to 3000kHz sample rate by changing this configuration in INI file:

`preferSampleRate4096kHz=true` ==> `preferSampleRate4096kHz=false`

PS: AbracaDABra 2.1 will be released soon, but meanwhile Windows users can replace DLL in installation folder of AbracaDABra 2.0

WIHDIA, 2023-03-14:
> Selectivity filter is implemented in DABSDR 2.1.0 released few minutes ago. This should fix selectivity issues reported in this ticket as well as on forum (RTLSDR default bandwidth was already set to 1530kHz before). All shared raw files play fine with exception of Airspy recording. I did some analysis of that file and it seems to be a problem of suppression caused by AGC driven by strong adjacent channel. Unfortunately Airspy does not have API for BW selection thus it is very difficult to fix this problem. It might help to switch to 3000kHz sample rate by changing this configuration in INI file:
> 
> `preferSampleRate4096kHz=true` ==> `preferSampleRate4096kHz=false`
> 
> PS: AbracaDABra 2.1 will be released soon, but meanwhile Windows users can replace DLL in installation folder of AbracaDABra 2.0

2.1 is working nicely using Win10.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-14:
Perfect! I am closing this issue, in case of any problems please do not hesitate to open new issue.

closed 2023-03-14 by KejPi