KejPi/AbracaDABra #63: AbracaDABra does not decode some ensembles working with DAB Player

KejPi, 2023-03-10:
Reported on forum here:
https://www.rundfunkforum.de/viewtopic.php?p=1703008#p1703008

Symptoms: 
- SNR is indicated lower than other tool
- not able to decode while other tool works fine
- recording shared
- raw files from AbracaDABra work in other tool

rf-oldie, 2023-03-10:
I do not know how Abra starts synchronizing and detects the null symbol. Do you take the total amplitude of the incoming IQ data and search for a minimum? 
If yes, a digital bandwidth filtering of the samples could be useful to remove neigbour channel components.

KejPi, 2023-03-10:
Correct, I plan to set 1530kHz as default on frontend and then I plan to do filtering on software. It will take some time to implement.

KejPi, 2023-03-14:
AbracaDABra 2.1 should fix selectivity issues, closing.

closed 2023-03-14 by KejPi