KejPi/AbracaDABra #64: 5A 5D DAB+ Vlaanderen2 crashes AbracaDABra

WIHDIA, 2023-03-10:
Hi,

AbracaDABra crashes when tuning to 5A and 5D DAB+ Vlaanderen2. No  problem on 5B or 6A.

mp4 of the crash: https://mega.nz/file/gJtxzITR#jMooWiycKJJ-tDpVA9atNmSDB0Ewo9eBvwFfhlOeP5s

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
BELGIUM editor [fmlist.org](https://www.fmlist.org)

andimik, 2023-03-10:
Herman, very interesting. 

From its content (according to FMLIST) I could not see any blocking point.

What about deleting the `*.ini` file and let AbracaDABra just search in 5A and 5D and play those?

And can you load a raw file (recorded with Qirx)? Does it crash as well?

KejPi, 2023-03-10:
Thanks for reporting! I thought andimik is the worst case user :-)))
Is it any service from this ensemble causing application to crash? What happens if you just select the channel from channel selector without selecting service?
Could you please share recording of the ensemble from other application? Is the recoding also causing crash?

KejPi, 2023-03-10:
@andimik  you are faster than me as usual ;-)

WIHDIA, 2023-03-10:
> Herman, very interesting.
> 
> From its content (according to FMLIST) I could not see any blocking point.
> 
> What about deleting the `*.ini` file and let AbracaDABra just search in 5A and 5D and play those?

**I did so. Tried it a a user and as admin on my pc. Rebooted. Deleted ini.....**
> 
> And can you load a raw file (recorded with Qirx)? Does it crash as well?
**It doesn't even load.**

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
BELGIUM editor [fmlist.org](https://www.fmlist.org/)

WIHDIA, 2023-03-10:
> Thanks for reporting! I thought andimik is the worst case user :-))) Is it any service from this ensemble causing application to crash? What happens if you just select the channel from channel [...]

It crashes on any service from 5A and 5D. 

I made a recording with QIRX and cannot even play it in AbracaDABra: it crashes AbracaDABra!

Recording here: https://wijnants.stackstorage.com/s/lfTyAzd5b4fcGaF3   File: 2023_03_10_20_22_55_2048000_fHz180064000.raw
(link expires in 2 days).

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
BELGIUM editor [fmlist.org](https://www.fmlist.org/)

andimik, 2023-03-10:
This is the backtrace of gdb under Ubuntu (it also crashes there).

```
	Service: Q-Foute Radio SID = 0xE0639D, SCIdS = 0
	Service: Joe 80's & 90's SID = 0xE06298, SCIdS = 0
	Service: Joe Easy SID = 0xE0639B, SCIdS = 0
	Service: POTZ-MusicIsLove SID = 0xE062A0, SCIdS = 0
	Service: CROOZE SID = 0xE063A4, SCIdS = 0
	Service: Business AM SID = 0xE063A5, SCIdS = 0
	Service: RTBF MIX SID = 0xE0639A, SCIdS = 0
	Service: Willy Class X SID = 0xE063A7, SCIdS = 0
	Service: N R J SID = 0xE06309, SCIdS = 0

Thread 12 "radioControlThr" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd17f9640 (LWP 71457)]
0x00005555555b4382 in RadioControl::ensembleConfigurationString() const ()
(gdb) bt
#0  0x00005555555b4382 in RadioControl::ensembleConfigurationString() const ()
#1  0x00005555555b6f88 in RadioControl::ensembleConfigurationDispatch() ()
#2  0x00007ffff69ac023 in  () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#3  0x00007ffff69ba4be in QTimer::timeout(QTimer::QPrivateSignal) ()
    at /lib/x86_64-linux-gnu/libQt6Core.so.6
#4  0x00007ffff69a062f in QObject::event(QEvent*) ()
    at /lib/x86_64-linux-gnu/libQt6Core.so.6
#5  0x00007ffff777fd36 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#6  0x00007ffff6953a58 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#7  0x00007ffff6abc203 in QTimerInfoList::activateTimers() ()
    at /lib/x86_64-linux-gnu/libQt6Core.so.6
#8  0x00007ffff6b7bbfc in  () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#9  0x00007ffff5d4ad3b in g_main_context_dispatch ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007ffff5d9f6c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007ffff5d483e3 in g_main_context_iteration ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007ffff6b7beae in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#13 0x00007ffff6960adb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
--Type <RET> for more, q to quit, c to continue without paging--
#14 0x00007ffff6a4dfdc in QThread::exec() () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#15 0x00007ffff6abdfcf in  () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#16 0x00007ffff6094b43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#17 0x00007ffff6126a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 
```
[image]

[image]

andimik, 2023-03-10:
Hint: I also tried to write an eti and then generated a clean raw with odr-dabmod. This will crash, too.

[image]

Petr: one service starts with blanks which is quite unusal.

`FICDecoder: SId 0x6309: programme service label '  N R J' ('  N R J')`

KejPi, 2023-03-10:
Space should not harm but thanks for pointing that out. I will look at this problem tomorrow, it should not be difficult to find the rootcause.

andimik, 2023-03-10:
In a hex editor I've replaced `  N R J` by `andimik`, which will not crash (but the ETI structure will be wrong then).

[image]

[image]

andimik, 2023-03-10:
[bel_test_andimik.eti.zip](https://github.com/KejPi/AbracaDABra/files/10945970/bel_test_andimik.eti.zip)

Try this file (just hex replacement of `  N R J`). Note: The structure is NOT valid DAB+!

WIHDIA, 2023-03-10:
> [bel_test_andimik.eti.zip](https://github.com/KejPi/AbracaDABra/files/10945970/bel_test_andimik.eti.zip)
> 
> Try this file (just hex replacement of ` N R J`). Note: The structure is NOT valid DAB+!

**Makes AbracaDABra crash too. OK with QIRX.**

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
BELGIUM editor [fmlist.org](https://www.fmlist.org/)

andimik, 2023-03-11:
> Makes AbracaDABra crash too.

Strange, not here. There is no crash with that file under Ubuntu. But let me generate another file (selfmade DAB stream) with leading spaces.

KejPi, 2023-03-11:
I have analysed the recording and the problem is that the ensemble is not compliant with ETSI EN 300 401 V2.1.1, it is probably version 1.4. I have to apply fix in the library probably. The space at the beginning of the service name is not problem, AbracaDABra anyway ignores leading and trailing spaces.

KejPi, 2023-03-11:
Library updated to 2.0.2 for all supported platforms.

Windows users: it is possible to replace DLL in installation folder by version 2.0.2 committed today. 

Closing issue.

closed 2023-03-11 by KejPi