KejPi/AbracaDABra #67: libdabsdr.dll not compatible with Windows 10?

WIHDIA, 2023-03-11:
Hi,

I've just replaced libdabsdr.dll with the new one, published 2 hours ago. AbracaDABra won't start, saying this new dll is not compatible.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-11:
This is strange, it works on my test PC. I removed DLL distributed with 2.0 and copied DLL from GitHub and it works:

[image]

Did you copy libdabsdr.dll and not libdabsdr.dll.a? And what Windows explorer shows when you right-click on DLL and display Properties?

[image]

WIHDIA, 2023-03-11:
Hi,

When running the exe:
[image]

I do use libdabsdr.dll Not dll.a.

Properties:

[image]

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-11:
The file seems to be corrupted, even the size is wrong, try downloading it again.

WIHDIA, 2023-03-11:
Hi,

Properties now (size still 204 kB as at first download):

[image]

BUT exe still not running and giving

[image]

Perhaps you could send me the dll?  herman@wijnants.info

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

WIHDIA, 2023-03-11:
Hi,

It's fixed! I rightclicked and downloaded the dll as shown here:

[image]

I just should have opened the link and then downloaded:

[image]

The wrong download (204 kB) was also called libdabsdr.dll .

A bit confusing not using github very often.
Anyway 5A and 5D work well now.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

closed 2023-03-11 by KejPi