KejPi/AbracaDABra #69: 2.1 scanning crash

WIHDIA, 2023-03-14:
Hi,

I unzipped 2.1 in the AbracaDABra folder and it worked perfectly well. Then I deleted the ini file to have AbracaDABra rebuild it and start by scanning for muxes. This didn't work and the scanning crashed AbracaDABra after having scanned a few muxes.
Solution: restore my back-upped original ini or start a scan and stop after a few muxes. Then continue building the ini by manually tuning to all muxes one by one.

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

KejPi, 2023-03-14:
You do not have to delete INI to do the rescan - just go to menu - Clear service list. Nevertheless it should not crash, I will try to reproduce this problem on Windows, thanks for reporting.

WIHDIA, 2023-03-14:
> You do not have to delete INI to do the rescan - just go to menu - Clear service list. Nevertheless it should not crash, I will try to reproduce this problem on Windows, thanks for reporting.

I know :-)  Just trying to break it :-)

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

WIHDIA, 2023-03-14:
2.1 also crashing when doing a band scan, so not deleting the ini file! Solution: delete the libdabsdr.dll and replace it with the previous one!

73

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
[BELGIUM editor fmlist.org](https://www.fmlist.org/)

andimik, 2023-03-14:
Yes, confirmed. Just had a crash in manual scan. But tried to reproduce it with `gdb` and a video but did not see it again.

In my case I have a very weak signal in 8A of the same mux which I have locally on 6A. So, it's the same content.

andimik, 2023-03-14:
After more than 10 trials, this is a video (crash in minute 1:05) and `gdb` output:
[Bildschirmaufzeichnung vom 15.03.2023, 00:04:38.webm](https://user-images.githubusercontent.com/24510556/225162312-e16d3f80-8c24-48e2-91db-1eb0697859e5.webm)

```
Allocating 15 zero-copy buffers
	Service: Radio Maria SID = 0xE0A3DD, SCIdS = 0
	Service: ENERGY SID = 0xE0AC51, SCIdS = 0
	Service: 'WELLE 1 SID = 0xE0AD58, SCIdS = 0
	Service: *STEPHANSDOM* SID = 0xE0AC52, SCIdS = 0
	Service: Mein Kinderradio SID = 0xE0AD55, SCIdS = 0
	Service: ERF Süd SID = 0xE0AD24, SCIdS = 0
	Service: #Technikum ONE SID = 0xE0AD2A, SCIdS = 0
	Service: Rock Antenne SID = 0xE0AD27, SCIdS = 0
	Service: jö.live SID = 0xE0AD56, SCIdS = 0
	Service: arabella RELAX SID = 0xE0AD54, SCIdS = 0
	Service: KLASSIK RADIO SID = 0xE0AD53, SCIdS = 0
	Service: * 88.6 * SID = 0xE0AC47, SCIdS = 0
	Service: -AUSTRIA SID = 0xE0A3E0, SCIdS = 0
	Service: Radio Flamingo SID = 0xE0AD5A, SCIdS = 0
	Service: Antenne Österrei SID = 0xE0AD57, SCIdS = 0
	Service: arabella HOT SID = 0xE0AD59, SCIdS = 0
[Thread 0x7fffa4fff640 (LWP 8697) exited]
[New Thread 0x7fffa4fff640 (LWP 8723)]
Allocating 15 zero-copy buffers
*** stack smashing detected ***: terminated

Thread 13 "dabsdr" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffd0ff8640 (LWP 8509)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=4664814603921077312) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: Datei oder Verzeichnis nicht gefunden.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=4664814603921077312) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=4664814603921077312) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=4664814603921077312, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff6042476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff60287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff60896f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff61db943 "*** %s ***: terminated\n")
    at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff613676a in __GI___fortify_fail (msg=msg@entry=0x7ffff61db92b "stack smashing detected") at ./debug/fortify_fail.c:26
#7  0x00007ffff6136736 in __stack_chk_fail () at ./debug/stack_chk_fail.c:24
#8  0x00007ffff7f9d373 in  () at /home/andreas/apps/AbracaDABra/lib/linux_x86_64/libdabsdr.so.2
#9  0x00007ffff7f9b33e in  () at /home/andreas/apps/AbracaDABra/lib/linux_x86_64/libdabsdr.so.2
#10 0x4192ea38410266d4 in  ()
#11 0x41d6fa0b41a2aaf2 in  ()
#12 0xc0dbe89ec1c68a84 in  ()
#13 0xc1bfff5ac092e6f6 in  ()
#14 0x40e795f041f3f252 in  ()
#15 0x41fab3ac41880265 in  ()
#16 0x40b14ed34164c0c3 in  ()
#17 0xc105bf76c19e7118 in  ()
#18 0x41a4131ec1b716cb in  ()
#19 0x419448c241c03682 in  ()
#20 0xbfa88d54c1829678 in  ()
#21 0x40863b3fc18db4e7 in  ()
#22 0x416a335442a836c9 in  ()
#23 0x421381da4298727e in  ()
#24 0x4227dada40a37a2e in  ()
#25 0x41bbf3be416ec8cd in  ()
#26 0x41d356cb4196c8c5 in  ()
#27 0x418d5891c20fb335 in  ()
#28 0xbfa398a0c285d6c0 in  ()
#29 0x4146e85ac127863c in  ()
#30 0x41fd257d423408ff in  ()
#31 0x41d101a6418f7448 in  ()
#32 0xc024b90640043dee in  ()
#33 0xc11306fe4119fb94 in  ()
#34 0x400f0dfec0e75680 in  ()
#35 0xc180ed8ac0bac6c3 in  ()
--Type <RET> for more, q to quit, c to continue without paging--
#36 0x41033db5c168a502 in  ()
[... 166 lines cut ...]
#199 0xc128a8e0c19c20da in  ()
#200 0xc23093c1c2304e49 in  ()
#201 0xc2010e70416d5909 in  ()
#202 0xc17befe641c47db6 in  ()
#203 0x4165dde0c17ce662 in  ()
#204 0xc0d5e7c5c0b91e14 in  ()
#205 0xc1670910414e9fa5 in  ()
#206 0x42234f8841127ed3 in  ()
#207 0x421f7cd4410266bf in  ()
#208 0xc16f8c0cc0c4be84 in  ()
#209 0xc1cc8fd1c171d956 in  ()
#210 0x40748466c049cbe8 in  ()
#211 0x4188ffefc1874bc4 in  ()
#212 0xc129da84c1989c5c in  ()
#213 0xc212ebf4417aefb6 in  ()
#214 0xc19545963ee24820 in  ()
#215 0x40d625dec246f416 in  ()
#216 0x3f224890c25823ac in  ()
#217 0xc14d3264c1512a29 in  ()
#218 0xc1a0219cc091e4ac in  ()
#219 0xc1f689d2c25c2ad2 in  ()
#220 0xc1a730f0c20a5bf2 in  ()
--Type <RET> for more, q to quit, c to continue without paging--q
Quit
```

andimik, 2023-03-15:
This is auto scan, this time the `gdb` is combined with `bt full`, crash is at min 0:30.

[Bildschirmaufzeichnung vom 15.03.2023, 07:20:16.webm](https://user-images.githubusercontent.com/24510556/225223875-d4a9960f-d7cf-455e-909c-2e8b1f7face5.webm)

```
Allocating 15 zero-copy buffers
	Service: ENERGY SID = 0xE0AC51, SCIdS = 0
	Service: Rock Antenne SID = 0xE0AD27, SCIdS = 0
	Service: jö.live SID = 0xE0AD56, SCIdS = 0
	Service: arabella RELAX SID = 0xE0AD54, SCIdS = 0
	Service: KLASSIK RADIO SID = 0xE0AD53, SCIdS = 0
	Service: * 88.6 * SID = 0xE0AC47, SCIdS = 0
	Service: -AUSTRIA SID = 0xE0A3E0, SCIdS = 0
	Service: Radio Flamingo SID = 0xE0AD5A, SCIdS = 0
	Service: Antenne Österrei SID = 0xE0AD57, SCIdS = 0
	Service: arabella HOT SID = 0xE0AD59, SCIdS = 0
	Service: Radio Maria SID = 0xE0A3DD, SCIdS = 0
	Service: 'WELLE 1 SID = 0xE0AD58, SCIdS = 0
	Service: *STEPHANSDOM* SID = 0xE0AC52, SCIdS = 0
	Service: Mein Kinderradio SID = 0xE0AD55, SCIdS = 0
	Service: ERF Süd SID = 0xE0AD24, SCIdS = 0
	Service: #Technikum ONE SID = 0xE0AD2A, SCIdS = 0
[Thread 0x7fffa4fff640 (LWP 3498) exited]
[New Thread 0x7fffa4fff640 (LWP 3501)]
Allocating 15 zero-copy buffers
[Thread 0x7fffa4fff640 (LWP 3501) exited]
[New Thread 0x7fffa4fff640 (LWP 3517)]
Allocating 15 zero-copy buffers
[Thread 0x7fffa4fff640 (LWP 3517) exited]
[New Thread 0x7fffa4fff640 (LWP 3533)]
Allocating 15 zero-copy buffers
[Thread 0x7fffa4fff640 (LWP 3533) exited]
[New Thread 0x7fffa4fff640 (LWP 3549)]
Allocating 15 zero-copy buffers
[Thread 0x7fffd27fb640 (LWP 3423) exited]
[Thread 0x7fffd1ffa640 (LWP 3424) exited]
[Thread 0x7fffd2ffc640 (LWP 3422) exited]
[Thread 0x7fffd37fd640 (LWP 3421) exited]
[Thread 0x7fffa4fff640 (LWP 3549) exited]
[New Thread 0x7fffa4fff640 (LWP 3567)]
Allocating 15 zero-copy buffers
[Thread 0x7fffa4fff640 (LWP 3567) exited]
[New Thread 0x7fffa4fff640 (LWP 3583)]
Allocating 15 zero-copy buffers
[Thread 0x7fffa4fff640 (LWP 3583) exited]
[New Thread 0x7fffa4fff640 (LWP 3599)]
Allocating 15 zero-copy buffers
*** stack smashing detected ***: terminated

Thread 13 "dabsdr" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffd0ff8640 (LWP 3426)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=4704720117865852527) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: Datei oder Verzeichnis nicht gefunden.
(gdb) bt full
#0  __pthread_kill_implementation
    (no_tid=0, signo=6, threadid=4704720117865852527)
    at ./nptl/pthread_kill.c:44
        tid = <optimized out>
        ret = 0
        pd = 0x414a860541072e6f

                    old_mask = {__val = {5135414499197818050, 844420635172768, 5111485144741147566, 5100353538698859242, 5102147671079163678, 5049696976408998468, 5003206699916594508, 5096578266736415062, 5088020995354521478, 5103422425954868098, 5070567661322076942, 5091722990891856436, 5118051836179729939, 5119641446534316339, 5121660875757037502, 5105627973280572640}}
        ret = <optimized out>
#1  __pthread_kill_internal (signo=6, threadid=4704720117865852527)
    at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=4704720117865852527, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#3  0x00007ffff6042476 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
        ret = <optimized out>
#4  0x00007ffff60287f3 in __GI_abort () at ./stdlib/abort.c:79
        save_stage = 1

--Type <RET> for more, q to quit, c to continue without paging--
                  act = {__sigaction_handler = {sa_handler = 0x42fe000042fe0000, sa_sigaction = 0x42fe000042fe0000}, sa_mask = {__val = {4827295851711692800, 4658333644329532812, 13991482427474898576, 18446744073709551615, 4294967295, 4667609291916585885, 14102599351746446004, 5087180272700060979, 5140520815857161604, 5126816232353548748, 5088858629956579413, 5082468775188123784, 5096402907500531179, 5073864345069015174, 5050926333489339165, 5004960317890553120}}, sa_flags = 1193695756, sa_restorer = 0x47096e1c46840c61}

                    sigs = {__val = {32, 0, 1123942400, 0, 9223372034707292159, 9223372034707292159, 18446462598732840960, 18446462598732840960, 0, 18446744069414584320, 14022163449043487080, 4845175989533431086, 4674175652670079005, 4805365280556183388, 71777214294589695, 71777214294589695}}
#5  0x00007ffff60896f6 in __libc_message
    (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff61db943 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:155

                    ap = {{gp_offset = 24, fp_offset = 0, overflow_arg_area = 0x7fffd0fef9a0, reg_save_area = 0x7fffd0fef930}}
        fd = <optimized out>
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#6  0x00007ffff613676a in __GI___fortify_fail
--Type <RET> for more, q to quit, c to continue without paging--
    (msg=msg@entry=0x7ffff61db92b "stack smashing detected")
    at ./debug/fortify_fail.c:26
#7  0x00007ffff6136736 in __stack_chk_fail () at ./debug/stack_chk_fail.c:24
#8  0x00007ffff7f9d373 in  ()
    at /home/andreas/apps/AbracaDABra/lib/linux_x86_64/libdabsdr.so.2
#9  0x00007ffff7f9b33e in  ()
    at /home/andreas/apps/AbracaDABra/lib/linux_x86_64/libdabsdr.so.2
#10 0x3c973280416fc269 in  ()
#11 0x40c94113416fe746 in  ()
#12 0x3d9b9480c08e3c2c in  ()
#13 0xc09dc2a8bf6518f8 in  ()
#14 0xc09c47ba4195c4da in  ()
#15 0xc181b9ed40ce3c95 in  ()
#16 0xc1203efdc14d45aa in  ()
#17 0x40e3e6e2c1091b7a in  ()
#18 0x412cc925c1310064 in  ()
#19 0x4099fa4ac0b7ae90 in  ()
#20 0xc1a5d32bbdf4a360 in  ()
#21 0xc1fd249dc13af625 in  ()
#22 0xc03d32a23fd839a0 in  ()
#23 0x40d0cf624081e76e in  ()
#24 0xc10a24b7c18229f0 in  ()
#25 0xc107f293bfd81462 in  ()

```

andimik, 2023-03-15:
Update: the crash is at 7D.

This channel is used by two different muxes (Slovenia and Italy), but both are too weak.

[Bildschirmaufzeichnung vom 15.03.2023, 07:30:18.webm](https://user-images.githubusercontent.com/24510556/225225472-20791a19-8737-488d-8fde-141298aeee78.webm)

andimik, 2023-03-15:
I fear we were talking about two different versions.

I meant AbracaDABra version and you DABSDR version ...

Because my 1.99.4+ (exec file in different folder under Ubuntu) already uses DABSDR version 2.1.0.

[image]

What would you like me to test? An older version of DABSDR?

KejPi, 2023-03-15:
OK, it makes sense now :-) Then we are aligned, the bug is in DABSDR 2.1. I will release DABSDR binaries soon today and then I would like to ask you to test it with your setup to be sure that I have not overseen anything before release new version of AbracaDABra.

andimik, 2023-03-15:
Confirmed, no crash now with 2.1.1 :-)

BTW: What does

```
RTL-SDR: dropping 1638400 bytes...
RTL-SDR: dropping 1638400 bytes...
RTL-SDR: dropping 1638400 bytes...
RTL-SDR: dropping 1638400 bytes...
```
mean?

KejPi, 2023-03-15:
It means a problem. It happens when DABSDR is not able to process the data fast enough. Does it start to happen after update to 2.1.1? If this is the case then there is still some problem :-(

andimik, 2023-03-15:
I've tested all three options in the gain control.

[image]

The message appears in all 3 cases and **mainly** in that channel which was crashing before (in my case 7D), but **also** on some empty channels (= not on all).

KejPi, 2023-03-15:
Well, synchronization is quite CPU demanding processing, now it needs even more MIPS due to selectivity filter. I am running the app on Ubuntu on >10 years old laptop now and do not see any samples dropping. I am trying all channels and it does not happen. 
Do you see high CPU usage when this happens? Maybe some process hanging in the memory? I hate to say that but have you tried to restart the PC?

andimik, 2023-03-15:
On a very very weak signal `top` says, Abra uses more than 100%, on a normal signal (which can be decoded in the end) the value is 10-15 % typically.

[Bildschirmaufzeichnung vom 15.03.2023, 14:44:55.webm](https://user-images.githubusercontent.com/24510556/225327763-122d1224-3168-41a6-b628-619947b4bbbe.webm)

KejPi, 2023-03-15:
Yes, CPU usage is the problem. I assume that you are running Release build. But I do not see such huge difference between channel with signal and without. Does it happen with prebuilt AppImage too? 
Do you also have high CPU usage when running from file (the same channel recoded from other tool)?

andimik, 2023-03-15:
Qt-DAB: 60% in top on a very weak channel

Then I've made a recording of the same channel and played it (continuously) in Abra.

[Bildschirmaufzeichnung vom 15.03.2023, 15:08:18.webm](https://user-images.githubusercontent.com/24510556/225334287-f1c7bb32-f867-4a11-bdf6-06b618a974a0.webm)

KejPi, 2023-03-16:
Version 2.1.1 with fix of this issue has been released, closing.

closed 2023-03-16 by KejPi