KejPi/AbracaDABra #72: Problem with .ts-files

gvanem, 2023-03-20:
Trying to select **Deutch** in the Settings menu and restarting `Abracadabra.exe`, does not show any German texts.
But the Settings menu still claims I have **Deutch** as the active language. 
So 2 problems here:
* Some problem with this `lrelease` program in my case? Exactly what steps are done for these `.ts` files and
   how are they added into the build process? As resources?
* Failure to load a language should revert to **English**. That does not happen.

The 2nd problem I fixed like this:
```diff
--- a/gui/main.cpp 2023-02-28 13:22:42
+++ b/gui/main.cpp 2023-03-20 10:40:51
@@ -74,8 +74,6 @@

     QLocale::Language lang = QLocale::codeToLanguage(settings->value("language", QString("")).toString());

-    delete settings;
-
     QTranslator translator;
     if (QLocale::AnyLanguage == lang)
     {   // system default
@@ -90,7 +88,13 @@
         {
             a.installTranslator(&translator);
         }
+        else
+        {
+          settings->setValue("language", QLocale::languageToCode(QLocale::English));
+          qWarning() << "Restoring language to QLocale::English";
     }
+    }
+    delete settings;

```

But what about the 1st?

KejPi, 2023-03-20:
Translation files (*.ts) are transformed to *.qm during build and stored to resources. Thus it cannot happen that translation is not available if build was done correctly. I suppose that official Windows binary works correctly. 

Pleas check your building environment, it really seems to be problem with `lrelease`, translations are built at the beginning:

```
$ make
[  2%] Generating AbracaDABra_cs.qm
Updating '/Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_cs.qm'...
    Generated 637 translation(s) (637 finished and 0 unfinished)
[  4%] Generating AbracaDABra_de.qm
Updating '/Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_de.qm'...
    Generated 566 translation(s) (565 finished and 1 unfinished)
    Ignored 71 untranslated source text(s)
[  6%] Generating AbracaDABra_pl.qm
Updating '/Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_pl.qm'...
    Generated 636 translation(s) (633 finished and 3 unfinished)
    Ignored 1 untranslated source text(s)
[  6%] Built target AbracaDABra_lrelease

...
```

gvanem, 2023-03-20:
> $ make

Thanks. But can I get the `make VERBOSE=1` version of that? 
I assume some `.qm` file should be read as a resource? If so, where and how?

KejPi, 2023-03-20:
On Mac it is enough to provide `-DCMAKE_PREFIX_PATH=$QT_PATH/lib/cmake` to cmake command and then cmake takes care of everything. I am using QT Creator under Windows that can handle it correctly. I would assume that `CMAKE_PREFIX_PATH` should work under Windows too.

`.qm` files are compiled in application resources, they are not as file on the disk. 

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f gui/CMakeFiles/AbracaDABra_lrelease.dir/build.make gui/CMakeFiles/AbracaDABra_lrelease.dir/depend
cd /Users/kejpi/Devel/AbracaDABra/build-test && /opt/homebrew/Cellar/cmake/3.26.0/bin/cmake -E cmake_depends "Unix Makefiles" /Users/kejpi/Devel/AbracaDABra /Users/kejpi/Devel/AbracaDABra/gui /Users/kejpi/Devel/AbracaDABra/build-test /Users/kejpi/Devel/AbracaDABra/build-test/gui /Users/kejpi/Devel/AbracaDABra/build-test/gui/CMakeFiles/AbracaDABra_lrelease.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f gui/CMakeFiles/AbracaDABra_lrelease.dir/build.make gui/CMakeFiles/AbracaDABra_lrelease.dir/build
[  2%] Generating AbracaDABra_cs.qm
cd /Users/kejpi/Devel/AbracaDABra/build-test/gui && /Users/kejpi/Devel/Qt/6.4.3/macos/bin/lrelease /Users/kejpi/Devel/AbracaDABra/gui/l10n/AbracaDABra_cs.ts -qm /Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_cs.qm
Updating '/Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_cs.qm'...
    Generated 637 translation(s) (637 finished and 0 unfinished)
[  4%] Generating AbracaDABra_de.qm
cd /Users/kejpi/Devel/AbracaDABra/build-test/gui && /Users/kejpi/Devel/Qt/6.4.3/macos/bin/lrelease /Users/kejpi/Devel/AbracaDABra/gui/l10n/AbracaDABra_de.ts -qm /Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_de.qm
Updating '/Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_de.qm'...
    Generated 566 translation(s) (565 finished and 1 unfinished)
    Ignored 71 untranslated source text(s)
[  6%] Generating AbracaDABra_pl.qm
cd /Users/kejpi/Devel/AbracaDABra/build-test/gui && /Users/kejpi/Devel/Qt/6.4.3/macos/bin/lrelease /Users/kejpi/Devel/AbracaDABra/gui/l10n/AbracaDABra_pl.ts -qm /Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_pl.qm
Updating '/Users/kejpi/Devel/AbracaDABra/build-test/gui/AbracaDABra_pl.qm'...
    Generated 636 translation(s) (633 finished and 3 unfinished)
    Ignored 1 untranslated source text(s)
[  6%] Built target AbracaDABra_lrelease
```

gvanem, 2023-03-20:
Thanks. So basically it does:
```
lrelease gui/l10n/AbracaDABra_cs.ts -qm gui/AbracaDABra_cs.qm
lrelease gui/l10n/AbracaDABra_de.ts -qm gui/AbracaDABra_de.qm
lrelease gui/l10n/AbracaDABra_pl.ts -qm gui/AbracaDABra_pl.qm
```

> .qm files are compiled in application resources, they are not as file on the disk.

That's what I do not understand. How exactly? No in effin CMake terms. but the actual command. 
I assume it'a `rcc ..something`.

KejPi, 2023-03-20:
This is probably what you are looking for. 

```
[ 10%] Generating qrc_resources.cpp
cd /Users/kejpi/Devel/AbracaDABra/build-test/gui && /Users/kejpi/Devel/Qt/6.4.3/macos/libexec/rcc --no-zstd --name resources --output /Users/kejpi/Devel/AbracaDABra/build-test/gui/qrc_resources.cpp /Users/kejpi/Devel/AbracaDABra/gui/resources.qrc
[ 12%] Running rcc for resource AbracaDABra_translations
cd /Users/kejpi/Devel/AbracaDABra/build-test/gui && /Users/kejpi/Devel/Qt/6.4.3/macos/libexec/rcc --output /Users/kejpi/Devel/AbracaDABra/build-test/gui/.rcc/qrc_AbracaDABra_translations.cpp --name AbracaDABra_translations /Users/kejpi/Devel/AbracaDABra/build-test/gui/.rcc/AbracaDABra_translations.qrc --no-zstd
```

I do not understand why don't you simply run CMake on your system and do your investigations if you want to port it to other environment. Qt is providing set of *.cmake files that are prepared for easy integration of Qt inside CMake environment.

gvanem, 2023-03-20:
> I do not understand why don't you simply run CMake on your system and ...

CMake here fails miserably. I thinks it just makes the build process a lot harder to understand. Even compared to 
automake etc.

gvanem, 2023-03-21:
I finally figured it out with the help of a Wireshark build. No thanks to the cryptic macros in Qt and CMake.
* *Compile* into .qm:
 ```
 lrelease gui/l10n/AbracaDABra_cs.ts -qm objects/AbracaDABra_cs.qm
 lrelease gui/l10n/AbracaDABra_de.ts -qm objects/AbracaDABra_de.qm
 lrelease gui/l10n/AbracaDABra_pl.ts -qm objects/AbracaDABra_pl.qm
 ```
* Then manually generate this `objects/i18n.qrc:`
   ```
  <RCC>
   <qresource prefix="/i18n">
     <file>AbracaDABra_cs.qm</file>
     <file>AbracaDABra_de.qm</file>
     <file>AbracaDABra_pl.qm</file>
   </qresource>
  </RCC>
   ```
   It **must** be `<RCC>`. Not `<rcc>`!!
* Convert the above to C++:   `rcc --output objects/qrc_i18n.cpp objects/i18n.qrc`
* Compile it: `clang-cl -c @clang-cl.args -Fo./objects/qrc_i18n.obj objects/qrc_i18n.cpp`
* Then finally link this `objects/qrc_i18n.obj` to the rest.

It now shows **Deutch**:
[image]

closed 2023-03-21 by gvanem