KejPi/AbracaDABra #73: Segfault when rtl_tcp gets lost

andimik, 2023-03-21:
Using the rtl_tcp from Oldenburger ( [64fb83ced0](https://github.com/old-dab/rtlsdr/commit/64fb83ced0770ed1a44bd42b5072366f282a620d) ) I get a segfault when the connection gets lost:

```
RTL-TCP: error:  Connection reset by peer
RTL-TCP: server disconnected.
[Thread 0x7fffb1dfe640 (LWP 86854) exited]
[Thread 0x7fffb0a83640 (LWP 86857) exited]

Thread 1 "AbracaDABra_wri" received signal SIGSEGV, Segmentation fault.
unlink_chunk (p=p@entry=0x5555595771c0, av=<optimized out>) at ./malloc/malloc.c:1641
1641	./malloc/malloc.c: No such file or directory.
(gdb) bt full
#0  unlink_chunk (p=p@entry=0x5555595771c0, av=<optimized out>) at ./malloc/malloc.c:1641
        fd = 0x7ffff621a4b0 <main_arena+2096>
        bk = 0x7ffff621a4b0 <main_arena+2096>
#1  0x00007ffff60a4841 in _int_malloc (av=av@entry=0x7ffff6219c80 <main_arena>, bytes=bytes@entry=1048576) at ./malloc/malloc.c:4204
        p = <optimized out>
        iters = <optimized out>
        nb = 1048592
        idx = 126
        bin = 0x7ffff621a4b0 <main_arena+2096>
        victim = 0x5555595771c0
        size = 1638448
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = <optimized out>
        tcache_nb = 0
        tc_idx = 65535
        return_cached = <optimized out>
        __PRETTY_FUNCTION__ = "_int_malloc"
#2  0x00007ffff60a51b9 in __GI___libc_malloc (bytes=1048576) at ./malloc/malloc.c:3329
        ar_ptr = 0x7ffff6219c80 <main_arena>
        victim = <optimized out>
        tbytes = <optimized out>
        __PRETTY_FUNCTION__ = "__libc_malloc"
#3  0x00007ffff6f73110 in QImageData::create(QSize const&, QImage::Format) () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#4  0x00007ffff6f733bf in QImage::QImage(QSize const&, QImage::Format) () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#5  0x00007ffff6f88642 in QImageIOHandler::allocateImage(QSize, QImage::Format, QImage*) () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#6  0x00007ffff72f1155 in  () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#7  0x00007ffff6f8d0d1 in QImageReader::read(QImage*) () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#8  0x00007ffff6f8d879 in QImageReader::read() () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
--Type <RET> for more, q to quit, c to continue without paging--
#9  0x00007ffff6fa8313 in QPlatformPixmap::fromFile(QString const&, char const*, QFlags<Qt::ImageConversionFlag>) ()
    at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#10 0x00007ffff6f9e40d in QPixmap::load(QString const&, char const*, QFlags<Qt::ImageConversionFlag>) ()
    at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#11 0x0000555555618334 in Ui_AboutDialog::setupUi(QDialog*) ()
#12 0x0000555555616a99 in AboutDialog::AboutDialog(QWidget*) ()
#13 0x0000555555597f79 in MainWindow::showAboutDialog() ()
#14 0x00007ffff69ac023 in  () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#15 0x00007ffff72d5f66 in QAction::triggered(bool) () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#16 0x00007ffff72d8b2d in QAction::activate(QAction::ActionEvent) () at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#17 0x00007ffff794e3a2 in  () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#18 0x00007ffff7956231 in  () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#19 0x00007ffff77d90e8 in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#20 0x00007ffff777fd36 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#21 0x00007ffff77892d7 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#22 0x00007ffff6953a58 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#23 0x00007ffff7787de2 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#24 0x00007ffff77ec2b4 in  () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#25 0x00007ffff77ee635 in  () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#26 0x00007ffff777fd36 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#27 0x00007ffff6953a58 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#28 0x00007ffff6fcb6bf in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
    at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#29 0x00007ffff7013c8c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/x86_64-linux-gnu/libQt6Gui.so.6
#30 0x00007ffff1a7886e in  () at /usr/lib/x86_64-linux-gnu/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6
#31 0x00007ffff5d4ad3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff5d9f6c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff5d483e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007ffff6b7beae in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/x86_64-linux-gnu/libQt6Core.so.6
#35 0x00007ffff6960adb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#36 0x00007ffff695c0f3 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt6Core.so.6
#37 0x000055555557d5d7 in main ()

```

KejPi, 2023-03-21:
Are there specific steps/conditions to reproduce it? I cannot reproduce the crash on Ubuntu.

andimik, 2023-03-27:
yes, this fixes it.

closed 2023-03-27 by andimik