KejPi/AbracaDABra #77: Collapsing mux list highlights wrong service

andimik, 2023-04-28:
[Bildschirmaufzeichnung vom 28.04.2023, 06:29:47.webm](https://user-images.githubusercontent.com/24510556/235055238-d2caf270-c475-4157-af00-d1397a7eefc8.webm)

This is just a cosmetic issue which I could reproduce only sometimes.

In this video I was clicking on the 1st service of each mux. After collapsing the mux list, a wrong service got highlighted on the right, which was never selected before. You still can listen to the service marked on the left of course.

KejPi, 2023-05-04:
Can it be reproduced just selecting some (maybe first) service in the MUX and then keeping collapsing and uncollapsing the ensemble? Without selecting services from other ensembles before?

andimik, 2023-05-05:
I'll try at home.

andimik, 2023-05-08:
I found the root cause (drag&drop) but had no luck with the parameters in Qt-Designer in `serviceTreeView`

[image]

Here I've just held the left mouse button while moving ... (like drag&drop, but this not activated)

[image]

[Bildschirmaufzeichnung vom 08.05.2023, 19:32:44.webm](https://user-images.githubusercontent.com/24510556/236891653-a14bb9a3-10e4-4f8a-a992-03b95408e457.webm)

KejPi, 2023-05-08:
This is something I can finally reproduce. I agree that it behaves undesired way leading to selection of the service in service list that does not play but can this behavior explain the what you have observed before?

andimik, 2023-05-08:
I guess it explains it. Because when you collapse the list, you are holding down the button.

KejPi, 2023-05-09:
OK, it could be. I will try to find a way to fix this behavior.

KejPi, 2023-05-09:
I have fixed "drag" issue in e7da6213dd28d5e545815ddffb60be3aa36aaa3a Please check, that the original issue is fixed too.

KejPi, 2023-05-09:
Fixed in f29385e19ace4f4ddbf34cccd3a3131b94790158

andimik, 2023-05-10:
Yes, looks good!

BTW: This is the updated German text
[AbracaDABra_de.zip](https://github.com/KejPi/AbracaDABra/files/11444383/AbracaDABra_de.zip)

Hint: "Application log" and the three buttons on the bottom cannot be translated at the moment. I tried to `lupdate` but it does not work.

closed 2023-05-10 by andimik