KejPi/AbracaDABra #78: Make DLS selectable by mouse

andimik, 2023-05-12:
The following patch allows a user to select the DL text and copy it to clipboard for instance:

[image]

```
Sie hören: CROWDED HOUSE - WEATHER WITH YOU auf Technikum ONE
```

```
diff --git a/gui/mainwindow.ui b/gui/mainwindow.ui
index 08649b8..875bbed 100644
--- a/gui/mainwindow.ui
+++ b/gui/mainwindow.ui
@@ -542,6 +542,10 @@ QScrollBar::add-page:vertical, .QScrollBar::sub-page:vertical {
               <property name="wordWrap">
                <bool>true</bool>
               </property>
+              <property name="textInteractionFlags">^M
+               <set>Qt::LinksAccessibleByMouse|Qt::TextSelectableByMouse</set>^M
+              </property>^M
+^M
              </widget>
             </item>
            </layout>
```
Don't know what `^M` stands for ...

KejPi, 2023-05-15:
Implemented bit different way but copying works. Closing.

closed 2023-05-15 by KejPi

andimik, 2023-05-15:
Wow, cool feature, I agree, it's different, but unique!