KejPi/AbracaDABra #85: Crash when switching "Audio output"

gvanem, 2023-06-18:
When using the menu *"Audio output"* and switch from *"Speakers (Realtek)"* to *"Realtek Digital Output"*
(or vise versa) the audio gets muted but Abracadabra crashes deep inside my Qt 6.2.4 with a 
NULL-ptr access:
```
  test  byte ptr [rsi+30h],2   ; rsi = 0
```
Building for `_DEBUG`-mode, the instruction is:
```
   mov  eax,dword ptr [rax+30h]  ; rax=dddddddddddddddd
```
It seems to be a timer-function that's called after it has been deleted. But which?
This started to happen with some change 3 or 4 days ago. Probably [this](https://github.com/KejPi/AbracaDABra/commit/496b40a56d549d5bbeb6e2e56b0a3c1d32e07bde).

In `_DEBUG`-mode it happens every time. The last console output before the crash:
```
03:30:36.675 [W] MOTObject: Removing duplicate header parameter 5
03:30:36.675 [W] MOTObject: Removing duplicate header parameter 37
03:30:36.676 [W] MOTObject: Removing duplicate header parameter 38
03:30:36.680 [W] MOTObject: Removing duplicate header parameter 39
03:30:55.824 [I] Application: Audio device selected "Realtek Digital Output (Realtek(R) Audio)"
03:30:55.840 [I] AudioOutput: Muting... [available 12000 samples]
```

The call-stack in `_DEBUG` mode:
```
Qt6Cored!doActivate<0>(class QObject * sender = 0x000001b5`04be1c90, int signal_index = 0n3, void ** argv = 0x000000a2`f62f9468)+0x46
Qt6Cored!QMetaObject::activate(class QObject * sender = 0x000001b5`04be1c90, struct QMetaObject * m = 0x00007ff8`d83c5690, int local_signal_index = 0n0, void ** argv = 0x000000a2`f62f9468)+0x68
Qt6Multimediad!QPlatformAudioSink::errorChanged(QAudio::Error _t1 = UnderrunError (0n3))+0x56
Qt6Multimediad!QWindowsAudioSink::deviceStateChange(QAudio::State state = IdleState (0n3), QAudio::Error error = UnderrunError (0n3))+0x272
Qt6Multimediad!QWindowsAudioSink::pullSource(void)+0x1e4
Qt6Multimediad!QtPrivate::FunctorCall<QtPrivate::IndexesList<>,QtPrivate::List<>,void,void (<function> * f = 0x00007ff8`d82022bb, class QWindowsAudioSink * o = 0x000001b5`04be1c90, void ** arg = 0x000000a2`f62f9aa8)+0x1d
Qt6Multimediad!QtPrivate::FunctionPointer<void (<function> * f = 0x00007ff8`d82022bb, class QWindowsAudioSink * o = 0x000001b5`04be1c90, void ** arg = 0x000000a2`f62f9aa8)+0x28
Qt6Multimediad!QtPrivate::QSlotObject<void (int which = 0n1, class QtPrivate::QSlotObjectBase * this_ = 0x000001b5`04aa0e00, class QObject * r = 0x000001b5`04be1c90, void ** a = 0x000000a2`f62f9aa8, bool * ret = 0x00000000`00000000)+0x82
Qt6Cored!QtPrivate::QSlotObjectBase::call(class QObject * r = 0x000001b5`04be1c90, void ** a = 0x000000a2`f62f9aa8)+0x43
Qt6Cored!doActivate<0>(class QObject * sender = 0x000001b5`04be1ce0, int signal_index = 0n3, void ** argv = 0x000000a2`f62f9aa8)+0x908
Qt6Cored!QMetaObject::activate(class QObject * sender = 0x000001b5`04be1ce0, struct QMetaObject * m = 0x00007ff8`a5c9c170, int local_signal_index = 0n0, void ** argv = 0x000000a2`f62f9aa8)+0x68
Qt6Cored!QTimer::timeout(struct QTimer::QPrivateSignal _t1 = struct QTimer::QPrivateSignal)+0x55
Qt6Cored!QTimer::timerEvent(class QTimerEvent * e = 0x000000a2`f62fb438)+0x65
Qt6Cored!QObject::event(class QEvent * e = 0x000000a2`f62fb438)+0xaf
Qt6Widgetsd!QApplicationPrivate::notify_helper(class QObject * receiver = 0x000001b5`04be1ce0, class QEvent * e = 0x000000a2`f62fb438)+0x1ab
Qt6Widgetsd!QApplication::notify(class QObject * receiver = 0x000001b5`04be1ce0, class QEvent * e = 0x000000a2`f62fb438)+0x510
Qt6Cored!QCoreApplication::notifyInternal2(class QObject * receiver = 0x000001b5`04be1ce0, class QEvent * event = 0x000000a2`f62fb438)+0x157
Qt6Cored!QCoreApplication::sendEvent(class QObject * receiver = 0x000001b5`04be1ce0, class QEvent * event = 0x000000a2`f62fb438)+0x2f
Qt6Cored!QEventDispatcherWin32Private::sendTimerEvent(int timerId = 0n43)+0xb9
Qt6Cored!QEventDispatcherWin32::event(class QEvent * e = 0x000001b5`065c6610)+0x1fd
Qt6Widgetsd!QApplicationPrivate::notify_helper(class QObject * receiver = 0x000001b5`0657a360, class QEvent * e = 0x000001b5`065c6610)+0x1ab
Qt6Widgetsd!QApplication::notify(class QObject * receiver = 0x000001b5`0657a360, class QEvent * e = 0x000001b5`065c6610)+0x510
Qt6Cored!QCoreApplication::notifyInternal2(class QObject * receiver = 0x000001b5`0657a360, class QEvent * event = 0x000001b5`065c6610)+0x157
Qt6Cored!QCoreApplication::sendEvent(class QObject * receiver = 0x000001b5`0657a360, class QEvent * event = 0x000001b5`065c6610)+0x2f
Qt6Cored!QCoreApplicationPrivate::sendPostedEvents(class QObject * receiver = 0x00000000`00000000, int event_type = 0n0, class QThreadData * data = 0x000001b5`04aaa710)+0x55f
Qt6Cored!QEventDispatcherWin32::sendPostedEvents(void)+0x8f
Qt6Cored!QEventDispatcherWin32::processEvents(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0xbf
Qt6Cored!QEventLoop::processEvents(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0x7c
Qt6Cored!QEventLoop::exec(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0x1f4
Qt6Cored!QThread::exec(void)+0xe8
Qt6Cored!QThread::run(void)+0x14
Qt6Cored!QThreadPrivate::start(void * arg = 0x000001b5`0657aef0)+0x260
KERNEL32!BaseThreadInitThunk+0x14
ntdll!RtlUserThreadStart+0x21
```

At the crash-site, I note that `QObject *sender` contains junk:
```
sender     == 0x000001b5`04be1c90 class QObject *
 |___vfptr == 0xdddddddd`dddddddd
 |
 |___ signal_spy_set == 0xcccccccc`cccccccc struct QSignalSpyCallbackSet *
 |___ sp             == 0xdddddddd`dddddddd class QObjectPrivate *
```

KejPi, 2023-06-18:
I cannot reproduce the crash on MacOS with Qt 6.4.3 nor on Windows wit Qt 6.5.1. Qt 6.2 is too old, I will probably disable the feature fro Qt < 6.4 and make PortAudio default as before. It has other issues but at least not crashing under normal conditions.

closed 2023-06-18 by KejPi

gvanem, 2023-06-19:
>  ... nor on Windows wit Qt 6.5.1. 

I'm installing Qt 6.5.1 right now (a 5 GByte download). I'll then check if this is an issue with Qt 6.2.4 or with Abracadabra.
But due to the `0xdddddddd`dddddddd` pointer-values, it looks like a *use after free* issue somewhere.

gvanem, 2023-06-19:
Now I've verified it was indeed a Qt 6.2.4 issue. I rebuilt everything from `master` using Qt 6.5.2 and editing 
my `%APPDATA%/AbracaDABra/AbracaDABra.ini` to say `audioFramework=1`. (i.e. *QtAudio*).

No crashes seen now when switching back and forth between  "*Speakers (Realtek)*" and "*Realtek Digital Output*".
Tested both `_RELEASE` and `_DEBUG` modes. Good job!

KejPi, 2023-06-19:
I am glad it works but I am still not satisfied with Qt Audio framework stability. I will focus on PortAudio improvements and try to implement some kind of device selection for it as well even though how-plug is not supported by PortAudio.