KejPi/AbracaDABra #88: DAB Melbourne 1 crashes

b1he, 2023-08-19:
Hello,

AbracaDABra most of the time is experiencing crashes when selecting any service within the ensemble DAB Melbourne 1. This started to happen when services iHeartAustralia and iHeartDance were added to the ensenble, I am using AbracaDABra 2.2.1 on Windows.  I have attached unsigned 8 bit recordings where 20-32-13_202928000Hz.bin is working and 20-30-50_202928000Hz.bin is crashing.
[sample.zip](https://github.com/KejPi/AbracaDABra/files/12385596/sample.zip)

KejPi, 2023-08-19:
Hello, thanks for reporting and sharing the recording. I can reproduce the issue on my side. Could you please share longer version of non working case (at least 30sec) ? I know how to fix the symptoms but I would like to understand what is really going on.

andimik, 2023-08-19:
https://www.digitalbitrate.com/dtv.php?mux=9A&liste=1&live=628&lang=en

For those who are interested, there is a scanner in Melbourne, too.

b1he, 2023-08-20:
Thanks for the quick response and release! It will no longer immediately crash. I noticed sometimes if I quickly switch between RSN Xtra 2 and other services, it is unable to change and produces these warnings:

```
21:31:04.359 [W] RadioControl: RadioControlEvent::SERVICE_SELECTION error 2
21:31:38.881 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F010C3, SCIdS 00: status 2
```

There are still some crashes occurring when I attempt to add favourite services. Curiously the ensemble country says Malaysia not Australia. Attached is a one minute recording. https://mega.nz/file/kdJWFT7D#C4ThaOi3IOUOJ-PTwPfOz5UHgoL6m_XzUKmXXfC3EaI

KejPi, 2023-08-20:
Thank you for sharing the recording, very appreciated!

I can meanwhile answer "Malaysia"issue - it is because ECC is 0xF0 and ensemble ID is 0xF003 thus country code (first 4 bits of ensemble ID) is 0xF. According to [ETSI TS 101 756 V2.4.1](https://www.etsi.org/deliver/etsi_ts/101700_101799/101756/02.04.01_60/ts_101756v020401p.pdf) table 7 it is interpreted  as Malaysia. For Australia, ensemble ID should start with country ID 1-8. It seems that other ensembles in Melbourne are violating standard too. 

I will try to reproduce other issues. I am very surprised about favourite crashes since this feature is widely used and nobody complained. Did you observe this problem since you started to use the app or has it started to happen recently?

b1he, 2023-08-21:
Interesting. Looking at the digitalbitrate.com site, Perth is also using F.

After further testing, it seems the favourites crashing issue is not related to a particular service/ensemble but instead caused by using the services list (not the ensemble-tree). Choosing any services from this list then clicking the star will cause a crash. Using the ensemble-tree list instead does not cause a crash. 2.1.1 is not affected by 2.2.1 and 2.2.2 is.

KejPi, 2023-08-22:
I have just released 2.2.3 that should fix the issues. Please let me know if it works.

b1he, 2023-08-23:
There were some issues with services removed from the ensemble tree but after a re-scan, it is all fixed. Other than that, it is ok.

```
15:12:51.653 [I] ServiceList: Removing service: [F0F003] RSN Xtra 2         F010C3 : 0
15:12:51.653 [I] ServiceList: Removing service: [F0F003] RSN Xtra 1         F010C1 : 0
15:12:51.653 [I] ServiceList: Removing service: [F0F003] TAB Lounge         F010C2 : 0
```

```
15:24:57.957 [I] RadioControl: Tuning 202.928 -> 206.352 MHz ...
15:24:57.971 [I] AudioOutput: Muting... [available 1920 samples]
15:24:59.077 [I] RadioControl: Current frequency 206.352 MHz
15:24:59.898 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS PopAsia        F03903 : 255
15:24:59.899 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS Arabic24       F03904 : 255
15:24:59.902 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS PopDesi        F03905 : 255
15:24:59.905 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC Classic        F03104 : 255
15:24:59.907 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC Country        F0310B : 255
15:24:59.911 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC Sport          F03100 : 255
15:24:59.913 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC Jazz           F03109 : 255
15:24:59.916 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC KIDS listen    F0310A : 255
15:24:59.919 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC NewsRadio      F0310C : 255
15:24:59.921 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABCRadioNational   F03103 : 255
15:24:59.924 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] ABC MELBOURNE      F03102 : 255
15:24:59.927 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] Double J           F03108 : 255
15:24:59.929 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] triple j           F03106 : 255
15:24:59.932 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] triplejUnearthed   F03107 : 255
15:24:59.934 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS Radio 1        F03900 : 255
15:24:59.937 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS Radio 2        F03901 : 255
15:24:59.939 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS Radio 3        F03906 : 255
15:24:59.942 [I] ServiceList:           [F03100 @ 206352 kHz |  9C] SBS Chill          F03902 : 255
15:24:59.944 [I] ServiceList: Removing service: [F03100] triplejUnearthed   F03107 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC Sport          F03100 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC NewsRadio      F0310C : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS PopAsia        F03903 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC MELBOURNE      F03102 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS Radio 3        F03906 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS Chill          F03902 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] Double J           F03108 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS PopDesi        F03905 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS Radio 1        F03900 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS Radio 2        F03901 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC Jazz           F03109 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC Classic        F03104 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] triple j           F03106 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC Country        F0310B : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABC KIDS listen    F0310A : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] ABCRadioNational   F03103 : 0
15:24:59.944 [I] ServiceList: Removing service: [F03100] SBS Arabic24       F03904 : 0
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03100, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F0310B, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03104, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03905, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03904, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03903, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03108, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03102, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03103, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F0310C, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F0310A, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03109, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03902, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03906, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03901, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03900, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03107, SCIdS FF: status 2
15:25:01.082 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId F03106, SCIdS FF: status 2
```

KejPi, 2023-08-23:
It clearly does not work 🙁 I will try to fix it today, but I am not sure I will manage that. This is unexpected side effect from the change I did and it will probably affect all users 🙁

andimik, 2023-08-23:
What is the technical background? Too many services?

KejPi, 2023-08-23:
The background is that this ensemble is not compliant with current DAB standard (SCIdS: The primary service component shall use the value 0) and the library is not implemented for supporting it and thus there is some unexpected behaviour. I am trying to find a way to support it without sacrifying performance on compliant ensembles.

KejPi, 2023-08-23:
I have released 2.2.4 with new version of DABSDR library. I hope it will fix the issues. Please try and let me know.

b1he, 2023-08-24:
No problems so far 🙂

KejPi, 2023-08-25:
I am glad you do not see any problem so far, but I am afraid it still needs some fixes 🙁

KejPi, 2023-09-04:
Could you please try DABSDR 2.3.4 from attachment. Just replace DLL in AbracaDABra folder. You shoud see DABSDR version 2.3.4 in About application dialog. 
Please let me know if you see any side effects. 

[libdabsdr.zip](https://github.com/KejPi/AbracaDABra/files/12516296/libdabsdr.zip)

b1he, 2023-09-05:
After upgrading, the affected services become missing on the ensemble tree until a re-scan is performed
[image]

KejPi, 2023-09-06:
The services should update automatically once you tune the ensemble. Do I understand correctly that 3 services marked yellow appeared in the service list (left) but not in tree (right) until full band scan was performed? Or was it enough just to tune 9A channel to fix it?

b1he, 2023-09-06:
This is correct. A band scan or clear services list then selecting 9A is needed to have services appear correctly in the tree. Subsequent close and re-open of the application does not have this issue. 

Scan immediately after updating DLL

```
01:15:02.285 [I] RadioControl: Tuning 0.000 -> 202.928 MHz ...
01:15:02.737 [I] RadioControl: Current frequency 202.928 MHz
01:15:05.160 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] MMM 2000s          F0113B : 0
01:15:05.162 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] SEN Track          F01108 : 0
01:15:05.178 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Smooth 91.5        F0111B : 0
01:15:05.179 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] RSN Racing&Sport   F010C0 : 0
01:15:05.196 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] SEN 1116           F01106 : 0
01:15:05.211 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] 3MBS Fine Music    F0114E : 0
01:15:05.227 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] MMM 80s            F01137 : 0
01:15:05.242 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] LightMIX           F0114D : 0
01:15:05.257 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Coles Radio        F0111C : 0
01:15:05.272 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Light89.9          F01151 : 0
01:15:05.287 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] SEN SYDNEY         F0110A : 0
01:15:05.304 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] MMM COUNTRY        F01139 : 0
01:15:05.319 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] 3RRR Digital       F0114C : 0
01:15:05.334 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Light Christmas    F01152 : 0
01:15:05.350 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] RSN Xtra 1         F010C1 : 1
01:15:05.366 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Radio Maria        F0111E : 0
01:15:05.388 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] NICHE RADIO        F0110B : 0
01:15:05.404 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Chill Pop Hits     F0113A : 0
01:15:05.426 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] TAB Lounge         F010C2 : 2
01:15:05.444 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Nova 100           F0111A : 0
01:15:05.461 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] iHeartAustralia    F01132 : 0
01:15:05.479 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] SEN FAN            F01107 : 0
01:15:05.495 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Radio Maria Ital   F01008 : 0
01:15:05.510 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] VA IRIS Melb       F01150 : 0
01:15:05.527 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] DISRUPT            F0110C : 0
01:15:05.543 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] RSN Xtra 2         F010C3 : 4
01:15:05.577 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] NICHE RADIO 2      F0110D : 0
01:15:05.592 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] iHeartDance        F01133 : 0
01:15:05.611 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Smooth Relax       F0111D : 0
01:15:05.628 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] easy music 3MP     F01109 : 0
01:15:05.644 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] 3ZZZ Ethnic        F0114F : 0
01:15:05.661 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] MMM CLASSIC ROCK   F01138 : 0
01:15:05.678 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Smooth Vintage     F0111F : 0
01:15:05.697 [I] ServiceList: Removing service: [F0F003] TAB Lounge         F010C2 : 0
01:15:05.697 [I] ServiceList: Removing service: [F0F003] RSN Xtra 1         F010C1 : 0
01:15:05.697 [I] ServiceList: Removing service: [F0F003] RSN Xtra 2         F010C3 : 0
```

Scan after clearing services list
```
01:17:24.888 [I] RadioControl: Tuning 202.928 -> 0.000 MHz ...
01:17:27.335 [I] RadioControl: Tuning 0.000 -> 202.928 MHz ...
01:17:27.805 [I] RadioControl: Current frequency 202.928 MHz
01:17:30.227 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] SEN Track          F01108 : 0
01:17:30.230 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] RSN Xtra 2         F010C3 : 4
01:17:30.230 [I] ServiceList: Adding F010C3 : 4 as primary service [old DAB standard]
01:17:30.247 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] iHeartDance        F01133 : 0
01:17:30.250 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Smooth 91.5        F0111B : 0
01:17:30.266 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] RSN Racing&Sport   F010C0 : 0
01:17:30.284 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] SEN 1116           F01106 : 0
01:17:30.300 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] 3MBS Fine Music    F0114E : 0
01:17:30.316 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] MMM 80s            F01137 : 0
01:17:30.332 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] LightMIX           F0114D : 0
01:17:30.349 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Light Christmas    F01152 : 0
01:17:30.364 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Smooth Relax       F0111D : 0
01:17:30.382 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] easy music 3MP     F01109 : 0
01:17:30.398 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] Light89.9          F01151 : 0
[... cut ...]
01:17:30.682 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] RSN Xtra 1         F010C1 : 1
01:17:30.682 [I] ServiceList: Adding F010C1 : 1 as primary service [old DAB standard]
01:17:30.701 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] DISRUPT            F0110C : 0
01:17:30.721 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] MMM 2000s          F0113B : 0
01:17:30.740 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] NICHE RADIO 2      F0110D : 0
01:17:30.761 [I] ServiceList:           [F0F003 @ 202928 kHz |  9A] 3ZZZ Ethnic        F0114F : 0
```

KejPi, 2023-09-06:
I found the rootcause but I have decided not to fix it because it is only affecting old DAB standard transmission and it should not happen anymore unless SCIdS will change in future. It was caused by changing SCIdS from 0 (stored service list with older DLL) to N (service list using new DLL) on affected services.
By the way restarting of the application fixes it too, no need to rescan the band.

Side note: old standard 1.4.1 is from 2006, latest standard 2.1.1 is from 2017 and there is already new version in preparation. No transmission should use SCIdS != 0 for primary service today.

b1he, 2023-09-08:
Fair enough. I don't have anything to say. Good job!