KejPi/AbracaDABra #89: Leading space in short label not shown

andimik, 2023-09-11:
See attached xml file from Slovene DAB ensemble R1.

Some services have a leading space in the short label, but this is not shown.

[2023-09-12_005137_10D.zip](https://github.com/KejPi/AbracaDABra/files/12580902/2023-09-12_005137_10D.zip)

```
FICDecoder: EId 0x9200: ensemble label 'SLO DAB+ R1' ('SLO R1')
FICDecoder: SId 0x9203: programme service label 'ARS' ('ARS')
FICDecoder: SId 0x9201: programme service label 'PRVI' ('PRVI')
FICDecoder: SId 0x9202: programme service label 'VAL 202' (' 202')
FICDecoder: SId 0x933A: programme service label 'Radio Si' ('Radio Si')
FICDecoder: SId 0x9414: programme service label 'Center Slo' ('Center S')
FICDecoder: SId 0x9516: programme service label 'Rock Radio Slo' ('Rock S')
FICDecoder: SId 0x932F: programme service label 'Radio Ognjisce' ('Ognjisce')
FICDecoder: SId 0x9442: programme service label 'Radio 1 DAB' (' 1 DAB')
FICDecoder: SId 0x9459: programme service label 'Radio 1 80-a' ('Ra 1 80')
FICDecoder: SId 0x9415: programme service label 'Radio City' (' City')
FICDecoder: SId 0x9440: programme service label 'Radio Student' ('Student')
FICDecoder: SId 0x9463: programme service label 'Radio Salomon' (' Salomon')
FICDecoder: SId 0x9449: programme service label 'Radio Veseljak' ('Veseljak')
FICDecoder: SId 0x942B: programme service label 'Radio Net FM' ('Net FM')
FICDecoder: SId 0x9462: programme service label 'Ekspres Slo' ('Eksp Slo')
FICDecoder: SId 0x941C: programme service label 'Radio Aktual' (' Aktual')
```
Other tools (here: dablin_gtk) show that, on the other hand.

[image]

KejPi, 2023-09-12:
Yes, I am trimming both service name and short name. Honestly I do not like that they are trying to get prominent position in service list by starting service/short name with space. Since most of the information is displayed using html technology keeping space at the beginning of short name will probably have no visible impact. Is it important to show that show name starts with space?

andimik, 2023-09-12:
In Ensemble Information window it makes sense **not** to trim it, as you already quote it. Plan is to export this to FMLIST or similar.

KejPi, 2023-11-10:
This is what is actually transmitted for VAL 202 service:

```
	0x9202	'VAL 202         ' [ '1E00' ]   PTY: 09  (static) | charset 0x00 | NumComponents 1 
		AudioComponent (primary), SCIdS: 0, Label: 'VAL 202         ' ['1E00'], ASCTy: 0x3F, 
			SubChId: 1, Language: 38, StartCU: 96, NumCU: 102, EEP 3-A, Datarate: 136kbps, NumUserApps: 1
				UserApp 1/1: UAType: 0x2 [0C3C], X-PAD data: Type=12, DG=0, DSCTy=0x3C, Label: ''

```

1E00 is bitfield that means `' 202'` 

So why to keep leading space and remove trailing space? I am not sure about this issue.

andimik, 2023-11-10:
They have changed the labels, btw. 

The leading space is now only for 4 services.

```
Nov 10 21:33:44 FICDecoder: EId 0x9200: ensemble label 'SLO DAB+ R1' ('SLO R1')
Nov 10 21:33:44 FICDecoder: SId 0x9203: programme service label 'ARS' ('ARS')
Nov 10 21:33:44 FICDecoder: SId 0x9201: programme service label 'PRVI' ('PRVI')
Nov 10 21:33:44 FICDecoder: SId 0x9202: programme service label 'VAL 202' ('VAL 202')
Nov 10 21:33:44 FICDecoder: SId 0x933A: programme service label 'Radio Si' ('Radio Si')
Nov 10 21:33:44 FICDecoder: SId 0x9414: programme service label 'Center Slo' ('Center S')
Nov 10 21:33:44 FICDecoder: SId 0x9516: programme service label 'Rock Radio Slo' ('Rock S')
Nov 10 21:33:45 FICDecoder: SId 0x932F: programme service label 'Radio Ognjisce' ('Ognjisce')
Nov 10 21:33:45 FICDecoder: SId 0x9442: programme service label 'Radio 1 DAB' (' 1 DAB')
Nov 10 21:33:45 FICDecoder: SId 0x9459: programme service label 'Radio 1 80-a' ('Ra 1 80')
Nov 10 21:33:45 FICDecoder: SId 0x9415: programme service label 'Radio City' (' City')
Nov 10 21:33:45 FICDecoder: SId 0x9440: programme service label 'Radio Student' ('Student')
Nov 10 21:33:45 FICDecoder: SId 0x9463: programme service label 'Radio Salomon' (' Salomon')
Nov 10 21:33:45 FICDecoder: SId 0x9449: programme service label 'Radio Veseljak' ('Veseljak')
Nov 10 21:33:45 FICDecoder: SId 0x942B: programme service label 'Radio Net FM' ('Net FM')
Nov 10 21:33:45 FICDecoder: SId 0x9462: programme service label 'Ekspres Slo' ('Eksp Slo')
Nov 10 21:33:45 FICDecoder: SId 0x941C: programme service label 'Radio Aktual' (' Aktual')
```

KejPi, 2023-11-10:
The question is valid. I have tried to remove trimming but it will not help in ensemble view. I would have remove formatting and  display it in fixed font to visualise all spaces as they are (eg. trailing spaces in service label)

KejPi, 2023-11-10:
This is how it looks like when service labels are not trimmed and `<pre>` tag is used, I do not like it :-(

[image]

andimik, 2023-11-11:
> <pre>

If this is HTML, why not replace each space with `&nbsp;`?

KejPi, 2023-11-11:
This is exactly what I do not want to do.

andimik, 2024-05-16:
BTW

If the leading space is in the **long** label, then you see it in AbracaDABra.

https://www.rundfunkforum.de/download/file.php?id=18102&mode=view

KejPi, 2024-05-16:
Well, that surprises me too :-D 
But looking at the related commit, it seem that I have changed the implementation in the way that I am removint trailing spaces not leading. But it does not look good in the service list to be honest :-(