KejPi/AbracaDABra #90: Station logo source

bostjane, 2023-09-12:
After investigating why some stations show logo and others do not I discovered a mess that is currently in DAB+ world. Some parties interpret station logo is in the DAB SPI, while others that it is in the DAB SlideShow. The result is that I see station logo in my car (Ford Sync 3) just for the opposite stations that SDR SW shows, including AbracaDABra.

My request would be if you could show logos from both sources, DAB SPI and also DAB Slideshow, respectively. Therefore we could determine where particular station puts logo and see it in any case

andimik, 2023-09-12:
I disagree.

What is the benefit? If you want to see SPI logos, please use Qirx.

And note, that reading the complete SPI content takes currently much much longer (depending on the multiplex) than just waiting for the next SLS picture.

How long does it take to display the logo in your car? If you see it immediately right after tuning, then the source is not SPI, but local cache or database (like in my Volkswagen Passat), which is even maintained externally.

KejPi, 2023-09-12:
I am currently implementing station logos support and it should be available in couple of weeks. 
There is only one "pure" DAB source for station logos, it is SPI application. Whatever comes from SLS application is slide with any visual contents. Nonetheless SPI can also deliver logo in format suitable to display as slide (320x240) and on top of that there are also internet sources.
What car radio concerns, they typically have local logo database and they use it to display station logo for better user experience. For example, when you select station, no slide is available, thus it makes sense to display station logo instead of any generic placeholder. As Andi writes, SPI application takes long time to decode but whatever it decodes, can be also stored in local cache.
AbracaDABra is going to support logos from SPI and from internet and it is going to show them as small logo and SLS logo if available.

bostjane, 2023-09-12:
Although I'm a programmer and technical enthusiast, I'm quite new to DAB, especially in deep technical understanding. I started digging around, because only 1/16 stations showed logo in my Ford Sync 3 infotainment and I wanted to understand why is that so.

By cross-comparison results in car and various SDR PC software and digging internet I found out that Sync 3 also has internal logo database and is shows logo of exactly the one station that SDR PC software don't. I don't know if that logo is from internal database or received over air. That is what I'm searching further.

I agree with all written by KeyPi. Also having one SW that could also serve as debugging tool is also welcome.

One thing is sure so far. Apparently Slovenia's multiplex operator (and other probably also) send stations logo as SLS and not SPI as they should. If you say it takes much longer to receive it that might be the reason - although it's wrong and receiver's manufacturers that play by the book don't get it. But on the other hand receiver's manufacturers should take approach as suggested by KeyPi, take SPI or local database first, then replace it by SLS if/when arrives.
Even though SPI receiving takes long long time it is still better than nothing. Once received, car could cache it indefinitelly.

@andimik : I'm not sure how long it took to display logo in car for the first time. Now is surely displays it immediately but that could be because it is cached or accessed in local database.

Does anyone knows how/where car manufacturers get data for their static logo database? radiodns.org?

KejPi, 2023-09-12:
They get it from freely available sources, RadioDNS is a good source. By the way, AbracaDABra is going to utilize RadioDNS too. 
Sending station logo as SLS is not against the standard, they can practically provide any visual contents in SLS, if it is static logo, then it is their choice. From receiver point of view it is just some slide without special meaning.

bostjane, 2023-09-12:
Great. Please consider showing info where logo came from, radiodns, SPI, SLS, etc,.. Or give possibility to choose between either of those plus combined option (best effort) It'll be great for debugging.

I understand that having logo in SLS is not strictly against standard, but it is not logical and produces some problems in receiver's implementation. Of course that is if all content within SLS is the same and all slides are same with no tags like this one is logo, this one is album, etc. (I don't know that).
In case all slides are the same receiver's manufacturers have dilemma when implementing GUI. They can't simply cache last SLS content as logo and show it in list of saved stations or show it immediately when selecting particular station. Last content might be album of last song and therefore it's nonsense to show it in the list of saved stations or when recalling some station some time later.
Bottom line, logo in SLS if fine in some situations, but much better in SPI - at least as far as I understand it so far.

KejPi, 2023-09-12:
Receivers typically do not consider SLS being anything special, they do not use it as station logo. So there is no additional complexity.
I have no plan to show where the logo comes from, I will only implement a possibility to disable RadioDNS and maybe also to disable SPI. SPI and RadioDNS are defined as hybrid radio here: https://www.etsi.org/deliver/etsi_ts/102800_102899/102818/03.03.01_60/ts_102818v030301p.pdf

andimik, 2023-09-12:
Dear Boštjan, 

I know Slovenian DAB very well since its beginning in 2016 (I even was invited to the kick-off-party on transmitter Nanos).

> Apparently Slovenia's multiplex operator (and other probably also) send stations logo as SLS and not SPI as they should.

No this is not true, as some days ago broadcaster RTV Slovenia has introduced all logos for Prvi, VAL 202, Ars and Radio Slovenia International (but not for the other programms) via SPI. In the past years, only one single logo for Prvi was transmitted, instead of 4 variants as usual.

[image]

Croatia, Austria (both multiplexes) are using that, too (and are up-to-date).

And please note, that lots of ensembles try to avoid png-files for SLS as there is buggy consumer hardware which cannot display that filetype.

KejPi, 2023-09-13:
Loading time of SPI application depends on data rate of the subchannel (or X-PAD) and size of the MOT directory. When the directory is small and dedicated data service is used for SPI, it can take less than 1 minute. If the directory is huge (meaning many logo pictures, many XML files) it can take several minutes to load. Of course bad raception quality increases load time too. 
SPI application does not require RadioDNS, RadioDNS is and additional source of station logos. 

Coudl you please share recording of the ensemble with SPI logos? Longer better, I would like to use it for testing, I have not SPI application on air here :-(

bostjane, 2023-09-14:
Sure, I'll do it in the afternoon when I return from work. Just give me instructions what to do. Is it logging inside AbracaDABra?

KejPi, 2023-09-14:
Yes, you can record it in AbracaDABra.
Firstly, please check Settings -> Others -> Raw Data Recording -> Include XML header (this is default)
Then just pick any service from the ensemble with SPI to tune it and then go to Menu -> Ensemble information -> Record raw data (bottom right). Chose the file and recording starts. Ensemble dialog will show you the length of recording and file size and you can stop it there too. The files are huge, one minute about 240MB.  I would need at least 2 minutes if possible. Resulting file has .uff extension. If you do not have simple possibility to share the file using some could services, we can discuss it privately by mail.

bostjane, 2023-09-14:
Here is log for 3 minute of Val 202, R1 (https://we.tl/t-dJWrJGYDSA). It should contain SPI, at least if they really introduced it days ago as andimik suggested. I noticed logo change, so SLS is probably there.

KejPi, 2023-09-14:
Thanks a lot for sharing the recording however it is not long enough to receive complete SPI MOT directory. So from this you can see how long it can take to receive complete SPI application. Nonetheless some of the service support RadioDNS :-)

[image]

bostjane, 2023-09-14:
Can you see how much SPI was received so I'll repeat with enough time?

KejPi, 2023-09-14:
The MOT directory has 17 objects in carousel, I was able to receive 12. The calculation is of time needed is not simple, it is just matter of luck when you start the reception / recording. Maybe one more minute would be enough, it is difficult to say.

bostjane, 2023-09-15:
Now it's more than 6 min, hopefully enough: https://we.tl/t-cH0nMwfAkx

KejPi, 2023-09-15:
Yes, enough, thank you! It takes about 250 seconds to receive the complete MOT directory that actually contains logos for four services: 9201, 9202, 9203, 933A

closed 2023-09-18 by bostjane

andimik, 2023-12-12:
Don't want to open a new issue, but I've checked various satellite DAB muxes. Therefore, I have sent them to a fifo files

```
dvbstream -o - 8192 | [mux parameters] | odr-dabmod -f stdin.raw  -F u8
```

Mux | SPI working | files | Comment
--|--|--|--
WDR 9A (not on air yet) | yes | 129 | 
WDR 11D | yes | maybe the same | 
Bundesmux 2 | yes | not shown in log | 
Bundesmux 1 | yes | not shown in log |
ORS Austria | yes | 65 | terrestrial 
R1 Slovenia | yes | forgot | terrestrial
R2W Slovenia | only RadioDNS transmitted | 0 | terrestrial
R2E Slovenia | only RadioDNS transmitted | 0 | terrestrial
EuroDAB Italia | yes | 93 | will not load more than 68 out of 93 files
DAB Italia | yes | 68 | 
D1 National UK | no SPI transmitted | | 
SDL National UK | no SPI transmitted | | 
M1 France | no SPI transmitted | | 
M2 France | no SPI transmitted | | 
RAI Italy | no SPI transmitted | | 
Riks Norway | yes | 81 and 100 | used old file from 2022
NRK 1 Norway | yes | 206 | used old file from 2022
RadioDIG Umbria Italy | no | ? | used file from summer

# Remarks: 

For Umbria the log says,

```
RadioControl: Found user application type 'SPI' within ensemble in data service SId E0500001, SCIdS 0. Starting the service...
```

but nothing will happen. According to digitalbitrate the SPI contains logos. If you want, I can share the eti file.

EuroDAB will work, althought there is a message `19:21:12.228 [I] SPIApp: 0: MOT directory NOT complete (decoded 68 / 93)` and 10 folders with 20 pngs saved.

Riks will execute both SPI streams :-) and show the following log

```
19:38:38.869 [I] SPIApp: 1: MOT directory NOT complete (decoded 39 / 100)
19:38:39.244 [I] SPIApp: 1: MOT directory NOT complete (decoded 39 / 100)
19:38:50.050 [I] SPIApp: 1: MOT directory NOT complete (decoded 49 / 100)
19:38:50.466 [I] SPIApp: 1: MOT directory NOT complete (decoded 49 / 100)
19:39:00.473 [I] SPIApp: 1: MOT directory NOT complete (decoded 59 / 100)
19:39:00.842 [I] SPIApp: 1: MOT directory NOT complete (decoded 59 / 100)
19:39:07.666 [I] SPIApp: 0: MOT directory NOT complete (decoded 64 / 81)
19:39:10.449 [I] SPIApp: 1: MOT directory NOT complete (decoded 69 / 100)
19:39:10.844 [I] SPIApp: 1: MOT directory NOT complete (decoded 69 / 100)
19:39:21.269 [I] SPIApp: 1: MOT directory NOT complete (decoded 79 / 100)
19:39:21.271 [I] SPIApp: 1: MOT directory NOT complete (decoded 79 / 100)
19:39:31.672 [I] SPIApp: 1: MOT directory NOT complete (decoded 89 / 100)
19:39:31.674 [I] SPIApp: 1: MOT directory NOT complete (decoded 89 / 100)
```

Hint: The German rundfunkforum does not allow Markdown tables, but it was easy to write them here.

KejPi, 2023-12-12:
Interesting experiment :-) My comments:
**Umbria** - please share, there can be still some bugs in application 
**EuroDAB** - yes, this is a change in last version, that application processes incomplete directory, I am working on further improvement
**Riks** - In theory, the application shall handle up to 3 data subchannels each with more data services (packat addresses) in parallel, but I do not have any testcase to verify it.

andimik, 2023-12-12:
ok, will send you Umbria in the next days as ETI.

andimik, 2023-12-13:
http://andimik.bplaced.net/temp/dab/20230715_umbria.zip

KejPi, 2023-12-13:
Thank you! It seems to be a very nice case because FEC scheme is not applied for SPI data service. This configuration is discouraged be standard but still valid. It should be supported by library but it does not seem to work.

andimik, 2023-12-13:
No, Qt-DAB does not detect it...

Maybe the file was from a earlier stage and the content was not working smoothly like you see it now in digitalbitrate.com