KejPi/AbracaDABra #91: Audio recording capability

KejPi, 2023-09-22:


andimik, 2023-11-08:
Tested Recording with DAB+. 

AAC frames are ok.
Path is localized name for Music :-)

Hint: thanks for escaping the annoying `*` character which is found in some service labels in lots of countries ...

`AudioRecorder: Recording file: "/home/andreas/Musik/2023-11-08-230057_E0AC47__ 88.6 _.aac"`

[image]

KejPi, 2023-11-09:
You can change the recording folder in settings. There are actually more difficult characters, I hope I am removing all that may cause problems.

andimik, 2023-11-09:
> You can change the recording folder in settings. 

I know as I was translating that as well :-)

closed 2023-11-23 by KejPi