KejPi/AbracaDABra #92: Show real service during Alarm and Alarm test

andimik, 2023-10-20:
I've realized that during an active Alarm or Alarm test announcement the audio switches to that service, but you cannot see it in the window. The only index is the bitrate and protection level, which is **not** that from the displayed service, but from the announcement service.

In this window I expect switching to the following SubCh 

```
15:35:06.803 [I] RadioControl: "Alarm Test" announcement in subchannel 10 cluster ID 254
```

[image]

KejPi, 2023-10-20:
Showing the original service name is done by purpose and it is expected behavior of the receiver. If technical info about bitrate etc. does not match Radio Horeb it is probably a bug and I will fix it. It woudl be helpful if you could share the recording for testing. 
PS: some infomation about source of the announcement is in tooltip of the announcement icon.

KejPi, 2023-10-25:
What about this? Parameters are in disabled state but showing original service information during announcement from other service. 

[image]

andimik, 2023-11-10:
I realized that there is a tool tip saying the service. I guess it can be closed.

Tested with 🇩🇪 Munich 10D (other 📢 announcements)

closed 2023-11-10 by andimik