KejPi/AbracaDABra #95: Unable to access BBC channels

DavidGoodenough, 2023-11-02:
I am using welle.io which receives BBC channels correctly with my sdrrtl setup but which drops out on the hour.

I raised this with the welle.io team and they that the problem is that BBC use reconfigure which they do not support.  They recommended I use abracadabra.

So I downloaded it and ran it up but none of the BBC stations appear on the channel list.

Any ideas?

KejPi, 2023-11-02:
This is interesting, I am pretty confident that AbracaDABra can receive anything that you can receive by welle.io. Are you using USB stick directly (it should be set according to screenshot below) or through RTL-TCP server? 
I would be very helpful if you could share some IO recording for analysis. Just switch to Expert UI (Settings -> Other), select desired channel, go to Ensemble -> Record raw data. Longer recording better but 30 sec should be enough. 

[image]
[image]
[image]
[image]

andimik, 2023-11-03:
Do other multiplexes work? Is your signal good enough?

DavidGoodenough, 2023-11-03:
Well I am not quite sure how, but going into expert mode seems to have cured it, and suddenly the BBC National DAB appeared along with its channels.  

I did record the ensemble information, it is attached (I had to zip it, github does not like raw uff files).
[uff.zip](https://github.com/KejPi/AbracaDABra/files/13249934/uff.zip)

andimik, 2023-11-03:
Can you please repeat it (delete the service list and use the standard gui)? Maybe the Mux can really be found in expert mode??

DavidGoodenough, 2023-11-03:
I turned off expert mode and cleared the list and then did a bank scan.  It found the BBC stations but there is no sound when I select a station on the BBC set (12B) nothing happens and the ensemble information is filled in a the top with the right frequency etc but no information in the lower box.  

I then selected one on another set (the Swindon set as it happens) and that worked and then I selected BBC Radio 3 and it worked.  It is very odd.

I do need a better antenna I think, I am only get an SNR of 4.6db

andimik, 2023-11-03:
This is really too less.

KejPi, 2023-11-03:
Expert mode should not have any influence on reception quality - it only changes UI. What OS do you use? Are you using 2.2.4 binary downloaded from GitHub or did you compile by yourself?  And what HW do you use? Are you using RTL SDR and is it configured as on screenshot I shared yesterday? 

In general, if band scan finds the ensemble then it should be possible to receive it with reasonable quality. From your description it seems to be some problem with AGC (tuning elewhere and going back gives different result).

DavidGoodenough, 2023-11-03:
Changing the Gain Control to Software raised the SNR to 9.6db, which is much cleaner audio quality.

I am running on Debian sid, using a git clone of the current Github repository which I compiled myself.

I have a very old Terratec R820T USB dongle.

KejPi, 2023-11-03:
In your configuration (self compiled on Debian), Software gain control will be probably the only working option. For sure HW AGC will not work unless you linked against olddab's rtlsdr. 
Can I close the issue or is there still any open point or doubt?

DavidGoodenough, 2023-11-03:
It can probably be closed.  It might be worth documenting that Software gain control is the only reasonable option for self compiled (on Debian if relevant).

andimik, 2023-11-03:
On Ubuntu I am using software gain control, although I am compiling myself. So it might be Debian related.

KejPi, 2023-11-03:
I think that SW AGC is still default option in the application. HW AGC may work if rtlsdr driver handles it correctly like rtlsdr modified by olddab does. Debian is known to use old software versions so the problem might be might be more severe. I am closing the issue but feel free to open new in case you face another one.

closed 2023-11-03 by KejPi