KejPi/AbracaDABra #98: ECC detection leads to wrong ensemble information window

andimik, 2023-11-08:
If ECC is not correctly detected, the ensemble information might be printed twice.

```
Ensemble:
0x9400 SLO DAB+ R2 W [ SLO R2 W ] ECC: 0xE4, UTC 60 min, INT: 1, alarm announcements: 0
Services (24):
0x9409 Radio 94 [ Radio 94 ] ECC: 0x00, Country: Unknown, PTy: Varied (static), Announcements: No
0x940B Radio Antena [ Antena ] ECC: 0x00, Country: Unknown, PTy: Pop Music (static), Announcements: No
0x9411 Capodistria [ Capodist ] ECC: 0x00, Country: Unknown, PTy: Information (static), Announcements: No
0x9412 Radio Capris [ Capris ] ECC: 0x00, Country: Unknown, PTy: Pop Music (static), Announcements: No
0x9421 Radio Koper [ Koper ] ECC: 0x00, Country: Unknown, PTy: Varied (static), Announcements: No
0x9422 Radio Kranj [ Kranj ] ECC: 0x00, Country: Unknown, PTy: Varied (static), Announcements: No
0x9436 Radio Robin [ Robin ] ECC: 0x00, Country: Unknown, PTy: Other Music (static), Announcements: No
0x9438 Best FM [ Best FM ] ECC: 0x00, Country: Unknown, PTy: Other Music (static), Announcements: No
0x943C Radio Sora [ Sora ] ECC: 0x00, Country: Unknown, PTy: Other Music (static), Announcements: No
0x9444 Radio Triglav [ Triglav ] ECC: 0x00, Country: Unknown, PTy: Other Music (static), Announcements: No
0x945B Radio BOB [ BOB ] ECC: 0x00, Country: Unknown, PTy: Rock Music (static), Announcements: No
0x952E Alpski Val [ Alp Val ] ECC: 0x00, Country: Unknown, PTy: Other Music (static), Announcements: No
0x9409 Radio 94 [ Radio 94 ] ECC: 0xE4, Country: Slovenia, PTy: Varied (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio 94' [ 'Radio 94' ], ASCTy: 0x3F (AAC)
      SubChId: 3, Language: Slovene, StartCU: 132, NumCU: 42, EEP 3-A, Coderate: 1/2, Bitrate: 56kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x940B Radio Antena [ Antena ] ECC: 0xE4, Country: Slovenia, PTy: Pop Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Antena' [ 'Antena' ], ASCTy: 0x3F (AAC)
      SubChId: 10, Language: Slovene, StartCU: 456, NumCU: 42, EEP 3-A, Coderate: 1/2, Bitrate: 56kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9411 Capodistria [ Capodist ] ECC: 0xE4, Country: Slovenia, PTy: Information (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Capodistria' [ 'Capodist' ], ASCTy: 0x3F (AAC)
      SubChId: 2, Language: Slovene, StartCU: 66, NumCU: 66, EEP 3-A, Coderate: 1/2, Bitrate: 88kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9412 Radio Capris [ Capris ] ECC: 0xE4, Country: Slovenia, PTy: Pop Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Capris' [ 'Capris' ], ASCTy: 0x3F (AAC)
      SubChId: 9, Language: Slovene, StartCU: 396, NumCU: 60, EEP 3-A, Coderate: 1/2, Bitrate: 80kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9421 Radio Koper [ Koper ] ECC: 0xE4, Country: Slovenia, PTy: Varied (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Koper' [ 'Koper' ], ASCTy: 0x3F (AAC)
      SubChId: 1, Language: Slovene, StartCU: 0, NumCU: 66, EEP 3-A, Coderate: 1/2, Bitrate: 88kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9422 Radio Kranj [ Kranj ] ECC: 0xE4, Country: Slovenia, PTy: Varied (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Kranj' [ 'Kranj' ], ASCTy: 0x3F (AAC)
      SubChId: 12, Language: Slovene, StartCU: 552, NumCU: 42, EEP 3-A, Coderate: 1/2, Bitrate: 56kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9436 Radio Robin [ Robin ] ECC: 0xE4, Country: Slovenia, PTy: Other Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Robin' [ 'Robin' ], ASCTy: 0x3F (AAC)
      SubChId: 7, Language: Slovene, StartCU: 306, NumCU: 48, EEP 3-A, Coderate: 1/2, Bitrate: 64kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9438 Best FM [ Best FM ] ECC: 0xE4, Country: Slovenia, PTy: Other Music (static), Announcements: Traffic News, Cluster IDs [08]
AudioComponent (primary), SCIdS: 0, Label: 'Best FM' [ 'Best FM' ], ASCTy: 0x3F (AAC)
      SubChId: 8, Language: Slovene, StartCU: 354, NumCU: 42, EEP 3-A, Coderate: 1/2, Bitrate: 56kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x943C Radio Sora [ Sora ] ECC: 0xE4, Country: Slovenia, PTy: Other Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Sora' [ 'Sora' ], ASCTy: 0x3F (AAC)
      SubChId: 4, Language: Slovene, StartCU: 174, NumCU: 42, EEP 3-A, Coderate: 1/2, Bitrate: 56kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x9444 Radio Triglav [ Triglav ] ECC: 0xE4, Country: Slovenia, PTy: Other Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio Triglav' [ 'Triglav' ], ASCTy: 0x3F (AAC)
      SubChId: 6, Language: Slovene, StartCU: 264, NumCU: 42, EEP 3-A, Coderate: 1/2, Bitrate: 56kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x945B Radio BOB [ BOB ] ECC: 0xE4, Country: Slovenia, PTy: Rock Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Radio BOB' [ 'BOB' ], ASCTy: 0x3F (AAC)
      SubChId: 11, Language: Slovene, StartCU: 498, NumCU: 54, EEP 3-A, Coderate: 1/2, Bitrate: 72kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x952E Alpski Val [ Alp Val ] ECC: 0xE4, Country: Slovenia, PTy: Other Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Alpski Val' [ 'Alp Val' ], ASCTy: 0x3F (AAC)
      SubChId: 5, Language: Slovene, StartCU: 216, NumCU: 48, EEP 3-A, Coderate: 1/2, Bitrate: 64kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
```

Only workaround is to change the mux and go back, then the window is ok.

KejPi, 2023-11-09:
Coudl you please share some recording so that I can reproduce the issue. I have never seen this problem before.

andimik, 2023-11-09:
I have had it several times and now opened that issue. 

I don't expect the recording will help you, as at the time the recording starts, it might be ok again.

andimik, 2023-11-09:
So my proposal is the reset the content when opening the ensemble information window.

andimik, 2023-11-09:
The only file which *might* help is the debug raw file.

Let me re-build the application (having this debug option activated) and test again. Hope I can catch such an error...

KejPi, 2023-11-09:
I will try to simulate this behavior - in general it is quite unexpected - ensemble without ECC is not considered as valid thus the application shall wait with service list retrieval until it gets ensemble UEID that is valid. Maybe there is some timing issue.

andimik, 2023-11-09:
> timing 

Yes, I guess so.

KejPi, 2023-11-09:
Does it happen only with this ensemble or have you observed it randomly with any?

andimik, 2023-11-10:
Good question, but I only have seen it when reception is difficult.

KejPi, 2023-11-10:
Just to be sure - please confirm that you have:

```
#define RADIO_CONTROL_TEST_MODE 0
```

andimik, 2023-11-10:
yes, line 55 in `radiocontrol.h`. I've not changed it.

KejPi, 2023-11-10:
I have tried to fix it, please check and let me know. I hope I have not created new problem.

closed 2024-10-18 by andimik