KejPi/AbracaDABra #101: libdabsdr.so.2 not found after make install - ldconfig needed

andimik, 2023-11-20:
Seems the `make install` is not working properly, because when I run the app from the build-directory `~/apps/AbracaDABra/build/gui/AbracaDABra` it works fine, while 

```
LANG=C sudo make install
[  5%] Built target AbracaDABra_lrelease
[  7%] Automatic MOC and UIC for target AbracaDABra
[  7%] Built target AbracaDABra_autogen
[100%] Built target AbracaDABra
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/bin/AbracaDABra
-- Up-to-date: /usr/local/share/applications/AbracaDABra.desktop
-- Up-to-date: /usr/local/share/icons/hicolor/512x512/apps/AbracaDABra.png
-- Up-to-date: /usr/local/lib/libdabsdr.so.2.4.0
-- Up-to-date: /usr/local/lib/libdabsdr.so.2
```

will end in

```
$ AbracaDABra 
AbracaDABra: error while loading shared libraries: libdabsdr.so.2: cannot open shared object file: No such file or directory
```

The solution is the following line

```
sudo ldconfig
```

Maybe you can add it to the `README.md`

KejPi, 2023-11-21:
Some Linux distributions do not even have /usr/local/lib in their ldconf. This is generic linux issue, no application specific, all application that install any library in /usr/local have this problem.

closed 2023-11-21 by KejPi

andimik, 2023-11-21:
Ah, ok

KejPi, 2023-11-22:
I have added short note about installation prefix, not sure it helps dummy users, advanced users should know already.

andimik, 2023-11-22:
Better than nothing ...