KejPi/AbracaDABra #106: Audio drops on 32 KHz.  stations

jpuigs, 2023-12-11:
I can not hear correctly one MUX in which all stations are sampled at 32 KHz. 

Log says:

```
23:05:00.718 [I] RtlTcpInput: Trying to connect to: 192.168.23.254:1234
23:05:00.734 [I] RtlTcpInput: Connected to: 192.168.23.254:1234
23:05:00.734 [I] RtlTcpInput: RTLSDR_TUNER_R820T
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] GRAN VIA RADIO     E10912 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] PEPE RADIO         E18930 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] HITS 1 RADIO       E1B014 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] LEBONMIX           E1B124 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] DABclassic         E1EC06 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] GOZADERA FM        E1F771 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] NOSTALGIA FM       E1F772 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] ONIRIA RADIO       E1F774 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] laFRESCA           E1F776 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] LA URBAN RADIO     E1F777 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] BOM RADIO          E1F778 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] EDM RADIO          E1F779 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] M40                E1F783 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] DJ INTERNATIONAL   E1F785 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] LAKALLE            E1F786 : 0
23:05:00.734 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] Pyrénées           E1FEFE : 0
23:05:00.734 [I] ServiceList:           [E2EEF4 @ 188928 kHz |  7A] 1 DAB              E2EE01 : 0
23:05:00.734 [I] ServiceList:           [E2EEF4 @ 188928 kHz |  7A] ESPY DAB           E2EE02 : 0
23:05:00.734 [I] ServiceList:           [E2EEF4 @ 188928 kHz |  7A] NETRADIO           E2EE03 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] DAB+ R MARIA       E2EFED : 0
23:05:00.734 [I] ServiceList:           [E2EFFC @ 218640 kHz | 11B] RNE-SUSPILNE UKR   E2EFEE : 0
23:05:00.734 [I] ServiceList:           [E2EFFC @ 218640 kHz | 11B] RADIO CLASICA      E2EFEF : 0
23:05:00.734 [I] ServiceList:           [E2EFFC @ 218640 kHz | 11B] RADIO 3            E2EFF0 : 0
23:05:00.734 [I] ServiceList:           [E2EFFC @ 218640 kHz | 11B] REE EUROPA         E2EFF1 : 0
23:05:00.734 [I] ServiceList:           [E2EFFC @ 218640 kHz | 11B] MegaStar           E2EFF2 : 0
23:05:00.734 [I] ServiceList:           [E2EFFC @ 218640 kHz | 11B] LOS40 CLASSIC      E2EFF3 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] SER Digital        E2EFF4 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] Onda Cero Radio    E2EFF5 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] KISS FM DIGITAL    E2EFF6 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] RADIO MARIA        E2EFF7 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] Cadena100          E2EFF8 : 0
23:05:00.734 [I] ServiceList:           [E2EFFD @ 195936 kHz |  8A] EUROPA FM          E2EFF9 : 0
23:05:00.734 [I] ServiceList:           [E2EFFE @ 209936 kHz | 10A] Radio 1            E2EFFA : 0
23:05:00.734 [I] ServiceList:           [E2EFFE @ 209936 kHz | 10A] RADIO 5            E2EFFB : 0
23:05:00.734 [I] ServiceList:           [E2EFFE @ 209936 kHz | 10A] COPE DIGITAL       E2EFFC : 0
23:05:00.734 [I] ServiceList:           [E2EFFE @ 209936 kHz | 10A] IntereconomiaDAB   E2EFFD : 0
23:05:00.734 [I] ServiceList:           [E2EFFE @ 209936 kHz | 10A] RADIO MARCA        E2EFFE : 0
23:05:00.734 [I] ServiceList:           [E2EFFE @ 209936 kHz | 10A] El Mundo Radio     E2EFFF : 0
23:05:09.282 [I] RadioControl: Tuning 0.000 -> 223.936 MHz ...
23:05:09.720 [I] RadioControl: Current frequency 223.936 MHz
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] GRAN VIA RADIO     E10912 : 0
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] GOZADERA FM        E1F771 : 0
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] NOSTALGIA FM       E1F772 : 0
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] PEPE RADIO         E18930 : 0
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] ONIRIA RADIO       E1F774 : 0
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] M40                E1F783 : 0
23:05:11.740 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] laFRESCA           E1F776 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] LA URBAN RADIO     E1F777 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] BOM RADIO          E1F778 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] EDM RADIO          E1F779 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] DABclassic         E1EC06 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] LEBONMIX           E1B124 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] Pyrénées           E1FEFE : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] LAKALLE            E1F786 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] HITS 1 RADIO       E1B014 : 0
23:05:11.741 [I] ServiceList:           [E1F409 @ 223936 kHz | 12A] DJ INTERNATIONAL   E1F785 : 0
23:05:11.742 [I] RadioControl: Playing: [E1F409 @ 223936 kHz | 12A] LA URBAN RADIO     E1F777 : 0
23:05:11.744 [I] RadioControl: Starting user application 'SlideShow' from XPAD.
23:05:12.134 [I] AudioDecoder: HE-AAC v2 32 kHz stereo
23:05:12.134 [I] AudioDecoder: Output sample rate 32000 Hz, channels: 2
23:05:12.135 [I] AudioOutput: Audio device "Speakers (Creative SB X-Fi)" [default]
23:05:12.956 [I] RadioControl: Starting user application 'SlideShow' from XPAD.
23:05:12.976 [I] AudioOutput: Unmuting audio
23:05:28.987 [I] AudioDecoder: Muting audio (decoding errors)
23:05:38.538 [I] AudioDecoder: Umuting audio
23:05:41.908 [I] AudioDecoder: Muting audio (decoding errors)
23:05:42.018 [I] AudioDecoder: Umuting audio
23:05:43.108 [I] AudioDecoder: Muting audio (decoding errors)
23:05:43.218 [I] AudioDecoder: Umuting audio
23:05:43.468 [I] AudioDecoder: Muting audio (decoding errors)
23:05:43.578 [I] AudioDecoder: Umuting audio
23:05:44.187 [I] AudioDecoder: Muting audio (decoding errors)
23:05:44.298 [I] AudioDecoder: Umuting audio
23:05:44.578 [I] AudioDecoder: Muting audio (decoding errors)
23:05:44.658 [I] AudioDecoder: Umuting audio
23:05:45.033 [I] AudioDecoder: Muting audio (decoding errors)
23:05:45.138 [I] AudioDecoder: Umuting audio
23:05:45.746 [I] AudioDecoder: Muting audio (decoding errors)
23:05:45.859 [I] AudioDecoder: Umuting audio
23:05:45.986 [I] AudioDecoder: Muting audio (decoding errors)
23:05:46.100 [I] AudioDecoder: Umuting audio
23:05:48.025 [I] AudioDecoder: Muting audio (decoding errors)
23:05:48.139 [I] AudioDecoder: Umuting audio
23:06:13.110 [I] AudioDecoder: Muting audio (decoding errors)
23:06:13.220 [I] AudioDecoder: Umuting audio
23:06:14.010 [I] AudioDecoder: Muting audio (decoding errors)
23:06:14.121 [I] AudioDecoder: Umuting audio
23:06:17.190 [I] AudioDecoder: Muting audio (decoding errors)
23:06:17.300 [I] AudioDecoder: Umuting audio
23:06:18.950 [W] MOTObject: Removing duplicate header parameter 5
23:06:25.651 [I] AudioDecoder: Muting audio (decoding errors)
23:06:25.822 [I] AudioDecoder: Umuting audio

```

jpuigs, 2023-12-11:
Just fixed !
I don't delete post, because It could help anybody who may have same problem.
Soundblaster output settings were set to 48 KHz.
I've set them now to 96 KHz.
I understand that as 96 KHz is multiple (3x) of 32 KHz., it works fine....

jpuigs, 2023-12-11:
Setting SB settings to 96 KHz. instead of 48 KHz., works much better, but still some drops....

andimik, 2023-12-11:
No, I cannot confirm it.

```
23:31:43.395 [I] AudioDecoder: HE-AAC 32 kHz mono
23:31:43.396 [I] AudioDecoder: Output sample rate 32000 Hz, channels: 2
23:31:43.746 [W] SPIApp: DNS lookup failed: "0.9440.9200.9e4.dab.radiodns.org"
23:31:43.809 [I] AudioOutput: Unmuting audio
```
I only have one 32 kHz program (in mono), but there are no dropouts.

KejPi, 2023-12-12:
I also have only 32kHz mono service, could you please share the recording? 
What is your operating system? 
Do you use prebuild binaries or have you built from source? 

**Analysis of the problem:** 
The dropouts are probably caused by decoding errors because the log says: `AudioDecoder: Muting audio (decoding errors)` What is a quality of the signal you are receiving (SNR)? 

There is probably an additional problem with output device because something has changed when you changed sampling rate. Application does not do any sampling rate conversion and fully relies on audio framework. If you use prebuilt binaries and did not change defaults, PortAudio is used (_by the way, this information should be at the beginning of the log and I do not see it in the log you have shared_). You can try for switch to Qt audio framework to make it running @ 48kHz: https://github.com/KejPi/AbracaDABra#expert-settings

jpuigs, 2023-12-12:
I was using windows version 2.2.3 (windows 10 pro 64b), and I upgraded to  2.3.3, but results were the same ones.
signal quality is 15-16 dB.
Since changing sundblaster output to 96 KHz (which is 3 x 32 KHz), it works fine.... only one error in 10 minutes.....

Here you have 2 minutes recording  (470 MB):
https://we.tl/t-dwrqb7cJt0

KejPi, 2023-12-12:
Thank you for the recording, much appreciated. Was the error happening also during this recoding? If yes, the what service did you listen to? 
I am playing the file on Windows using bluetooth headphones and it works without any problem.

What do you see in the log when you hear the dropout? 
I would still recommend to try with Qt Audio framework and 48kHz output rate. It might work better with SoundBlaster. 

And by the way - according to encemble label I would say it is a testing ensemble from Barcelona, am I right? But the country ID is set to France.

jpuigs, 2023-12-12:
During this recording was NO dropouts.
When I hear dropouts, is appears what you saw in first post.

Later I'll try qtaudio.
Yes, yes, you're right, it's from Barcelona and the country ID is wrong !!!

I've noticed that RTL_TCP doesn't start automatically on v. 2.3.2 and 2.3.3.
I have to change the manual gain value +1/-1 to wake it up !
On version 2.2.3 worked fine.

KejPi, 2023-12-12:
It seems that SB does not support 32kHz input - just theory. 
I will check RTL_TCP  manual gain settings, thanks for reporting. 

If the log says the same as in the first post, it is due to decoding errors - this could be caused by reception distortions but even by errors in brodcasting. I have one service here that also has these problems while other services in the ensemble play fine and there are no FIB errors. You can check in other tool if you have these dropouts too.

jpuigs, 2023-12-12:
> If the log says the same as in the first post, it is due to decoding errors 

Yes, same messages.

On radio devices like Technisat Viola 2, and on car receiver, there is no problem with those 32 KHz. services.
So I guess it's SB problem.

Later I'll comment a little bug in _ensemble information_ in another post.

KejPi, 2023-12-12:
Audio decoding problem (log on top) has nothing to do with audio device. It would be interesting to see what is really happening during the dropout.

jpuigs, 2023-12-12:
I can add some info.
When SB is set to 48 KHz. and problems start, it seems audio is still in input buffer, because I close program, and I have still audio during several seconds. I'll try to do a video later.

KejPi, 2023-12-12:
@jpuigs  I have created issue #107  for RTL_TCP problem, please comment there

closed 2024-06-22 by KejPi