KejPi/AbracaDABra #107: RTL_TCP doesn't start automatically on v2.3.x

KejPi, 2023-12-12:
[jpuigs](https://github.com/jpuigs) ported following in Issue #106 : 

I've noticed that RTL_TCP doesn't start automatically on v. 2.3.2 and 2.3.3. using manual option in settings.
Device and software options work fine.

@jpuigs:  could you please be more specific about the behaviour and your expectations? I have tried to reproduce the issue but I did not see anything wrong

jpuigs, 2023-12-15:
Sorry, I was away from home and I couldn't post this video.

In this video you'll see....
You execute version 2.2.4, the one I had for several months, and it works fine using rtl_tcp server.
I have manual gain selection, and you see that as I open program, reception starts correctly.
Then I close v 2.2.4 and open version 2.3.2 , and signal is KO.
Until I move manually gain +1 / -1 , it doesn't start to work.
Then I close again and open version 2.3.3 , and signal is KO too, until I wake it up by moving manual gain....
Now I've seen that version 2.3.4 is published,  I've done tests and it works like 2.3.2 and 2.3.3

https://youtu.be/Hx6ag3bfXW8

jpuigs, 2023-12-15:
I've done another test.
I've downloades version 2.3.0-1, and it doesn't work too.
So last version in which RTL TCP works just starting in manual mode correctly is 2.2.4

KejPi, 2023-12-15:
I cannot reproduce this behaviour using RTL_TCP server running locally. However I have done one change with RTL_TCP gain handling, I will try to remove this modification in next version.
Is the server running on other machine or on the same as application? I would be curious about server messages.

jpuigs, 2023-12-15:
RTL TCP server is on a raspberry model 3B

jpuigs, 2023-12-15:
Here you have logs:

I open server
```

pi@raspberrypi:~ $ rtl_tcp -a 192.168.23.254 -f 188928000 -g 44.5 -P -2
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner error set to -2 ppm.
[R82XX] PLL not locked!
Tuned to 188928000 Hz.
Tuner gain set to 44.500000 dB.
listening...
Use the device argument 'rtl_tcp=192.168.23.254:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

```

I open abracaDABra v 2.2.4 ans it gets signal.

```
client accepted! 192.168.23.2 53430
Allocating 15 zero-copy buffers
set gain mode 1
set tuner gain by index 23
set freq 223936000

```
I close v 2.2.4

```
comm recv bye
Signal caught, exiting!
worker cond timeout
Signal caught, exiting!
all threads dead..
listening...
Use the device argument 'rtl_tcp=192.168.23.254:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

```

I open v 2.3.3

```
client accepted! 192.168.23.2 53431
Allocating 15 zero-copy buffers
set gain mode 1
set gain mode 1
set agc mode 0
set tuner gain by index 23
set freq 223936000
set gain mode 1
set agc mode 0

```
Nothing happens...
When I move manual gain.....

`set tuner gain by index 24
`

It starts working

KejPi, 2023-12-15:
Thanks you for sharing to log. I think I understand why it does not work but I do not understand why it works in my case. I did that change on purpose, I hope that anything else won't be broken.

jpuigs, 2024-01-25:
Yes, it works fine now !.
Thanks.

closed 2024-01-26 by KejPi