KejPi/AbracaDABra #108: libdabsdr distributed as binaries

GabMus, 2023-12-13:
Hello, I wanted to try doing something similar to abracadabra but from a command line application, and I saw that in the lib folder there are a bunch of binary shared object files for libdabsdr. I was wondering why you're distributing those as binaries instead of source? Are these binaries not open source?

KejPi, 2023-12-13:
Hello, these binaries are not open source, but you are free to use them, they are distributed under MIT license.

closed 2023-12-13 by KejPi

GabMus, 2023-12-13:
thanks for the quick answer, but I think the follow up question is: why are they not open source? did you make them? are they using some tainted data that you cannot redistribute?

KejPi, 2023-12-13:
I am developing the library and I have decided not to disclose the source code.

GabMus, 2023-12-13:
That makes things a bit harder for packaging and redistributing, particularly if one wants to support other architectures like riscv or wants to package this for downstream distributions or even flatpak. Can you elaborate on why you're not sharing the library source?

KejPi, 2023-12-13:
I want to keep my know-how. I have developend the library from the scratch, I have spent a lot of time with optimizations of the reception performance and CPU usage and the library is what makes AbracaDABra unique.

GabMus, 2023-12-13:
That's definitely a thing to be proud of, and something I can appreciate, but I don't think closing it down will achieve anything but reduce adoption. If you're afraid of third party companies profiting from your hard work, you can use a more "restrictive" license like GPL3 or AGPL3, this way anyone can use it but they have to share the derived source code (don't take this as legal advice, I'm not a lawyer). Besides if a scummy third party wants to use this they can already do so since the shared object and headers are available.

Another issue is that this way all the work on this library is delegated to you, with no possibility of external contributions or forks. If at some point you get busy with other stuff and cannot develop it anymore it would effectively waste all the work you've put into it, particularly if at some point the DAB standard changes in a way that breaks existing implementations.

I greatly appreciate your work, and the effort you've put in the library and the app, it's evidently a labor of love and it shows, but unless you have some commercial ambitions for it, I think this is ultimately hurting the project overall.

KejPi, 2023-12-13:
I do not have commercial ambitions, that is why it is under MIT that allows anyone to use it as it is without restrictions or payments. Maybe I will change my mind in teh future. Originally I did not even have any ambitions to make the application available to public, I developed it primiarily for myself because I was not satisfied with freely available tools (both open and closed source) and I wanted do develop my own DAB library to understand how it works but then I decided to publish it because others may like it too but it is still primary my toy to play width :-)

GabMus, 2023-12-13:
I personally think that unless you're doing something you're planning to sell and don't want to open source, sharing the source should be the default: in the best case it will make someone else happy, and in the worst case it'll just be out there, no harm done. I really hope you do make up your mind about sharing the source :slightly_smiling_face: