KejPi/AbracaDABra #109: Service Names and Component Names

jpuigs, 2023-12-15:
I've found a small bug.

I noticed that on all home & car receivers I have, some stations long names (16 char) are different from the ones that abracaDABra shows.
On DAB/DAB+ programs, Audio Service has a name, and Audio Component may have or not have a name.
Receivers have to look wether Audio Component has name or not, and if Audio Component has one, they have to show it, and if Audio Component has no name, they have to show the "main" name from Audio Service.

AbracaDABra always shows the Audio Service Name. It should show the component name when it's available.

On AbracaDABra ensemble info, the Service and the Component names are always the same ones, and that's the bug.
Here an example:

[image]

Red (service name) and Orange (Component name) are equal, but they are not  really equal.
If you look at radio receiver, or you look at other PC programs like DAB Player, you'll see that names are different.

Here you have a DAB Player ensemble information, as you'll see, it shows the differences that AbracaDABra doesn't (orange Component name)

[image]

_________________________________________________________________________________________________________________

AbracaDABra:

[image]

Component name is not shown anywhere.

_________________________________________________________________________________________________________________

DAB Player:

[image]

So, please, would it be possible to fix the **ensemble information** to show correct service and component names everywhere ?
And show component name (where available) instead of Service name in main app.

Thanks,

KejPi, 2023-12-15:
Thank you for reporting. This is actually the first time I see the primary service component to have dedicated name. I am glad that I have a test case for that and I will fix it in next version.

KejPi, 2023-12-16:
I was checking the implementation and it is done this way intentionally because 2.1.1 standard from 2017 does not permit to broadcast primary service component name:

[image]

Since this change could have some unwanted side effects, I have decided not to change current implementation at the moment.

jpuigs, 2023-12-16:
OK, I understand that what is shown on the program list is OK, but inside the _**enseble information**_ it should appear.
It should appear what is transmitted....

jpuigs, 2023-12-16:
In UK............

[image]

jpuigs, 2023-12-16:
BBC in the UK, has 2 channels with 2 components.

[image]

If you always look names at service label and not on component label (if exist), you'll be listening to Parliament, and program label will be BBC Radio 4

KejPi, 2023-12-16:
Where this comes from? My quote was taken from  ETSI EN 300 401 V2.1.1 (2017-01) http://www.etsi.org/deliver/etsi_en/300400_300499/300401/02.01.01_60/en_300401v020101p.pdf

Component label is shown, but only for secondary services that shall have label unlike primary services.

jpuigs, 2023-12-16:
OK, Ok, I know you do it correctly.
But, I repeat, although program behaviour is correctly according to ETSI standards, when we look at **_ensemble information_**, we would have to see correct component labels names, where they appear.
So, red and orange labels should appear different, as they are really transmitted different in this station and others,

[image]

Only here, in **_ensemble information_** .

KejPi, 2023-12-16:
I understand your point but at the moment it is not possible without side effects. The decoder is designed to be compliant with latest DAB standard so it does not consider with primary service component labels. It is not a problem to decode it, after all it is label like any other - in fact is is already decoded - the problem is that at the time the service list is read from decoder it may not be available and thus the service list loading would need to be redesigned and would always take longer just to support single case. The other option would be to implement labels update functionality, that is in my TODO list, but with very low priority at the moment.

KejPi, 2023-12-16:
Let's keep this issue open as TODO reminder like #9