KejPi/AbracaDABra #111: Selecting a station causes the RTLSDR device to disconnect

Pelican104, 2024-01-08:
I'm having an odd issue where selecting a station causes the RTLSDR dongle to disconnect. 

Had a similar problem when doing the initial station scan which required changing some USB memory setting??

20:24:09.415 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] YorkMix Radio      E1C1EE : 0
20:24:09.415 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 1        E1C221 : 0
20:24:09.415 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 2        E1C222 : 0
20:24:09.416 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 3        E1C223 : 0
20:24:09.416 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 4        E1C224 : 0
20:24:09.416 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 5 Live   E1C225 : 0
20:24:09.416 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 1Xtra    E1C22A : 0
20:24:09.416 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 6Music   E1C22B : 0
20:24:09.417 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 4Extra   E1C22C : 0
20:24:09.417 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC AsianNetwork   E1C236 : 0
20:24:09.417 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC WorldService   E1C238 : 0
20:24:09.425 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] HITS RADIO UK      E1C2E1 : 0
20:24:09.426 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Capital Yorkshre   E1C370 : 0
20:24:09.426 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Planet Rock        E1C3F8 : 0
20:24:09.426 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute 80s       E1C3F9 : 0
20:24:09.426 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute 90s       E1C3FA : 0
20:24:09.427 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute  Rock     E1C3FB : 0
20:24:09.427 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Scala Radio        E1C3FD : 0
20:24:09.427 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Capital Yorkshre   E1C570 : 0
20:24:09.428 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] HITS RADIO N YKS   E1C6ED : 0
20:24:09.428 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Viking FM          E1C7B0 : 0
20:24:09.428 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Grt Hits N Yorks   E1C7D0 : 0
20:24:09.428 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Grt Hits E Yorks   E1C8C4 : 0
20:24:09.429 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] YO1 Radio          E1C9F0 : 0
20:24:09.429 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Nation Radio       E1CA82 : 0
20:24:09.438 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Smooth UK          E1CACE : 0
20:24:09.439 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Smooth UK          E1CACE : 0
20:24:09.439 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] BBC Radio York     E1CC37 : 0
20:24:09.439 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] BBC Humberside     E1CD34 : 0
20:24:09.440 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Smooth Country     E1CDE9 : 0
20:24:09.440 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] This IsThe Coast   E1CDED : 0
20:24:09.440 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Your Harrogate     E1CEF2 : 0
20:24:09.441 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Fun Kids           E1CFD8 : 0
20:24:09.874 [I] RtlSdrInput: Found 1 devices
20:24:10.331 [I] RtlSdrInput: Opening rtl-sdr device #0: Generic RTL2832U OEM
20:24:10.601 [I] RtlSdrInput: Supported gain values 29
20:24:10.653 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] YorkMix Radio      E1C1EE : 0
20:24:10.653 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 1        E1C221 : 0
20:24:10.653 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 2        E1C222 : 0
20:24:10.653 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 3        E1C223 : 0
20:24:10.653 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 4        E1C224 : 0
20:24:10.654 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 5 Live   E1C225 : 0
20:24:10.654 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 1Xtra    E1C22A : 0
20:24:10.654 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 6Music   E1C22B : 0
20:24:10.654 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC Radio 4Extra   E1C22C : 0
20:24:10.655 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC AsianNetwork   E1C236 : 0
20:24:10.655 [I] ServiceList:           [E1CE15 @ 225648 kHz | 12B] BBC WorldService   E1C238 : 0
20:24:10.655 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] HITS RADIO UK      E1C2E1 : 0
20:24:10.655 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Capital Yorkshre   E1C370 : 0
20:24:10.655 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Planet Rock        E1C3F8 : 0
20:24:10.656 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute 80s       E1C3F9 : 0
20:24:10.656 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute 90s       E1C3FA : 0
20:24:10.656 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute  Rock     E1C3FB : 0
20:24:10.656 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Scala Radio        E1C3FD : 0
20:24:10.657 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Capital Yorkshre   E1C570 : 0
20:24:10.657 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] HITS RADIO N YKS   E1C6ED : 0
20:24:10.657 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Viking FM          E1C7B0 : 0
20:24:10.658 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Grt Hits N Yorks   E1C7D0 : 0
20:24:10.658 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Grt Hits E Yorks   E1C8C4 : 0
20:24:10.658 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] YO1 Radio          E1C9F0 : 0
20:24:10.659 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Nation Radio       E1CA82 : 0
20:24:10.659 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Smooth UK          E1CACE : 0
20:24:10.659 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Smooth UK          E1CACE : 0
20:24:10.659 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] BBC Radio York     E1CC37 : 0
20:24:10.660 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] BBC Humberside     E1CD34 : 0
20:24:10.660 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Smooth Country     E1CDE9 : 0
[... 106 lines cut ...]
20:26:23.199 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Planet Rock        E1C3F8 : 0
20:26:23.199 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute 80s       E1C3F9 : 0
20:26:23.199 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute 90s       E1C3FA : 0
20:26:23.200 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Absolute  Rock     E1C3FB : 0
20:26:23.200 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Scala Radio        E1C3FD : 0
20:26:23.201 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Capital Yorkshre   E1C570 : 0
20:26:23.203 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] HITS RADIO N YKS   E1C6ED : 0
20:26:23.204 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Viking FM          E1C7B0 : 0
20:26:23.204 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Grt Hits N Yorks   E1C7D0 : 0
20:26:23.204 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Grt Hits E Yorks   E1C8C4 : 0
20:26:23.205 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] YO1 Radio          E1C9F0 : 0
20:26:23.205 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Nation Radio       E1CA82 : 0
20:26:23.205 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Smooth UK          E1CACE : 0
20:26:23.205 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Smooth UK          E1CACE : 0
20:26:23.206 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] BBC Radio York     E1CC37 : 0
20:26:23.206 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] BBC Humberside     E1CD34 : 0
20:26:23.206 [I] ServiceList:           [E1C198 @ 215072 kHz | 10D] Smooth Country     E1CDE9 : 0
20:26:23.207 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] This IsThe Coast   E1CDED : 0
20:26:23.207 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Your Harrogate     E1CEF2 : 0
20:26:23.207 [I] ServiceList:           [E1C1B6 @ 213360 kHz | 10C] Fun Kids           E1CFD8 : 0
20:26:23.322 [I] RtlSdrInput: Setting bandwidth 1530 kHz
20:26:34.331 [I] RadioControl: Tuning 0.000 -> 213.360 MHz ...
20:26:34.836 [I] RadioControl: Current frequency 213.360 MHz
**20:26:35.887 [C] RtlSdrInput: Device unplugged.**
20:26:35.922 [I] RadioControl: Tuning 213.360 -> 0.000 MHz ...

Pelican104, 2024-01-08:
This seems to be fixed by permanently increasing the USBFS buffer size as described here [https://github.com/OpenKinect/libfreenect2/issues/807](https://github.com/OpenKinect/libfreenect2/issues/807)

KejPi, 2024-01-08:
This is known issue on Linux. I use 1024MB usually. I don't know how to fix it from application since it has to be set as root. I will add the hint in README because you are not the first facing this problem on Linux.