KejPi/AbracaDABra #112: MOT directory parsing failed

andimik, 2024-01-20:
Since this week, the second national muxes (R2 West and R2 East) in Slovenia have an SPI service, but - other than for my other ensembles here in Southern Austria - no files are written to `~/.cache/AbracaDABra`

The message on the console is this all the time:

```
MOT directory parsing failed
```

[slo_12c_short.zip](https://github.com/KejPi/AbracaDABra/files/13999130/slo_12c_short.zip)

But there are several logo pictures in the SPI, as I could extract them with Qt-DAB, for example

[image]

[image]

[image]

KejPi, 2024-01-21:
Thank you for reporting and for sharing the recording. It is actually very interesting because in this recording FEC scheme is not signalled (FIG 0/14) and actually seems really not to be used. It should not be the case on air today however still permitted by standard.  
I have analysed what is going on and I have found an embarrassing bug - I am surprised it did not appear sooner.

closed 2024-01-21 by KejPi

andimik, 2024-01-21:
ah, cool fix :-)

Well, it seems the SPI service is still under construction by the broadcaster as it only contains files for one program ...