KejPi/AbracaDABra #113: Feature: Timer to start application and do recording

KejPi, 2024-01-25:


andimik, 2024-02-29:
Timer is working, so you could set it to "solved" after you have merged it to the master branch.

I will also update the German translation.

KejPi, 2024-02-29:
Please wait with the transaltions until is it somehow stable to avoid double work. I will probably do some changes, I am not satisfied with current implementation.

andimik, 2024-02-29:
Understood

KejPi, 2024-02-29:
And by the way - current implementation relies on DAB time. That means recording will not start if you lose DAB sync. This is just a warning in case you want to try to schedule recording something that may be important for you ;-)

andimik, 2024-02-29:
Oh, good to know

KejPi, 2024-02-29:
Details: 
The scheduled recording starts at -30 sec by countdown, then service selection at -10 sec and recording starts at -2 sec. Then application waits for end time and stops ongoing recording. 
All the time points are in DAB time (recalculated to local time offset). DAB time shall be updated every second according to a standard but in the application the resolution as roughly 1.5 sec that is why recording starts bit sooner. If DAB time is lost application keeps updating time once a minute so it shoud not happen that recording will never finish.

Do you think that this implementation is a major limitation?

andimik, 2024-02-29:
Even an accuracy of 1 min is fine, as I always cut my recordings with https://github.com/mifi/lossless-cut

KejPi, 2024-02-29:
OK, I will keep the implementation as it is, thanks for feedback! :-)

andimik, 2024-02-29:
Regarding DAB time vs. PC time:

What about syncing it only once per mux, meaning when you connect to the Mux for the first time. This might prevent the sync loss problem you were talking about.

KejPi, 2024-02-29:
FYI: I have checked the DAB time wrt. system time with the ensembles I am receiving and it differs for each ensemble and only one ensemble was aligned (+- 500ms). The difference between ensembles was in the range of -500ms to +3500ms but two week ago I saw one ensemble with more than 1 minute delay. Based on this experiment I think more reliable implementation would be with system time without any consideration od DAB time.

andimik, 2024-02-29:
Yes, good idea. I don't want to rely on the DAB time. Have no idea if there are even ensembles with wrong time on air, but I know at least one DVB-T Slovene multiplex (several hours offset, no chance to use a Linux STB for recording on all multiplexes, as DVB time cannot be switched off) and many many RDS stations with time offset of several years (!).

KejPi, 2024-02-29:
I have just pushed the new implementation, you can try if you want.

KejPi, 2024-03-09:
@andimik Could you please update German translation? I think I will release the app as it is if you do not see any limitation of current schedule implementation. I have never used such feature so I am not sure if it is behaving expected way.

andimik, 2024-03-09:
Done

KejPi, 2024-03-09:
Thank you, I will merge it in the evening and make the official release hopefully.