KejPi/AbracaDABra #115: Feature: Support for APNG slides

KejPi, 2024-01-31:


thib66, 2025-04-27:
hello, to be more explicit we meet an issue with a new station since yesterday in Paris called l'Onde Porteuse
The logo displayed contains an alpha layer with transparency :

[image]

thib66, 2025-04-27:
Hello, sorry for my confusion. I thought that APNG was AlphaPNG. 
Thanks for the information concerning the possibility to change the background colour.

thib66, 2025-04-27:
Here is now the logo of the station after switching background colour to white: 

[image]

andimik, 2025-07-09:
Thanks to Github Pilot I have at least now the possibility to look for animated pngs (APNG) in a folder and subfolder.

In case of AbracaDABra, I just select it from the Download-Folder.

https://github.com/user-attachments/assets/04e81b83-8efb-4342-aba6-b34963443899

```
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <title>APNG Detector</title>
  <style>
    .image-container { display: flex; flex-wrap: wrap; gap: 16px; }
    .img-box { border: 1px solid #ccc; padding: 8px; text-align: center; }
    .img-box img { max-width: 200px; max-height: 200px; display: block; margin-bottom: 8px; }
  </style>
</head>
<body>
  <h2>APNG (Animated PNG) Detector</h2>
  
  <input type="file" id="fileInput" webkitdirectory multiple accept="image/*">
<div id="images"></div>
<script>
  document.getElementById('fileInput').addEventListener('change', function(event) {
    const container = document.getElementById('images');
    container.innerHTML = '';
    Array.from(event.target.files).forEach(file => {
      if (!file.type.match('image.*')) return;
      const reader = new FileReader();
      reader.onload = function(e) {
        const img = document.createElement('img');
        img.src = e.target.result;
        img.title = file.webkitRelativePath || file.name;
        img.style.maxWidth = '200px';
        img.style.margin = '8px';
        container.appendChild(img);
      };
      reader.readAsDataURL(file);
    });
  });
</script>

  
</body>
</html>
```