KejPi/AbracaDABra #116: EPG window empty (xml files received)

andimik, 2024-01-31:
In Slovene R2W there is a new EPG (since today).

[EPG.zip](https://github.com/KejPi/AbracaDABra/files/14114861/EPG.zip)

Although there are EPG data (see zip-file), the window is empty

[image]

Of course, the focus in the main window is the same service which covers the EPG.

KejPi, 2024-01-31:
Interesting, thanks for sharing the recording. Maybe my statement about EPG branch stability in #112 was too optimistic 😉

KejPi, 2024-01-31:
Well, the XML is somehow strange . 

0x9411: The scope is from 30.1. 0:00 (UTC+1) to 31.1 23:30 (it shall be only 1 day) and it actually covers whole day 31.1.
0x9421: The scope is from 30.1. 4:30 (UTC+0)  to 1.2 222:58 (it shall be only 1 day) and it actually covers whole day 31.1. 

I would even say it is invalid but application shows it. I do not know what is happening in your case - I am somehow faking the case using ETI you shared before and current time. For sure it is not the real case.

[image]

andimik, 2024-01-31:
Well, I have re-built from scratch, but still EPG window without any content.

And the console said, that all scopes are <1 day (right?)

```
20:03:03.280 [D] default: Processing object: "20240131_PI_9426"
20:03:03.280 [D] SPIApp: 	Programme Information "20240131_PI_9426"
20:03:03.280 [D] SPIApp: 		ScopeEnd "2024-01-31T00:58:00.000Z"
20:03:03.281 [D] SPIApp: 		ScopeID: "dab:9e4.9300.9426.0"
20:03:03.281 [D] SPIApp: 		ScopeStart "2024-01-30T04:10:00.000Z"
20:03:03.284 [I] SPIApp: 0: MOT directory NOT complete (decoded 4 / 8)
20:03:03.288 [D] default: Service scope ID: "dab:9e4.9300.9426.0"
20:03:06.480 [D] default: Processing object: "20240201_PI_9426"
20:03:06.480 [D] SPIApp: 	Programme Information "20240201_PI_9426"
20:03:06.480 [D] SPIApp: 		ScopeEnd "2024-02-01T00:58:00.000Z"
20:03:06.480 [D] SPIApp: 		ScopeID: "dab:9e4.9300.9426.0"
20:03:06.480 [D] SPIApp: 		ScopeStart "2024-01-31T04:00:00.000Z"
20:03:06.484 [I] SPIApp: 0: MOT directory NOT complete (decoded 5 / 8)
20:03:06.488 [D] default: Service scope ID: "dab:9e4.9300.9426.0"
20:03:09.289 [D] default: Processing object: "20240202_PI_9426"
20:03:09.289 [D] SPIApp: 	Programme Information "20240202_PI_9426"
20:03:09.289 [D] SPIApp: 		ScopeEnd "2024-02-02T00:58:00.000Z"
20:03:09.289 [D] SPIApp: 		ScopeID: "dab:9e4.9300.9426.0"
20:03:09.289 [D] SPIApp: 		ScopeStart "2024-02-01T04:10:00.000Z"
20:03:09.294 [I] SPIApp: 0: MOT directory NOT complete (decoded 6 / 8)
20:03:09.297 [D] default: Service scope ID: "dab:9e4.9300.9426.0"
20:07:34.479 [D] default: Processing object: "MARIBOR_128_128.png"
20:07:34.480 [I] SPIApp: 0: MOT directory NOT complete (decoded 7 / 8)
```

What do you get as default window (without full loaded EPG)?

KejPi, 2024-01-31:
I did the same test on Arch Linux and it works too.

I do not see that in console because I do not have the signal. I am using old signal to have these 2 services in the list and then application loads the XML files you have shared from the cache. The scope start and end in the log you can see is not read from XML but it comes from MOT object parameters. Please share the recording where I can receive at least 1 XML file so that I can analyse what is going on. 

What do you mean by default window?

andimik, 2024-01-31:
And one png is missing:

```
20:11:34.480 [D] default:     "MARIBOR_32_32.png"
20:11:34.480 [D] default:     "MARIBOR_112_32.png"
20:11:34.481 [D] default:     "MARIBOR_128_128.png"
20:11:34.481 [D] default:     "MARIBOR_320_240.png"
20:11:34.481 [D] default:     "20240131_PI_9426"
20:11:34.481 [D] default:     "20240201_PI_9426"
20:11:34.482 [D] default:     "20240202_PI_9426"
```

[image]

How does the epg window look like without epg content? I mean time, today, service list ...

KejPi, 2024-01-31:
It is not so sure, it depends what is in SI XML file. File name has no meaning.

EPG window without data shall not be accessible but it should contain the switches on to at least and maybe something else. In other words it should not be empty. 

BTW: when you start the app, it should load the XML from the cache and then you should get see some EPG like i do.

andimik, 2024-01-31:
BTW: Also the broadcaster made an error, the SPI logo for Alpski Val is that from another station (Radio Triglav) ...

[image]

andimik, 2024-01-31:
What about these error messages:

```
20:39:28.758 [D] default: void MetadataManager::onValidEpgTime()
20:39:28.762 [D] default: Loading: "/home/andreas/.cache/AbracaDABra/EPG/20240130_e49421.0_PI.xml"
20:39:28.766 [D] default: Service scope ID: "dab:9e4.9400.9421.0"
20:39:28.775 [D] default: Loading: "/home/andreas/.cache/AbracaDABra/EPG/20240130_e49411.0_PI.xml"
20:39:28.778 [D] default: Service scope ID: "dab:9e4.9400.9411.0"
20:39:28.797 [D] default: Loading: "/home/andreas/.cache/AbracaDABra/EPG/20240130_e49426.0_PI.xml"
20:39:28.802 [D] default: Service scope ID: "dab:9e4.9300.9426.0"
20:39:28.805 [D] default: Loading: "/home/andreas/.cache/AbracaDABra/EPG/20240131_e49426.0_PI.xml"
20:39:28.809 [D] default: Service scope ID: "dab:9e4.9300.9426.0"
20:39:28.811 [D] default: Loading: "/home/andreas/.cache/AbracaDABra/EPG/20240201_e49426.0_PI.xml"
20:39:28.814 [D] default: Service scope ID: "dab:9e4.9300.9426.0"
```

```
20:39:26.249 [W] default: qrc:/ProgrammeGuide/epg.qml:28:1: module "QtQuick.Controls" plugin "qtquickcontrols2plugin" not found 
     import QtQuick.Controls 
     ^
```

andimik, 2024-01-31:
```
sudo apt-get install qml6-module-qtquick-controls
```
and rebuilding did not help, btw.

KejPi, 2024-01-31:
That error message would explain the empty window, I do not understand why it is working for other cases. The module is correct and you would need more modules like this - check error messages.

andimik, 2024-01-31:
I finally solved it.

```
sudo apt-get install qml6-module-qtquick-*
```

And thanks to the debug message I also found 

```
sudo apt-get install libqt6core5compat6
```

https://launchpad.net/ubuntu/+source/qt6-5compat

and

```
sudo apt-get install qml6-module-qtqml-workerscript
```

https://packages.ubuntu.com/jammy/qml6-module-qtqml-workerscript

[image]

Therefore, please add that to the CMakeFiles.txt

andimik, 2024-01-31:
Another comment: please add scoll bars on the right and bottom :-)

[image]

andimik, 2024-01-31:
But you need it for EPG, see

```
21:11:12.287 [W] default: qrc:/ProgrammeGuide/epg.qml: module "QtQml.WorkerScript" is not installed
```

KejPi, 2024-01-31:
Yes, I saw that but I am not using it.

andimik, 2024-01-31:
yes, the xml was not the problem. So I will close the issue.

closed 2024-01-31 by andimik

KejPi, 2024-01-31:
By the way - I can see in your screenshot that you set to show only current ensemble - does it work? I have no possibility to test it :-(

andimik, 2024-01-31:
Yes, then you see the current ensemble only.