KejPi/AbracaDABra #124: Missing include will cause "QItemSelection does not name a type"

andimik, 2024-02-28:
# Foreword

Seems you have a newer Qt6 than me (6.2.4), `cmake` will complain that `Qt6QmlCompilerPlus` is not found.
```
//The directory containing a CMake configuration file for Qt6QmlCompilerPlus.
Qt6QmlCompilerPlus_DIR:PATH=Qt6QmlCompilerPlus_DIR-NOTFOUND
```

Then I found at least a hint on https://github.com/thanegill/nixpkgs/blob/b107159b26ac922e59ec4644d569bb2397e6c365/pkgs/development/tools/qtcreator/qt6.nix#L72 which then won't find the missing `Qt6QmlCompilerPlus`

`cmake .. -DQT_NO_FIND_QMLSC=TRUE`

# Problem

But, I still could not complile this commit as there was a bug causing an error in `audiorecitemdialog.h:59:35`

https://github.com/KejPi/AbracaDABra/blob/434b53204d48fd251636c932488455937a2a2e23/gui/audiorec/audiorecitemdialog.h#L59

```
[ 24%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGTabButton_qml.cpp
[ 25%] Generating .rcc/qmlcache/AbracaDABra_qml/epg_qml.cpp
[ 26%] Running rcc for resource AbracaDABra_raw_qml_0
[ 27%] Running rcc for resource qmake_ProgrammeGuide
[ 28%] Running rcc for resource translations
Consolidate compiler generated dependencies of target AbracaDABra
[ 29%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o
In file included from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/IMWZC7MZOR/moc_audiorecitemdialog.cpp:10,
                 from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/mocs_compilation.cpp:7:
/home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/IMWZC7MZOR/../../../../gui/audiorec/audiorecitemdialog.h:59:35: error: ‘QItemSelection’ does not name a type
   59 |     void onServiceSelection(const QItemSelection &selection);
      |                                   ^~~~~~~~~~~~~~
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:298: gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:141: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

I could fix it, by adding `#include <QItemSelectionModel>` into that file.

Later I had another issue with `QVariant` which seemed to be missing in several files.

# Try

[abracadabra.patch.txt](https://github.com/KejPi/AbracaDABra/files/14438742/abracadabra.patch.txt)

```
diff --git a/gui/audiorec/audiorecitemdialog.h b/gui/audiorec/audiorecitemdialog.h
index 43c2664..451eabf 100644
--- a/gui/audiorec/audiorecitemdialog.h
+++ b/gui/audiorec/audiorecitemdialog.h
@@ -28,6 +28,7 @@
 #define AUDIORECITEMDIALOG_H
 
 #include <QDialog>
+#include <QItemSelectionModel>
 #include "slmodel.h"
 #include "audiorecscheduleitem.h"
 
diff --git a/gui/audiorec/audiorecschedulemodel.cpp b/gui/audiorec/audiorecschedulemodel.cpp
index dc42aeb..9a853df 100644
--- a/gui/audiorec/audiorecschedulemodel.cpp
+++ b/gui/audiorec/audiorecschedulemodel.cpp
@@ -27,6 +27,7 @@
 #include <QTime>
 #include <QBrush>
 #include <QPixmap>
+#include <QVariant>
 #include "audiorecschedulemodel.h"
 
 AudioRecScheduleModel::AudioRecScheduleModel(QObject *parent)
diff --git a/gui/audiorec/audiorecschedulemodel.h b/gui/audiorec/audiorecschedulemodel.h
index 2469563..1d309e3 100644
--- a/gui/audiorec/audiorecschedulemodel.h
+++ b/gui/audiorec/audiorecschedulemodel.h
@@ -29,6 +29,7 @@
 
 #include <QAbstractTableModel>
 #include <QObject>
+#include <QVariant>
 #include <QList>
 #include "audiorecscheduleitem.h"
 #include "slmodel.h"
diff --git a/gui/metadatamanager.h b/gui/metadatamanager.h
index 006bbad..b39252b 100644
--- a/gui/metadatamanager.h
+++ b/gui/metadatamanager.h
@@ -31,6 +31,7 @@
 #include <QDomDocument>
 #include <QPixmap>
 #include <QHash>
+#include <QVariant>
 #include "servicelist.h"
 #include "epgmodel.h"
 
diff --git a/gui/servicelist.h b/gui/servicelist.h
index 82b5707..00fd7dc 100644
--- a/gui/servicelist.h
+++ b/gui/servicelist.h
@@ -37,7 +37,7 @@
 #include <stdint.h>
 #include <QHash>
 #include <QSettings>
-
+#include <QVariant>
 #include "radiocontrol.h"
 #include "servicelistid.h"
 #include "servicelistitem.h"

```

# ‘const QVariant’ is ambiguous

```
Consolidate compiler generated dependencies of target AbracaDABra
[ 26%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/audiorec/audiorecschedulemodel.cpp.o
/home/andreas/apps/AbracaDABra/gui/audiorec/audiorecschedulemodel.cpp: In member function ‘void AudioRecScheduleModel::save(QSettings&)’:
/home/andreas/apps/AbracaDABra/gui/audiorec/audiorecschedulemodel.cpp:220:62: error: conversion from ‘uint64_t’ {aka ‘long unsigned int’} to ‘const QVariant’ is ambiguous
  220 |         settings.setValue("ServiceId", item.serviceId().value());
      |                                        ~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/qlocale.h:43,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qcalendar.h:46,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qdatetime.h:47,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/QTime:1,
                 from /home/andreas/apps/AbracaDABra/gui/audiorec/audiorecschedulemodel.cpp:27:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:189:5: note: candidate: ‘QVariant::QVariant(float)’
  189 |     QVariant(float f);
      |     ^~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:188:5: note: candidate: ‘QVariant::QVariant(double)’
  188 |     QVariant(double d);
      |     ^~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:187:5: note: candidate: ‘QVariant::QVariant(bool)’
  187 |     QVariant(bool b);
      |     ^~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:186:5: note: candidate: ‘QVariant::QVariant(qulonglong)’
  186 |     QVariant(qulonglong ull);
      |     ^~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:185:5: note: candidate: ‘QVariant::QVariant(qlonglong)’
  185 |     QVariant(qlonglong ll);
      |     ^~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:184:5: note: candidate: ‘QVariant::QVariant(uint)’
  184 |     QVariant(uint ui);
      |     ^~~~~~~~
/usr/include/x86_64-linux-gnu/qt6/QtCore/qvariant.h:183:5: note: candidate: ‘QVariant::QVariant(int)’
  183 |     QVariant(int i);
      |     ^~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/QSettings:1,
                 from /home/andreas/apps/AbracaDABra/gui/servicelist.h:39,
                 from /home/andreas/apps/AbracaDABra/gui/metadatamanager.h:35,
                 from /home/andreas/apps/AbracaDABra/gui/slmodelitem.h:32,
                 from /home/andreas/apps/AbracaDABra/gui/slmodel.h:35,
                 from /home/andreas/apps/AbracaDABra/gui/audiorec/audiorecschedulemodel.h:35,
                 from /home/andreas/apps/AbracaDABra/gui/audiorec/audiorecschedulemodel.cpp:31:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qsettings.h:165:55: note:   initializing argument 2 of ‘void QSettings::setValue(const QString&, const QVariant&)’
  165 |     void setValue(const QString &key, const QVariant &value);
      |                                       ~~~~~~~~~~~~~~~~^~~~~
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:970: gui/CMakeFiles/AbracaDABra.dir/audiorec/audiorecschedulemodel.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:141: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

KejPi, 2024-02-28:
No problem to compile on macOS. I will fix it when I start my Ubuntu machine - I am seriously considering to drop support for Qt < 6.5.

andimik, 2024-02-28:
No, 34a985f22af0c0e5fe6b38595b547763a3cc9dc4 does not compile as well.

```
Consolidate compiler generated dependencies of target AbracaDABra
[ 29%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o
In file included from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/IMWZC7MZOR/moc_audiorecitemdialog.cpp:10,
                 from /home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/mocs_compilation.cpp:7:
/home/andreas/apps/AbracaDABra/build/gui/AbracaDABra_autogen/IMWZC7MZOR/../../../../gui/audiorec/audiorecitemdialog.h:59:35: error: ‘QItemSelection’ does not name a type
   59 |     void onServiceSelection(const QItemSelection &selection);
      |                                   ^~~~~~~~~~~~~~
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:299: gui/CMakeFiles/AbracaDABra.dir/AbracaDABra_autogen/mocs_compilation.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:141: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

KejPi, 2024-02-28:
As I wrote, I will hopefully fix it when I start Ubuntu. This commit fixed compilation under Arch Linux.

andimik, 2024-02-28:
And this small patch fixes another error message

```
diff --git a/gui/audiooutputqt.cpp b/gui/audiooutputqt.cpp
index 82a3bbe..da4119b 100644
--- a/gui/audiooutputqt.cpp
+++ b/gui/audiooutputqt.cpp
@@ -341,7 +341,7 @@ qint64 AudioIODevice::readData(char *data, qint64 len)
         }
         else
         {   // not enough samples ==> inserting silence
-            qCDebug(audioOutput, "Muted: Inserting silence [%llu ms]", bytesToRead / (m_bytesPerFrame * m_sampleRate_kHz));
+            qCDebug(audioOutput, "Muted: Inserting silence [%lu ms]", bytesToRead / (m_bytesPerFrame * m_sampleRate_kHz));
 
             memset(data, 0, bytesToRead);
 
@@ -480,7 +480,7 @@ qint64 AudioIODevice::readData(char *data, qint64 len)
     }
     else
     {   // mute can be requested when there is not enough samples or from HMI
-        qCInfo(audioOutput, "Muting... [available %llu samples]", numSamples);
+        qCInfo(audioOutput, "Muting... [available %lu samples]", numSamples);
         if (numSamples < AUDIOOUTPUT_FADE_TIME_MS*m_sampleRate_kHz)
         {
             float gain = 1.0;
```
which was mentioned in:

```
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/QLoggingCategory:1,
                 from /home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp:28:
/home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp: In member function ‘virtual qint64 AudioIODevice::readData(char*, qint64)’:
/home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp:344:34: warning: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘uint64_t’ {aka ‘long unsigned int’} [-Wformat=]
  344 |             qCDebug(audioOutput, "Muted: Inserting silence [%llu ms]", bytesToRead / (m_bytesPerFrame * m_sampleRate_kHz));
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                    |
      |                                                                                    uint64_t {aka long unsigned int}
/home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp:344:64: note: format string is defined here
  344 |             qCDebug(audioOutput, "Muted: Inserting silence [%llu ms]", bytesToRead / (m_bytesPerFrame * m_sampleRate_kHz));
      |                                                             ~~~^
      |                                                                |
      |                                                                long long unsigned int
      |                                                             %lu
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/QLoggingCategory:1,
                 from /home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp:28:
/home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp:483:29: warning: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 3 has type ‘uint64_t’ {aka ‘long unsigned int’} [-Wformat=]
  483 |         qCInfo(audioOutput, "Muting... [available %llu samples]", numSamples);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~
      |                                                                   |
      |                                                                   uint64_t {aka long unsigned int}
/home/andreas/apps/AbracaDABra/gui/audiooutputqt.cpp:483:54: note: format string is defined here
  483 |         qCInfo(audioOutput, "Muting... [available %llu samples]", numSamples);
      |                                                   ~~~^
      |                                                      |
      |                                                      long long unsigned int
      |                                                   %lu
```

andimik, 2024-02-29:
https://github.com/KejPi/AbracaDABra/blob/379c7474e775c97a5017e31454e18aedce500e3e/gui/setupdialog.cpp#L1151

Here is an example how you could solve this, too.

KejPi, 2024-02-29:
It is only minor warning under Linux that does not bother me.

andimik, 2024-03-01:
Now it looks better:

```
diff --git a/gui/audiooutputqt.cpp b/gui/audiooutputqt.cpp
index 82a3bbe..4363833 100644
--- a/gui/audiooutputqt.cpp
+++ b/gui/audiooutputqt.cpp
@@ -341,7 +341,11 @@ qint64 AudioIODevice::readData(char *data, qint64 len)
         }
         else
         {   // not enough samples ==> inserting silence
+            #ifdef Q_OS_MACOS
             qCDebug(audioOutput, "Muted: Inserting silence [%llu ms]", bytesToRead / (m_bytesPerFrame * m_sampleRate_kHz));
+            #else
+            qCDebug(audioOutput, "Muted: Inserting silence [%lu ms]", bytesToRead / (m_bytesPerFrame * m_sampleRate_kHz));
+            #endif
 
             memset(data, 0, bytesToRead);
 
@@ -480,7 +484,11 @@ qint64 AudioIODevice::readData(char *data, qint64 len)
     }
     else
     {   // mute can be requested when there is not enough samples or from HMI
+        #ifdef Q_OS_MACOS
         qCInfo(audioOutput, "Muting... [available %llu samples]", numSamples);
+        #else
+        qCInfo(audioOutput, "Muting... [available %lu samples]", numSamples);
+        #endif
         if (numSamples < AUDIOOUTPUT_FADE_TIME_MS*m_sampleRate_kHz)
         {
             float gain = 1.0;
```

There is no message anymore:

```
[ 26%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/audiooutputqt.cpp.o
[ 27%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/abracadabra_qmltyperegistrations.cpp.o
[ 28%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/epg_qml.cpp.o
[ 29%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGItem_qml.cpp.o
[ 30%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGSwitch_qml.cpp.o
[ 31%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGTabBar_qml.cpp.o
[ 31%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGTabButton_qml.cpp.o
[ 32%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGColors_qml.cpp.o
[ 33%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGHorizontalShadow_qml.cpp.o
[ 34%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/.rcc/qmlcache/AbracaDABra_qml/EPGScrollableText_qml.cpp.o
[ 35%] Linking CXX executable AbracaDABra
[100%] Built target AbracaDABra
```

KejPi, 2024-03-05:
Warning is only generated under Ubuntu, no problem on macOS, Windows or ArchLinux.

closed 2024-03-05 by KejPi

andimik, 2024-03-06:
But a user (like me) will wonder about the (only) warning message (meaning that the programming is very good for the rest of the files). My patch will prevent from showing the build-warning at all.

KejPi, 2024-03-06:
It is only happening under Ubuntu, not clear why. However I have committed a change  1f777d97f561537fe077f20da89c6bf92440a1a3 that should fix it and make the behavior unified under all OS.

tomneda, 2024-03-07:
@KejPi + @andimik : I had in my (also a DAB) project the issue that the `%llu` or `%lu` in the `printf()` method would not fit for both Windows and Linux (Ubuntu 22.04) while accessing an `uint64_t` variable. 

I could solve this with a special macro which is differently defined in Windows and in Linux. 
See https://github.com/tomneda/DABstar/commit/4915da9b3431d8a17913a25502fa86f8dfd8a2bd .

Linux defines `%PRIu64` as `%lu`, Windows as `%I64u`

Hope this will help you in your case.

KejPi, 2024-03-08:
@tomneda Thanks for the hint. Since I do not actually need full int64 range in the log message I did typecast to int and it seems to work too bacause @andimik stopped complaining 😁 In any case, I still consider this as no issue at all, it is only a warning. There are many projects out there with so many scary warnings during compilation and still they work and are used massively.

andimik, 2024-03-08:
I was not complaining anymore as I have a patch ;-)

tomneda, 2024-03-09:
@KejPi I can unterstand @andimik. I like my code completely warning-free (with `-Wall`). Especially there are really important warnings and with too many less important shown warnings a important warning could be overseen.

I have an example of a really important warning about mixing a signed `int` with an unsigend `int` in an if-clause:
(`warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’`)

Go to https://www.onlinegdb.com/online_c_compiler and try out this code:

```
#include <stdio.h>

int main()
{
    int s = -4;
    unsigned int u = 5;
    
    if (s > u)
    {
        printf("s (%d) > u (%u)", s, u);
    }
    else
    {
        printf("s (%d) <= u (%u)", s, u);
    }

    return 0;
}

```

The result is 
`s (-4) > u (5)`

Were you aware of that? 😁
I intuitively expected that the unsigned `int` is converted to an signed `int` but not the other way around.

KejPi, 2024-03-09:
Yes, I am aware of that sign related issues.
I like my code without warnings too, and it is clean under macOS on Apple Silicon that is my main OS. I do not care much about other OS as long as the project is compilable and there are no OS specific application problems reported. Actually, the most interesting part about that int64 related warning was that it appeared only under Ubuntu not on other platforms I use to compile the code.