KejPi/AbracaDABra #126: Source code for libdabsdr missing?

ksooo, 2024-03-18:
Hi, are there any plans to open source the code of libdabsdr?

```
if(EXISTS "${PROJECT_SOURCE_DIR}/dabsdr/CMakeLists.txt" AND NOT PROJECT_VERSION_RELEASE)
    message (STATUS "libdabsdr will be built from source")
    set(DABSDR_DIR "${PROJECT_SOURCE_DIR}/dabsdr")

    # dab library source code available
    add_subdirectory(${DABSDR_DIR} dabsdr)

    set(DAB_LINK_LIBRARIES dabsdr)
    set(DAB_INCLUDE_DIRS ${DABSDR_DIR}/src ${PROJECT_BINARY_DIR}/dabsdr/src)
else()
```

KejPi, 2024-03-18:
Hi, currently I have no plans to release source code of the library.

closed 2024-03-18 by KejPi

ksooo, 2024-03-18:
Okay, but this makes learning how things work /  debugging etc. in order to contribute to this project a real challenge. You really should consider making this open source if you want to build an actual community around your cool project.

KejPi, 2024-03-18:
You are not the first telling this but I still prefer to keep library as closed source, the library is actually the reason why AbracaDABra is so good in sensitivity and CPU usage. 
Most of the code is freely available, only demodulation and decoding library is closed so if you want to contribute to the project there is a lot of room for improvements ;-)

ksooo, 2024-03-18:
> the library is actually the reason why AbracaDABra is so good in sensitivity and CPU usage.

I cannot see how this is related to being open source. You as the project owner could be in full control which PRs are merged. 

But ofc your decision.

KejPi, 2024-03-18:
I want to keep my know-how ;-)

ksooo, 2024-03-18:
> I was not satisfied with features that could be supported by Kodi API

You are just talking to a guy knowing a lot about Kodi API. I hope you tried to implement a so-called PVR client add-on? From my point of view, that API provides everything you need. There is support for EPG, recordings, RDS, meta data, ... aside from the basics channels, channel groups, ...

ksooo, 2024-03-18:
Interesting, but I'd rather say almost everything is available, only some DAB specific features are missing in the API. DL can be handled easy this days by providing an "ID3 tag stream", which imo also could be used to implement SLS.

KejPi, 2024-03-18:
Maybe you are right, I have no experience with Kodi plugins development, I have just tried to do it starting from example template. But I am still not sure it is worth the effort.

ksooo, 2024-03-18:
> But I am still not sure it is worth the effort.

I thin it definitely is. A DAB(+) add-on for Kodi was requested by our users several times and I would be willing to implement and maintain it.

KejPi, 2024-03-18:
Good luck then!

ksooo, 2024-03-18:
> Good luck then!

Sure, I will find my way, with or without your support. With all the "want to protect my know-how" attitude I hope you do not forget how many people shared their know how with you so that you were able to get that far with your project. Imagine you would have to implement QT, rtl-sdr, ... everything on your own.

KejPi, 2024-03-18:
I am sharing most of my code under MIT license and the library is under MIT license too, but not open source. The API is also public, you can use it without restrictions. I can support you if you want to integrate the library in Kodi plugin.