KejPi/AbracaDABra #127: wrong counting of found frequencies

andimik, 2024-03-25:
I just had the strange situation that one frequency was counted twice. So the number of ensembles was 1 too high.

If you look into the log, then you see that 8A (Austrian Mux I from Graz) was counted twice.

[image]

```
23:03:46.366 [I] RadioControl: Current frequency 174.928 MHz
23:03:49.491 [I] RadioControl: Tuning 174.928 -> 176.640 MHz ...
23:03:50.366 [I] RadioControl: Current frequency 176.640 MHz
23:03:53.490 [I] RadioControl: Tuning 176.640 -> 178.352 MHz ...
23:03:54.366 [I] RadioControl: Current frequency 178.352 MHz
23:03:57.491 [I] RadioControl: Tuning 178.352 -> 180.064 MHz ...
23:03:58.366 [I] RadioControl: Current frequency 180.064 MHz
23:04:01.492 [I] RadioControl: Tuning 180.064 -> 181.936 MHz ...
23:04:02.366 [I] RadioControl: Current frequency 181.936 MHz
23:04:04.815 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] Rock Antenne       E0AD27 : 0
23:04:04.816 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] jö.live            E0AD56 : 0
23:04:04.819 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] arabella RELAX     E0AD54 : 0
23:04:04.825 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] KLASSIK RADIO      E0AD53 : 0
23:04:04.828 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] ENERGY             E0AC51 : 0
23:04:04.831 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] * 88.6 *           E0AC47 : 0
23:04:04.835 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] -AUSTRIA           E0A3E0 : 0
23:04:04.838 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] Radio Flamingo     E0AD5A : 0
23:04:04.842 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] Antenne Österrei   E0AD57 : 0
23:04:04.848 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] arabella HOT       E0AD59 : 0
23:04:04.852 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] ERF Süd            E0AD24 : 0
23:04:04.856 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] 'WELLE 1           E0AD58 : 0
23:04:04.866 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] Mein Kinderradio   E0AD55 : 0
23:04:04.868 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] Radio Maria        E0A3DD : 0
23:04:04.871 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] *STEPHANSDOM*      E0AC52 : 0
23:04:04.878 [I] ServiceList:           [E0A101 @ 181936 kHz |  6A] #Radio ONE         E0AD2A : 0
23:04:04.888 [I] RadioControl: Tuning 181.936 -> 183.648 MHz ...
23:04:05.817 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD27, SCIdS 00: status 2
23:04:05.818 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD56, SCIdS 00: status 2
23:04:05.821 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD54, SCIdS 00: status 2
23:04:05.826 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD53, SCIdS 00: status 2
23:04:05.830 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AC51, SCIdS 00: status 2
23:04:05.833 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AC47, SCIdS 00: status 2
23:04:05.836 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0A3E0, SCIdS 00: status 2
23:04:05.840 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD5A, SCIdS 00: status 2
23:04:05.845 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD57, SCIdS 00: status 2
23:04:05.850 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD59, SCIdS 00: status 2
23:04:05.854 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD24, SCIdS 00: status 2
23:04:05.858 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD58, SCIdS 00: status 2
23:04:05.862 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0A0AD50, SCIdS 00: status 2
23:04:05.867 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD55, SCIdS 00: status 2
23:04:05.869 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0A3DD, SCIdS 00: status 2
23:04:05.874 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AC52, SCIdS 00: status 2
23:04:05.880 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E0AD2A, SCIdS 00: status 2
23:04:05.966 [I] RadioControl: Current frequency 183.648 MHz
23:04:08.991 [I] RadioControl: Tuning 183.648 -> 185.360 MHz ...
23:04:09.966 [I] RadioControl: Current frequency 185.360 MHz
23:04:12.992 [I] RadioControl: Tuning 185.360 -> 187.072 MHz ...
23:04:13.965 [I] RadioControl: Current frequency 187.072 MHz
23:04:16.992 [I] RadioControl: Tuning 187.072 -> 188.928 MHz ...
23:04:17.965 [I] RadioControl: Current frequency 188.928 MHz
23:04:20.991 [I] RadioControl: Tuning 188.928 -> 190.640 MHz ...
23:04:21.565 [I] RadioControl: Current frequency 190.640 MHz
23:04:24.491 [I] RadioControl: Tuning 190.640 -> 192.352 MHz ...
23:04:25.565 [I] RadioControl: Current frequency 192.352 MHz
23:04:28.491 [I] RadioControl: Tuning 192.352 -> 194.064 MHz ...
23:04:29.565 [I] RadioControl: Current frequency 194.064 MHz
23:04:32.491 [I] RadioControl: Tuning 194.064 -> 195.936 MHz ...
23:04:33.565 [I] RadioControl: Current frequency 195.936 MHz
23:04:37.196 [I] ServiceList:           [E0A101 @ 195936 kHz |  8A] 'WELLE 1           E0AD58 : 0
23:04:37.198 [I] ServiceList:           [E0A101 @ 195936 kHz |  8A] *STEPHANSDOM*      E0AC52 : 0
23:04:37.200 [I] ServiceList:           [E0A101 @ 195936 kHz |  8A] Mein Kinderradio   E0AD55 : 0
23:04:37.203 [I] ServiceList:           [E0A101 @ 195936 kHz |  8A] ERF Süd            E0AD24 : 0
23:04:37.205 [I] ServiceList:           [E0A101 @ 195936 kHz |  8A] #Radio ONE         E0AD2A : 0
[... 187 lines cut ...]
23:05:45.834 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E49422, SCIdS 00: status 2
23:05:45.841 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E49569, SCIdS 00: status 2
23:05:45.850 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E4941E, SCIdS 00: status 2
23:05:45.857 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E4941F, SCIdS 00: status 2
23:05:45.863 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E4910001, SCIdS 00: status 2
23:05:45.871 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E49421, SCIdS 00: status 2
23:05:45.881 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E49411, SCIdS 00: status 2
23:05:45.888 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E49409, SCIdS 00: status 2
23:05:45.894 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E4943C, SCIdS 00: status 2
23:05:45.900 [W] RadioControl: DABSDR_NID_USER_APP_LIST SId E4952E, SCIdS 00: status 2
23:05:45.964 [I] RadioControl: Current frequency 229.072 MHz
23:05:48.991 [I] RadioControl: Tuning 229.072 -> 230.784 MHz ...
23:05:49.964 [I] RadioControl: Current frequency 230.784 MHz
23:05:52.992 [I] RadioControl: Tuning 230.784 -> 232.496 MHz ...
23:05:53.964 [I] RadioControl: Current frequency 232.496 MHz
23:05:56.990 [I] RadioControl: Tuning 232.496 -> 234.208 MHz ...
23:05:57.964 [I] RadioControl: Current frequency 234.208 MHz
23:06:00.991 [I] RadioControl: Tuning 234.208 -> 235.776 MHz ...
23:06:01.964 [I] RadioControl: Current frequency 235.776 MHz
23:06:04.991 [I] RadioControl: Tuning 235.776 -> 237.488 MHz ...
23:06:05.964 [I] RadioControl: Current frequency 237.488 MHz
23:06:08.991 [I] RadioControl: Tuning 237.488 -> 239.200 MHz ...
23:06:09.964 [I] RadioControl: Current frequency 239.200 MHz
23:06:12.991 [I] RadioControl: Tuning 239.200 -> 181.936 MHz ...
```

KejPi, 2024-03-26:
Thanks for reporting. I use band scan very rarely so I have never had this issue. Is it reproducible with that ensemble? I should have recording of Austrian Mux from Insbruck from February this year so I will check it if there is anything specific that might cause the issue.

andimik, 2024-03-26:
I have an idea, what about ECC, when it is received only some frames too late? Will the app discover two muxes then?? 

8A is weak and might cause some issues here.

KejPi, 2024-03-26:
You are right, it could be releated to ECC in some way.

KejPi, 2024-03-26:
ECC is handled in the library and it does not generate log.

andimik, 2024-10-18:
I'll close it, this seems to happen very very rarely. Could not reproduce it in the past months.