KejPi/AbracaDABra #128: Only one SLS pictures per service stored when filename is the same

andimik, 2024-04-15:
The function "Data storage" - "Overwrite existing files" does not work as expected.

I have lots of DAB services where the filename stays the same `PAD1_SLS1.JPG` for example.

But as soon a new picture is transmitted (with same file name), it's not saved.

[image]

When I tick that option, it will be saved, but I'll lose the old one.

[image]

# Idea

Check the file size, and as soon as it differs from the previous one, it's a new picture, which shall be stored (maybe with an additional time stamp in the file name).

KejPi, 2024-04-16:
Well, it works as expected. If the file exist is is either overwritten (if set in settings) or dropped.
If you want to store everything as it is received, you should use transportId in the pattern, for example something like: {transportId}_{contentNameWithExt} 
This shall be unique for a long time.

andimik, 2024-04-16:
oh, what an interesting undocumented feature! That's exactly what I was looking for. :-)

You have explained me SPI and `{directoryId} | transport ID of directory [only SPI]` is only updated once a day, so I thought I could NOT use the transport ID here.

Maybe you can add this to the readme and add the transport ID to the popup in the tool tip?

closed 2024-04-16 by andimik

KejPi, 2024-04-16:
Documentation is always painful task. I thought it was clear from the table that transportId applies to any MOT object and to both SLS and SPI application. I will try to improve the description.

andimik, 2024-04-16:
As you can see, not even I have understood that :-)